doudar/SmartSpin2k #25: Carrier PCB and crank

kswiorek, 2020-11-09:
I could design a carrier PCB for the three componets, that would fit in this space:
[image]
You don't have to use a buck converter for that, just a small power regulator is enough as the ESP doesn't take much power. 
I dont have the exact same ESP module, so I could either use mine, wait a month untill it arrives from China, or you could send me the distance between the headers on yours.

I could also provide code and the documentation for a DIY crank power meter. I also have a version that emulates the power meter directly on the crank, so it would be able pair with yours.

doudar, 2020-11-09:
I'm taking a bunch of measurements in F360 right now to figure out what orientation may work. I'll get back to you very shortly.

doudar, 2020-11-09:
Alright! I'm back! Thank you for taking a look btw!!!

I looked at putting a PCB into the top cover and maybe it could be done but the stepper motor nearly touches the cover, so there isn't much space. With a very small amount of rearranging, it could be possible however:
[image]

The biggest issue I see with this is that access to the boot button (shouldn't be needed but occasionally still is) might become complicated because I don't see a way of doing this without making a "PCB sandwich" with the boards attached to pin headers on opposite sides of the PCB like this:
[image]

So after careful consideration, I think it might be best to keep the esp32 in it's current location and just add stack onto the back side with the PCB containing headers on each side and the power supply on the lower half of the board. The upper half (nearest the usb interface on the esp32 should be where the stepper driver mounts, otherwise the shroud for the gear case may interfere with the heat sink on the stepper driver. 
[image]

There's currently about 40mm from the back side of the ESP32 to the side of the stepper motor, and the full stack I'm estimating at around 36mm.  

Another consideration if we decide to make a custom PCB, is that I'd really like to add serial support for the TMC2225, so a PCB should probably contain the those traces. I haven't done much research into what exactly is required here, but being able to adjust the driver amperage via software and/or detect stalls and travel limits would be invaluable on the software feature side of things so it cannot be ignored. 

Good Luck!!!! Let me know

kswiorek, 2020-11-09:
Could you give the whole cad assembly with the motor and board? I haven't found it in the repository.
I have some experience with TMC2130 - they use SPI, but are otherwise simmilar.

kswiorek, 2020-11-09:
The whole stack is quite thin - i don't use the normal headers:
[image]

doudar, 2020-11-09:
I'm waiting on the F360 export. It's taking a while. 

In the mean time, here's the info for the driver:
https://github.com/bigtreetech/BIGTREETECH-TMC2225-V1.0/blob/master/TMC2225%20V1.0%20manual.pdf

I have no issues sending some esp32's your way if you're committed to helping with this. I'd just need your address.

kswiorek, 2020-11-09:
I don't think international shipping would make sense. I found that I can order these boards locally so I will probably do that. I have an idea how to make it, so I will just design it and show you.

doudar, 2020-11-09:
The assembly is uploaded. If your idea only takes up a few mm I'm open to raising the recess in the cover a bit. I'd prefer to keep it to an absolute minimum though because it already impedes being able to lower the handlebar stem completely on certain bikes.

doudar, 2020-11-09:
But your idea does look very thin! Awesome!

kswiorek, 2020-11-16:
[Smart_trainer.zip](https://github.com/doudar/SmartSpin2k-PlatformIO/files/5547010/Smart_trainer.zip)
Ok, finally. After about an hour messing around with fusion I designed the PCBs. They weren't tested and they shoud fit, but I didn't manage to combine them with the assembly. 
One board uses the A4988, the other a TMC2130, because I didn't know how to set the TMC2225 up. The TMC2130 has a nice library with software SPI. 
This whole thing fits inside the "Tall_Cover" and could be made to hold with some latches, but I don't have the means to make the edge repeatably, so it needs to be tested with a milled PCB
I will try to test the electronics soon, but if I didnt make any mistakes with the order of pins, it should work.

doudar, 2020-11-16:
Thank you - I haven't looked yet, but I included them in the hardware/PCB folder.

I'll see if I can pull these into Fusion to make sure they fit and then if so, I'll try to get some made. 

The only reason I have the 2225 is that they are/were cheaper than the 2130 and they seem to work well. I *think* the pinout is the same and they may be compatible with the library, but I haven't messed around with either yet. The 2225 is single wire UART whereas the 2230 is SPI. I'm not really sure what the difference is as I thought SPI was basically single wire UART...  I guess I'll do some research! 

Don't hesitate to ask if you have any issues/questions if you build my project. While I've made a bunch of prototypes, I don't think anyone else has built it yet. (It hasn't been public long). I am fairly confident on it being robust - I've got probably 15 hours of regular (non-testing) riding with it and countless hours testing. 

Thanks for your contribution!

kswiorek, 2020-11-16:
It's worse :) the other one is flipped. The power supply pins should be near the edge, so the usb can be used from outside

kswiorek, 2020-11-16:
[universal.zip](https://github.com/doudar/SmartSpin2k-PlatformIO/files/5549113/universal.zip)
Ok, now I created a universal board that connects all the possible setup pins to the ESP (there is a lot GPIO left) and moved the buttons to the other side. You can delete the old PCBs with the mistake

This PCB is designed with quite large tolerances, so it's possible to make it at home with a laser printer

doudar, 2020-11-25:
Thank you very much for your contribution. It taught me how to walk through the PCB design process in Eagle, which I hadn't done before.

I ended up redesigning this to fit on the ESP32 mount on the side of the case because I couldn't come close to fitting it once the heat sinks were on the modules. I also think this position is a little better for heat dissipation and inevitable sweat ingress. 

I mostly kept your original schematic but I did have to change a couple things. The shifters are pull_up for less issues with bouncing & EMI. My software debounce also seems to be pretty robust, so I was able to get rid of the debounce caps. I also added some filtering to the 7805 because I'm guessing the stepper driver is making some pretty big ripples. 

I've since had a limited run fab'd (dang that's cheap nowadays!) and they are pretty good. Worked on the first try!

I do plan on making a couple more changes however. I haven't measured it, but the current draw on the ESP32 must be fairly high because the 7805 is getting too hot to touch. I could heat sink it, but since I'd like to try to stay with passive cooling, I'm going to try some TO-220 form factor switching power supplies. (RECOM R-78E5.0-0.5) https://bit.ly/3m87KPc

I also am unsure if I should keep the 5v and then power both modules directly with 5v, or switch the power supply out to a 3.3v model (to bypass using the 3.3v regulator on the ESP devkit. My first prototype with the PCB is putting out much less torque than before, but there were other variables changed (serial controlled amperage and other settings) on top of switching from 5v to the 3.3 off the ESP32. 

Anyway, lots of rambling. hopefully some of it is coherent. 

[image]

kswiorek, 2020-11-26:
That's nice! The 3.3V regulator would probably help. I didn't even think of that, because my design has to have 5V for the LCD. I'm using just this single regulator and it gets hot - by feel I'm guessing about 80°C, but the maximum rating is 125°C, so it probably won't be a problem.

doudar, 2020-11-26:
I did some testing and powering the driver directly with 5v doesn't make any difference. Everything seems to be operating the same either way. I'm able to get plenty of torque by setting driver rms current to 2000 (which seems very high) where my other serial prototype (non-pcb) subjectively puts out the same torque at roughly half that setting.

It could be this specific driver, or it could be that I changed something via serial that I shouldn't have on the driver (shiny new settings, so I had to mess with them all ;) )

doudar, 2020-11-29:
New PCB is working great as well as the DC switching power Supply. Marking this closed!

closed 2020-11-29 by doudar