doudar/SmartSpin2k #32: Occasional reset when handling 2 servers and one client

doudar, 2020-12-15:
Client connection works perfect with Zwift alone. 

If HR and Speed/Cadence/Power is paired and sent through to Zwift, we experience an occasional reset. More debugging is needed. This could be a divide by zero somewhere, a BLE stack overflow, or possibly just data received from one of the servers that is not handled properly by NimBLE.

doudar, 2020-12-16:
Maybe just maybe @h2zero will come by here, take a look at my BLE spaghetti and give a helping hand. One can dream...right??

h2zero, 2020-12-16:
Had a quick look, nothing obvious but there's a lot to look at. Any chance you have a log of the error to narrow it down?

doudar, 2020-12-16:
Whoa! Thank you! 

I'm on the road any away from any esp32's to get the most accurate (recent) logs or backtrace. I was able to grab some logs from my last testing session and have posted them below. They may be helpful but I'm not 100% certain that they are from the very latest .elf. If you'd like to save your effort for later, I'll be back to my workstation this weekend and can post the latest info.

Otherwise, if you have access to a 4mb esp32 devkit, I have the most current binaries in https://github.com/doudar/OTAUpdates for your testing pleasure.

To clarify the issue, the program seems to be stable when running as a server alone and fairly stable with one BLE client connected, but things fall off the rails with two clients connected (i.e. power meter and heart rate monitor), and a server running. Random crashes every 30 minutes or so. It could very easily be a divide by zero in one of my calculations but I'm starting to wonder if it's either task saturation or something in the BLE library.

ble_hs_hci_evt_acl_process(): conn_handle=2 pb=2 len=14 data=0x0a 0x00 0x04 0x00 0x12 0x1b 0x00 0x11 0x00 0x00 0x06 0x00 0x28 0x33 
rxed att command: write req; conn=2 handle=0x001b
11 0 0 6 0 28 33 <-- From APP  Target Incline: 0.06
txed att command: write rsp; conn=2 
host tx hci data; handle=2 length=5
ble_hs_hci_acl_tx(): 0x02 0x00 0x05 0x00 0x01 0x00 0x04 0x00 0x13 
Number of Completed Packets: num_handles=1
handle:0 pkts:1
Disconnection Complete: status=0 handle=0 reason=8
E NimBLERemoteService: "characteristicDiscCB() rc=7 "
E NimBLERemoteService: "Could not retrieve characteristics"
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x401b4b67  PS      : 0x00060730  A0      : 0x800d8194  A1      : 0x3ffd3b50  
A2      : 0x00000000  A3      : 0xe8c762f9  A4      : 0x3ffc45e4  A5      : 0xe8c762f9  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800ea1e8  A9      : 0x3ffd3b00  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000007  A13     : 0x00000000  
A14     : 0x0000000e  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000018  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x401b4b67:0x3ffd3b50 0x400d8191:0x3ffd3b70 0x400d8a3c:0x3ffd3c20 0x40093bf9:0x3ffd3c50

--------------------------------------------------------------------------------------------------------------------

6 0 <-- 0x2a37 HRM: 0
GATT procedure initiated: notify; att_handle=33
txed att command: notify req; conn=1 handle=0x0021
host tx hci data; handle=1 length=20
ble_hs_hci_acl_tx(): 0x01 0x00 0x14 0x00 0x10 0x00 0x04 0x00 0x1b 0x21 0x00 0x54 0x08 0x10 0x0c 0xb4 0x00 0x00 0x00 0x00 0xbd 0x00 0x00 0x00 

23 0 bd 0 0 ae 1 8c 79 <-- CPMC sent 
abort() was called at PC 0x4008359d on core 0

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x4008ea04:0x3ffbded0 0x4008ec81:0x3ffbdef0 0x4008359d:0x3ffbdf10 0x400836c1:0x3ffbdf40 0x401108fb:0x3ffbdf60 0x40109d35:0x3ffbe220 0x40103775:0x3ffbe270 0x400eb86b:0x3ffbe2a0 0x401b3526:0x3ffbe2c0 0x401a7652:0x3ffbe2e0 0x4001791d:0x3ffbe300 0x401a7c55:0x3ffbe320 0x401a7d32:0x3ffbe360 0x400188f5:0x3ffbe390 0x400175ee:0x3ffbe3d0 0x40017688:0x3ffbe3f0 0x4008d256:0x3ffbe420 0x4008d431:0x3ffbe440 0x401a75ff:0x3ffbe460 0x4008d448:0x3ffbe480 0x40081ab5:0x3ffbe4a0 0x400883b3:0x3ffb9bf0 0x4008d1fd:0x3ffb9c10 0x401b36c5:0x3ffb9c30 0x4019dcc8:0x3ffb9c50 0x400eb6ec:0x3ffb9c70 0x400827d7:0x3ffb9da0 0x40093bf9:0x3ffb9dc0

------------------------------------------------------------------------------------------------------------------------------------------------

E NimBLEClient: "characteristicDiscCB() rc=7 "
E NimBLEClient: "Could not retrieve services"
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400ea1da  PS      : 0x00060830  A0      : 0x800d8188  A1      : 0x3ffd3f30  
A2      : 0x00000000  A3      : 0x3ffd3fb4  A4      : 0x3ffc45e4  A5      : 0x56b25ddd  
A6      : 0x3ffd8c00  A7      : 0x3ffd8c00  A8      : 0x800e9014  A9      : 0x3ffd3f00  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000007  A13     : 0x400e90c8  
A14     : 0x3ffc28a0  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x400ea1da:0x3ffd3f30 0x400d8185:0x3ffd3f70 0x400d8a70:0x3ffd4020 0x40093bf9:0x3ffd4050

---------------------------------------------------------------------------------

host tx hci data; handle=Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x401b4b7e  PS      : 0x00060830  A0      : 0x800d7f48  A1      : 0x3ffd3f50  
A2      : 0x00000000  A3      : 0x00000bb8  A4      : 0x3ffc2424  A5      : 0x80a3c586  
A6      : 0x00000001  A7      : 0x80093bfc  A8      : 0x0000005d  A9      : 0x3ffd3f30  
A10     : 0x00000001  A11     : 0x0002d915  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060420  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000010  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x401b4b7e:0x3ffd3f50 0x400d7f45:0x3ffd3f70 0x400d8a70:0x3ffd4020 0x40093bf9:0x3ffd4050

------------------------------------------------------------------------------------------

h2zero, 2020-12-16:
>LoadProhibited, StoreProhibited

>This CPU exception happens when application attempts to read from or write to an invalid memory location. The address which was written/read is found in EXCVADDR register in the register dump. If this address is zero, it usually means that application attempted to dereference a NULL pointer. If this address is close to zero, it usually means that application attempted to access member of a structure, but the pointer to the structure was NULL. If this address is something else (garbage value, not in 0x3fxxxxxx - 0x6xxxxxxx range), it likely means that the pointer used to access the data was either not initialized or was corrupted.

Looks like something may be accessing a nullptr. Without a backtrace decode it's hard to determine where/why.

h2zero, 2020-12-16:
I think I see a couple problems. At [this line](https://github.com/doudar/SmartSpin2k/blob/7bcca38c8e0f86687bcb587ad8de95973cd44c6f/src/SBBLE_Server.cpp#L291) at client instance is being reused but creating a **new** callback class instance, so potential memory leak. 

In the next few lines you have:
```
BLERemoteService *pRemoteService = pClient->getService(serviceUUID);
pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
if (pRemoteCharacteristic == nullptr)
```
You are calling `getCharacteristic()` on pRemoteService that might be nullptr if the call to `getService()` was not successful. So you should test that there before the `getChracteristic()` call.

doudar, 2020-12-17:
The memory leak seems very probable as it's an issue that seems to develop over time. I'll look into it. 

Thank you so much!

doudar, 2020-12-17:
I just pushed that commit. I see exactly where I was checking the characteristic for nullptr instead of the service. Doh! It's amazing you could pick that needle out of a haystack so fast. Thanks!

Can't wait to test it out to see if it fixes the issue.

h2zero, 2020-12-17:
You're welcome, the logs did most of the work 😄. 
I just looked at your commit and I think you should still check if the characteristic pointer is null as well.

line 303 
```
if (pRemoteCharacteristic->canNotify())
```

Should maybe change to:
```
if (pRemoteCharacteristic && pRemoteCharacteristic->canNotify())
```

Or some other test you prefer.

doudar, 2020-12-17:
Added the additional test. 

Pushed:      

line 293

pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);

      if(pRemoteCharacteristic == nullptr){
        debugDirector("Couldn't find Characteristic");
        reconnectTries--;
        return false;
      }

h2zero, 2020-12-17:
Looks good, should resolve your issues. The rest of the code looks fine too, let me know when you test it 😄 .

doudar, 2020-12-17:
@h2zero - I just noticed that @kswiorek has a comment in his ble-ftms code that mentions interrupts may break bluetooth. He's using the original bluetooth library. I'm using interrupts to monitor buttons in my code, and it doesn't "seem" to be causing any issues with NIMble. Have you had any reports of interrupts causing problems?

h2zero, 2020-12-17:
No issues that I have seen or heard reports of. I use them in my projects as well. I suppose there could be an issue if too much time was spent in the handler, I just set a flag and return from mine.

doudar, 2020-12-18:
So the good news is that after fixing checking those nullptr checks there it seems to be stable with multiple devices connected again!

Bad news is that it is not getting the client callback on reconnects.

Looking at the code,  I just realized that I commented out both line 291 and 313, when @h2zero probably only suggested 291. 

I'll uncomment 313 see if that verkz.

doudar, 2020-12-18:
Quick reconnect test is good. I'll let this run for an hour and see if we're stable! 

Hopefully it is...I forgot to turn on the NIMble debugging messages :)  Terminal output is getting logged though so I should at least be able to get a backtrace if there's an issue.

h2zero, 2020-12-19:
Glad to hear it's working. Have you seen any other issues so far?

doudar, 2020-12-19:
I had one reset in about two hours that I wasn't able to catch the cause because I was using it on a bike. HTML debugging is great but there's no way to get the backtrace (that I know of) without a serial connection (and I didn't have the HTML debugging up at the time). It was a minute or two after I manually power cycled my heart monitor (to try to induce a problem), but I'm not certain it's related. Maybe I can get some sort of debugging cable setup but I'm a little hesitant with sweat and open electronics.

h2zero, 2020-12-19:
Well that seems to be an improvement at least. I'd be interested in seeing any logs of the crash if you can get it.

doudar, 2020-12-19:
Hot off a benchtest. Pasting here, I haven't even looked at it yet. 

23 0 bd 0 0 95 b f2 da <-- CPMC sent 
abort() was called at PC 0x400835bb on core 1

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x4008ea04:0x3ffd4b10 0x4008ec81:0x3ffd4b30 0x400835bb:0x3ffd4b50 0x400836b1:0x3ffd4b80 0x4011372c:0x3ffd4ba0 0x40083401:0x3ffd4bc0 0x4000bc69:0x3ffd4be0 0x40001850:0x3ffd4c00 0x400dd3c4:0x3ffd4c30 0x400d8d61:0x3ffd4c70 0x40093bf9:0x3ffd4c90

Decoding stack results
0x4008ea04: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 156
0x4008ec81: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 171
0x400835bb: lock_acquire_generic at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c line 148
0x400836b1: _lock_acquire at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c line 167
0x4011372c: get_adjusted_boot_time at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/time.c line 164
0x40083401: _gettimeofday_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/time.c line 298
0x400dd3c4: NimBLECharacteristic::setValue(unsigned char const*, unsigned int) at .pio\libdeps\esp32doit\NimBLE-Arduino\src\NimBLECharacteristic.cpp line 478
0x400d8d61: BLENotify(void*) at src\SBBLE_Server.cpp line 734
0x40093bf9: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

doudar, 2020-12-19:
I'm guessing it's this:

0x400d8d61: BLENotify(void*) at src\SBBLE_Server.cpp line 734

doudar, 2020-12-19:
Bam. Just got another one. Different spot but on notify again. 

23 0 bd 0 0 7c 7 38 2c <-- CPMC sent 
6 0 <-- 0x2a37 HRM: 0
23 0 bd 0 0 44 57 60 a2 <-- 0x2a63 CAD: 89.00
11 0 0 d 1 28 33 <-- From APP  Target Incline: 2.69
abort() was called at PC 0x400835bb on core 1

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x4008ea04:0x3ffd4b30 0x4008ec81:0x3ffd4b50 0x400835bb:0x3ffd4b70 0x400836b1:0x3ffd4ba0 0x4011372c:0x3ffd4bc0 0x40083401:0x3ffd4be0 0x4000bc69:0x3ffd4c00 0x40001850:0x3ffd4c20 0x400dd3c4:0x3ffd4c50 0x400d8c26:0x3ffd4c90 0x400d8d6a:0x3ffd4ce0 0x40093bf9:0x3ffd4d00

Decoding stack results
0x4008ea04: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 156
0x4008ec81: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 171
0x400835bb: lock_acquire_generic at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c line 148
0x400836b1: _lock_acquire at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c line 167
0x4011372c: get_adjusted_boot_time at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/time.c line 164
0x40083401: _gettimeofday_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/time.c line 298
0x400dd3c4: NimBLECharacteristic::setValue(unsigned char const*, unsigned int) at .pio\libdeps\esp32doit\NimBLE-Arduino\src\NimBLECharacteristic.cpp line 478
0x400d8c26: updateCyclingPowerMesurementChar() at src\SBBLE_Server.cpp line 843
0x400d8d6a: BLENotify(void*) at src\SBBLE_Server.cpp line 737
0x40093bf9: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

doudar, 2020-12-19:
Even though it's a different spot, it only seems to be doing it with the Heart rate monitor and a power meter connected. If only a power meter is connected, no issues. I haven't tested it with HR only.

h2zero, 2020-12-19:
Interesting,  it seems to be failing on the line `m_timestamp = time(nullptr);` in the library. I'm not sure why yet but I'll investigate further.

h2zero, 2020-12-19:
This seems to have something to do with being in ISR context when that line is called, perhaps an interaction with the critical section.
If you wouldn't mind testing, could you go to [this line](https://github.com/h2zero/NimBLE-Arduino/blob/9074bbabb85ecca05fa53af807dbfd21d53847f8/src/NimBLECharacteristic.cpp#L476) and change:
```
    portENTER_CRITICAL(&m_valMux);
    m_value = std::string((char*)data, length);
    m_timestamp = time(nullptr);
    portEXIT_CRITICAL(&m_valMux);
```
to:
```
    time_t t = time(nullptr);
    portENTER_CRITICAL(&m_valMux);
    m_value = std::string((char*)data, length);
    m_timestamp = t;
    portEXIT_CRITICAL(&m_valMux);
```

and let me know if that solves the issue?

doudar, 2020-12-19:
Sure thing. I'll make the change now

doudar, 2020-12-19:
Just saving another backtrace here before I restart the ESP32 with your code changes. Looks to be the same as the other two but this time on updateIndoorBikeDataChar();  with the same line in NimBLECharacteristic.cpp

11 0 0 d 1 28 33 <-- From APP  Target Incline: 2.69
11 0 0 d 1 28 33 <-- From APP  Target Incline: 2.69
abort() was called at PC 0x400835bb on core 1

ELF file SHA256: 0000000000000000000000000000000000000000000000000000000000000000

Backtrace: 0x4008ea04:0x3ffd4b60 0x4008ec81:0x3ffd4b80 0x400835bb:0x3ffd4ba0 0x400836b1:0x3ffd4bd0 0x4011372c:0x3ffd4bf0 0x40083401:0x3ffd4c10 0x4000bc69:0x3ffd4c30 0x40001850:0x3ffd4c50 0x400dd3c4:0x3ffd4c80 0x400d8bd7:0x3ffd4cc0 0x400d8d67:0x3ffd4ce0 0x40093bf9:0x3ffd4d00

Decoding stack results
0x4008ea04: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 156
0x4008ec81: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 171
0x400835bb: lock_acquire_generic at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c line 148
0x400836b1: _lock_acquire at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c line 167
0x4011372c: get_adjusted_boot_time at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/time.c line 164
0x40083401: _gettimeofday_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/time.c line 298
0x400dd3c4: NimBLECharacteristic::setValue(unsigned char const*, unsigned int) at .pio\libdeps\esp32doit\NimBLE-Arduino\src\NimBLECharacteristic.cpp line 478
0x400d8bd7: updateIndoorBikeDataChar() at src\SBBLE_Server.cpp line 833
0x400d8d67: BLENotify(void*) at src\SBBLE_Server.cpp line 736
0x40093bf9: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

doudar, 2020-12-19:
The library still works after the change. 

Waiting for the backtrace that hopefully doesn't happen :) Sometimes it happens within a few minutes, but it's also taken a couple hours before. I'll let this run as long as I can but eventually I'll need my heart monitor to work out :)

If we don't catch any errors before then, I'll let it log all night long and see if there's anything to come back to in the morning.

h2zero, 2020-12-19:
Thanks, I think it should solve the problem. Looking forward to longer term results.

doudar, 2020-12-19:
Still chugging along nicely after over two hours. I'm very confident it's fixed.

Closing the issue. 

Woohoo!!!!

Thanks @h2zero ! I would have never fixed that one without you!

closed 2020-12-19 by doudar

h2zero, 2020-12-20:
No problem, I wouldn't have known without your report/logs (I don't use server much). Thanks for testing out the fix.