doudar/SmartSpin2k #54: Add Heart Rate to Power Calculation

doudar, 2021-01-03:
Power Meters are still expensive and a barrier to entry. In order to open this up to more people we should be able to take the heart rate information and do a rough calculation of watt output via some math and user submitted variables. It seems the variables required are:

Age
Gender
Weight
FTP
Max Hr

@drz01 has kindly offered to help with this end of the project. I'm excited to see what we can come up with!

doudar, 2021-01-03:
@drz01 - If it ends up being just a simple equation, we could have this done in no time. If it requires a lookup table that's obviously a lot more work. Either way, I'm all for starting small with just an equation and then working our way up to more complicated means if necessary.

drz01, 2021-01-03:
> @drz01 - If it ends up being just a simple equation, we could have this done in no time. If it requires a lookup table that's obviously a lot more work. Either way, I'm all for starting small with just an equation and then working our way up to more complicated means if necessary.

Couldn't stop thinking about this today since I realized HR2VP would tie up the HR monitor, but SS2K does not allow use of the Companion app (to connect with the Watch), so here it is.

Estimated PO will never be "accurate" as such, but there is at least a mostly linear relationship to heart rate: https://www.cyclinganalytics.com/blog/2013/06/heart-rate-vs-power-chart

But what if FTP cannot be readily measured? According to "Training and Racing with a Power Meter" by Coggan _et al_, CP (critical power) can be used as a surrogate, given by total work performed over time. TWP(EE) in turn can be estimated by a validated HR formula:
https://www.researchgate.net/publication/7777759_Prediction_of_energy_expenditure_from_heart_rate_monitoring_during_submaximal_exercise

These ingredients should make a basic recipe, any additional "secret sauce" notwithstanding (e.g. METS, accelerometer access). But haven't had time to properly work through it today, though would welcome any initial thoughts (before I get too far down the rabbit hole).

doudar, 2021-01-03:
Companion app works with SS2K - the issue is you have to pair your sensors to SS2K prior to starting companion app (if you use companion app for Bluetooth adapter) Otherwise companion app steals all sensors that are not paired to anything weather it’s using them or not. It’s really irritating.

Also, when you pair both your HR and PM to SS2K, it will forward both of those to Zwift. Just select SS2K in all the connection boxes in Zwift. 

I did that initially so that people with AppleTV and/or limited Bluetooth connections can use the SS2K as a Bluetooth multiplexer. 4 sources of data but SS2K only takes one Bluetooth device slot for AppleTV.

doudar, 2021-01-03:
@drz01 - Nice Find! On his website I found this spreadsheet. It looks like it contains all of the information we need to make a HR power projection calculation!

https://drive.google.com/file/d/1vmRp1GJ9aOmBhr9H8acE2cAsWxvFA_q5/view?usp=sharing

[image]

The Formula he uses is: `=((F6*F8)-(F10*F4))/(F8-F4)+(170*((F6-F10)/(F4-F8)))`

Which Translates to `=((s1Pwr*s2HR)-(s2Pwr*S1HR))/(S2HR-s1HR)+(170*((s1Pwr-s2Pwr)/(s1HR-s2HR)))`
Where:
s1HR = Session 1 Heart Rate
s1Pwr = Session 1 Power
s2HR = Session 2 Heart Rate
s2Pwr = Session 2 Power

So we would just take that formula with the user supplied session data and use userConfig.getSimulatedHr() in place of 170 in the formula above. The C++ code would be:

`userConfig.setSimulatedWatts((s1Pwr*s2HR)-(s2Pwr*S1HR))/(S2HR-s1HR)+(userConfig.getSimulatedHr(*((s1Pwr-s2Pwr)/(s1HR-s2HR)));`

doudar, 2021-01-03:
The issues that need to further be resolved will be a modifier for instantaneous change (since the calculation is based on projecting average power over a long time), as well as another modifier for fatigue (as workout duration increases, HR is higher relative to power).

I also think we should come up with the formula calculating average power of a rider of X weight with a typical road bike over a 1 hour course in order to help people with no other means estimate their session 1 and session 2 numbers above. This calculation could easily be placed into a JavaScript helper on the webpage to be created.

doudar, 2021-01-03:
Hilariously you piqued my interest and I nearly have the base functionality completed. I might have this into master yet today.

@fireyeti & @drz01 ;
If either of you can dig up references on aiding the instantaneous part of the calculation that would help a ton. Currently I’m just working on the required data structures and implementing the equation I posted above.

fireyeti, 2021-01-03:
Great article  [here](https://www.cyclinganalytics.com/blog/2013/06/heart-rate-vs-power-chart)

Nice NIH s[tudy](https://pubmed.ncbi.nlm.nih.gov/10527323/) with FORMULA!

Reading the article now, I'll see if I can distill out the formula and post.

fireyeti, 2021-01-03:
I found this additional study from the NIH

https://pubmed.ncbi.nlm.nih.gov/10527323/

Looking to see if some instantaneous calculations are included.

doudar, 2021-01-03:
@fireyeti - here's a Completely untested Binary. I'm headed down to see if it works at all. Unzip it then manually upload it via the "update firmware" page. admin/admin 

unzip it first of course. 

[firmware.zip](https://github.com/doudar/SmartSpin2k/files/5762172/firmware.zip)

It ***should*** calculate power if: A HR is connected, a power Meter is ***not*** connected, and the "simulate power" button is slid right on the Bluetooth Simulator page. It's currently just using some default numbers I plugged into there for Session 1 and session 2. If it works at all and doesn't crash.

fireyeti, 2021-01-03:
http://www.kreuzotter.de/english/espeed.htm

This calculator can assist riders in calculating power to create their linear graph

HR2VP also has a FTP calculator on their site which analyzes a Strava ride, or makes a guess based on 10km running time.

https://bipr.fr/training-with-heart-rate-to-virtual-power

doudar, 2021-01-03:
Okay, so obviously the default numbers suck (at least for me) but this seems to work. Try the latest binary (below) if you'd like. Same Criteria. For some reason the Webpage seems very slow on this latest version. I'll have to track down what's going on. It will read 50w until your HR gets above 90bpm. You can ride this by just looking on the debug screen - it's not ready for a Zwift Session yet. 

[firmware.zip](https://github.com/doudar/SmartSpin2k/files/5762230/firmware.zip)

doudar, 2021-01-04:
Actually...doesn't Strava give us an estimate of watts when you don't have a power meter?

doudar, 2021-01-04:
Version 0.1.1.3 is pushed to OTAUpdates!

It includes the preliminary heart rate to watts. Let me know how it works. 

Also fixed a couple bugs, one of which was occasionally preventing automatic pairing on boot so that's nice.

doudar, 2021-01-05:
I did a little bit of limited testing tonight and it worked better than I expected. I calibrated it using two rides from my strava feed and it actually matched my power meter pretty close on average. Obviously it lags quite a bit on intervals though.

doudar, 2021-01-18:
HR->PWR functionality is working. Need to add instantaneous modifiers. Changing the name of the issue to reflect that.

doudar, 2021-01-20:
@drz01 or @fireyeti - either of you want to give this a try? It might be a great way to start playing with the code to figure it out. You'll find the function at the bottom of bleserver.cpp

void calculateInstPwrFromHR()
{

  int avgP = ((userPWC.session1Pwr * userPWC.session2HR) - (userPWC.session2Pwr * userPWC.session1HR)) / (userPWC.session2HR - userPWC.session1HR) + (userConfig.getSimulatedHr() * ((userPWC.session1Pwr - userPWC.session2Pwr) / (userPWC.session1HR - userPWC.session2HR)));

  if (avgP < 50)
  {
    avgP = 50;
  }

  if (userConfig.getSimulatedHr() < 90)
  {
    //magic math here for inst power
  }

  if (userConfig.getSimulatedHr() > 170)
  {
    //magic math here for inst power
  }

  userConfig.setSimulatedWatts(avgP);
  userConfig.setSimulatedCad(90);

  debugDirector("Power From HR: " + String(avgP));
}

doudar, 2021-01-20:
Where I've got the "magic math" comments is just a suggestion.

I'm guessing you'll want to define 2 or three static long int and set them to track the heartrate changes over time. 

BPM rapidly increasing means higher than average power output. BPM rapidly decreasing means lower than average power output. 

There should probably also be a %drop(fatigue) for every hour spent cycling and a way to automatically reset that timer after a period of inactivity. .

fireyeti, 2021-01-20:
I've drawn out a few graphs to wrap my head around what math would be helpful. I think a multiplier of BPM increase/decrease above 15 second average BOM over time will generate usable data. Having the timer will enable the fatigue percentage to be used as well. I need to research some C++ timer functions to see how a table would be filled.

doudar, 2021-01-20:
I don't know what storage requirement you have or what type of data you'd like to store (in the table), but if you need a fair amount and would like to keep it persistent, look up how JSON works as I'm already using arduinoJSON for storing and reading data to text files.

doudar, 2021-02-17:
Any updates @fireyeti ?

fireyeti, 2021-02-17:
Ummmmm. Does it count that I am still thinking about how to create this output?

Seriously, I am thinking about it, just haven't been able to come up with something usable.

drz01, 2021-03-15:
I am not quite comprehending how the code currently works -- a lot of passing back and forth; it may well be I need to go through the entire code base later. I did go through Grazzi _et al_ from @fireyeti ; the main idea being that the relationship is almost logarithmic -- which gave me an idea.

Delta(HR) is only significant at RHR, but diminishes up the scale: e.g. from 90 to 100 is 1.11, but from 170 to 180 is only 1.06; the "speed of change" over say, 500ms it should be small, but just observable. I am hoping to fully work up the modifier into an "experimental mode".

Modelling fatigue though introduces a lot more complexity: though cardiac drift be be seen _within_ a workout; Allen & Coggan pointed out the opposite is true _between_ workouts, with fatigue introducing a suboptimal HR response.

fireyeti, 2021-03-15:
I am of the opinion that he current algorithm is quite good for steady state riding, and would like to see an additional modifier of “instantaneous” rate of change. As you mention @drz01, 500ms would be sufficient to see a rate change and therefore model a power output increase or decrease.  A fatigue multiplier could be added later down the line after calculation on the below idea.

So, steady HR = steady power based on power calculator

for example (complete estimates based strictly on my unscientific observation)

- if HR increase 1 bpm over 2 second, power output increase by 20%

- if HR increase 2 bpm over 2 second, power output increase by 40%

- if HR decrease 1 bpm over 2 second, power output decrease by 20%, etc etc

Does this seem like a viable, implementable calculation? 

@doudar Is there a quick way to show (either in a log file which could be graphed, or a debug window output) HR calculated power AND actual power meter info? It could be a window on the HR calculation page.

doudar, 2021-03-15:
@drz01 
The whole HR->PWR calculation should be able to be contained within it's own function here.
 https://github.com/doudar/SmartSpin2k/blob/ac8ca3da2f13a826f3e88be3f31f9e174d280b4b/src/BLE_Server.cpp#L387

I think it just needs a rate of change addition and the associated math. I can probably insert the code to provide you guys with the rate of change hr. If only the last two data points are sufficient (are they or do we need 3 or more?), something like this could work as a base rate. Adding more data points is pretty easy if needed. 

double fx = (newHR-oldHR) / (BLE_CLIENT_DELAY/1000);

doudar, 2021-03-15:
> I am of the opinion that he current algorithm is quite good for steady state riding, and would like to see an additional modifier of “instantaneous” rate of change. As you mention @drz01, 500ms [...]

Yes, I could make a quick change with a flag in setting.h to always show calculated power in debugging.

doudar, 2021-03-15:
> I am of the opinion that he current algorithm is quite good for steady state riding, and would like to see an additional modifier of “instantaneous” rate of change. As you mention @drz01, 500ms [...]

Use the develop-release candidate branch and uncomment:
https://github.com/doudar/SmartSpin2k/blob/1d963cdb5aef639891c5f4f7fa60bc21d8b2921d/include/settings.h#L125

It will print the calculated output to the debug but it won't output it via BLE if there's a power meter attached.  
Here's the binary with that flag enabled:
[firmware.zip](https://github.com/doudar/SmartSpin2k/files/6142400/firmware.zip)

drz01, 2021-03-20:
Thanks for the changes! Is BLE_CLIENT_DELAY variable and in ms?

doudar, 2021-12-27:
Stale

closed 2021-12-27 by doudar