doudar/SmartSpin2k #58: Compilation error : redefinition of 'class MyServerCallbacks'

sspieser, 2021-01-05:
Hello, I'm new to VS Code and PlatformIO, so probably the reason why...

Trying to build the project, I get :

Building in release mode
Compiling .pio\build\esp32doit\src\BLE_Client.cpp.o
Compiling .pio\build\esp32doit\src\BLE_Server.cpp.o
Compiling .pio\build\esp32doit\src\BLE_Setup.cpp.o
Compiling .pio\build\esp32doit\src\HTTP_Server_Basic.cpp.o
Compiling .pio\build\esp32doit\src\Main.cpp.o
Compiling .pio\build\esp32doit\src\SBBLE_Server.cpp.o
Compiling .pio\build\esp32doit\src\SmartSpin_parameters.cpp.o
Generating partitions .pio\build\esp32doit\partitions.bin
Compiling .pio\build\esp32doit\lib961\NimBLE-Arduino\FreeRTOS.cpp.o
Compiling .pio\build\esp32doit\lib961\NimBLE-Arduino\NimBLE2904.cpp.o
src\SBBLE_Server.cpp:86:7: error: redefinition of 'class MyServerCallbacks'
 class MyServerCallbacks : public BLEServerCallbacks
       ^
In file included from include/Main.h:14:0,
                 from src\SBBLE_Server.cpp:12:
include/BLE_Common.h:55:7: error: previous definition of 'class MyServerCallbacks'
 class MyServerCallbacks : public BLEServerCallbacks
       ^
src\SBBLE_Server.cpp:545:7: error: redefinition of 'class MyCallbacks'
 class MyCallbacks : public BLECharacteristicCallbacks
       ^
In file included from include/Main.h:14:0,
                 from src\SBBLE_Server.cpp:12:
include/BLE_Common.h:61:7: error: previous definition of 'class MyCallbacks'
 class MyCallbacks : public BLECharacteristicCallbacks
       ^
*** [.pio\build\esp32doit\src\SBBLE_Server.cpp.o] Error 1

Clean/ build did not get out... thx!

doudar, 2021-01-05:
Is this with my current master or from the version that you forked?

doudar, 2021-01-05:
Yes, so it looks like you're using an old version. First try grabbing the most up to date code from the master in my repository

sspieser, 2021-01-05:
I just got the last master ZIP from your github, so... I do it again from scratch then...

doudar, 2021-01-05:
I think what you might have going on is that there are old/leftover files from a previous version, and it also looks like you have files from the newest version that are conflicting. Try deleting your whole local project and installing a new clone in the same location. (the file SBBLE_Server.cpp) does not exist in the newest version.

sspieser, 2021-01-05:
You're right, I do not know that platform or device; I used to code using ArduinoIDE for ESP8266... 

BTW : compilation success, thx a lot !

closed 2021-01-05 by sspieser

doudar, 2021-01-05:
You'll really like PlatformIO once you get the hang of it after using ArduinoIDE. There's a lot more builtin stuff to help debugging and development

doudar, 2021-01-05:
You may also want to do a "run task" "upload file system image" which will upload the contents of the /data folder to the esp32. 

If you don't do this, when you login to the esp32 it's going to ask for your WiFi network so it can download those files from my OTAUpdates folder. Either way they get there.

sspieser, 2021-01-05:
Thanks; I did build and upload, then connected using the phone to set my WIFI settings.
Since then, nothing really happens; the monitor log says : 

```
Reading File: /userPWC.txt
Config File Loaded: /userPWC.txt
Contents of file: /userPWC.txt
{"session1HR":129,"session1Pwr":100,"session2HR":154,"session2Pwr":150} 
Writing File: /userPWC.txt
ion 
 read:current=
ion
Setting up cpu Tasks
Connecting to: Freebox-7D6D6C
.Connected to Freebox-7D6D6C IP address: 192.168.0.3
Open http://SmartSpin2K.local/
Checking for newer firmware:
  -Server Ver 0.1.1.3
  -index.html not found. Forcing update
New firmware detected!
Upgrading from 0.1.1.3 to 0.1.1.3
Updating FileSystem
[E][WiFiClientSecure.cpp:133] connect(): start_ssl_client: -1
[E][HTTPUpdate.cpp:217] handleUpdate(): HTTP error: connection refused

Starting Arduino BLE Client application...
Starting BLE Server
Bluetooth Characteristic defined!
BLE Notify Task Started
Scanning for BLE servers and putting them into a list...
BLE Advertised Device found: Name: , Address: 02:e5:a4:50:f4:c9, manufacturer data: 0600010f2002e5c4e8871b841926bf8fabde7d143d62c3437bf9e6a80d
Bluetooth Client Found Devices: null
Scanning
any any
End BLE Setup
HTTP server started
Creating Shifter Interrupts
```

But the ping is 75% loss of packets, and timeout on browsing using CURL or Chrome...

sspieser, 2021-01-05:
New upload and monitor; this time : 

```
Connecting to: Freebox-7D6D6C
...........Couldn't Connect. Switching to AP mode
AP IP address: 192.168.4.1
Open http://SmartSpin2K.local/
```

So I guess that my WIFI password was wrong, as it has kept de WIFI name ?

Sadly then I can't connect to the SmartSpin2K WIFI :-( ... authentification issue maybe....

doudar, 2021-01-05:
Here are some questions for you to check on:

So it looks like maybe you connected to it once and changed the SSID to your network. Were those (default) pages responsive?

Does the network it's connected to currently have internet access?

Have you tried both http://192.168.0.3/index.html and http://smartspin2k.local/index.html ?

Does your version of the devkit have a built in antenna (black bar at the top with some traces in it). 

Does your Computer's USB port put out enough power for the ESP32? Maybe try another? Maybe try powering it with another source like a phone charger via USB or power supply via the 5v pin. 

Is the network router a long ways away? (ESP32 has pretty good WiFi range but not as good as a computer or phone)

Good Luck, Let me know!

doudar, 2021-01-05:
Yes, to avoid other issues, the SmartSpin2K network will use the password you have entered. I should probably document that somewhere.

 Hey @fireyeti :)

doudar, 2021-01-05:
To Clarify...Currently if it cannot connect to your WiFi network, it sets up an AP with the SmartSpin2K SSID and whatever password was in the Password field. 

That sound confusing...I'll probably revisit how the AP fallback works soon.

doudar, 2021-01-05:
Without shifters connected and without WiFi access, in order to reset the unit to defaults, you'll have to "build and upload filesystem image" with visual studio. 

Uploading files to file system SPIFFS

Run “Upload File System image” task in PlatformIO IDE or use PlatformIO Core (CLI) and pio run --target command with uploadfs target.
To upload SPIFFS image using OTA update please specify upload_port / --upload-port as IP address or mDNS host name (ending with the *.local).

Examples:

SPIFFS for Arduino
SPIFFS for ESP-IDF

sspieser, 2021-01-05:
Too bad!! I've just seen your mention "NEW! - Small quantity of kits availiable!"... would have been far more easier than buying parts one by one on Amazon + find a 3D printer guy to print... anyway : I pluggeg the ESP32 on a smartphone charger + just near my home WIFI box : 192.168.0.3 works !

I get this again : 

```
The webserver files
need to be updated.
Please enter the credentials for your network
or upload a new filesystem image using the
link below.
```

Weird as the credentials are OK... what's wrong?

If I hit "Update firmware", I get an "ESP32 Login" form... 

"smartspin2k.local" does not work

I do not see the same page as shown on the WIKI

Keeping trying anyway!

sspieser, 2021-01-05:
> shifters

Sorry for all my troubles and questions... I did not get : what are the shifters used for ?
Currently I have nothing attached to my ESP32 devkit...

sspieser, 2021-01-05:
> Without shifters connected and without WiFi access, in order to reset the unit to defaults, you'll have to "build and upload filesystem image" with visual studio.
>
> Uploading files to file system [...]

Sorry I'm lost here ; I will need time to understand this tool, seems quiet not intuitive to me (but not developping since ages).

doudar, 2021-01-06:
I've made a couple more minor changes to hopefully help you out. 

There isn't a binary for this yet, you'll have to recompile from master and upload again. 

Changes:
SSID will failback to 'SmartSpin2K' and 'password' if it cannot connect to your home network.

WiFi Power has been hardcoded to maximum. 

Small Bluetooth change which may improve WiFi.

doudar, 2021-01-06:
Two additional things I thought of:
1, Is there anything on your network that may be blocking internet access for the SmartSpin like some sort of landing page or anything?

2, Possibly being in France is causing an issue with getting the correct Github server? 
To test that, try clicking this link in your browser:
https://raw.githubusercontent.com/doudar/OTAUpdates/main/firmware.bin
That link should do nothing other than download the file to your computer. 

3, Is your network a WiFi Mesh network? (It shouldn't matter, but if it is I might have some more ideas to try on my end).

doudar, 2021-01-06:
There's a possibility that Github servers were down when you tried also. Good Luck! 

I'll be out all week next week with limited email. If you have any questions, @fireyeti might be able to help you a bit. 

Excited to help you get this going!