doudar/SmartSpin2k #59: Sorting Out WiFi Distance Issues

doudar, 2021-01-06:
@h2zero, 

I know this is one of your guys favorite topics, but....

Any way to get more range out of the ESP32 WiFi these days with both BLE and WiFi running at the same time (without using esp-idf)?

It seems some of this projects users are having issues with WiFi connectivity at reasonable distances from their routers. Phones and computers can stream videos in these areas and have never had an issue. 

We're using WiFi on this project for an occasional user interface and OTA updates while continuously using the BLE for connection to 2 servers and one client for fitness data (Needs to be refreshed at about 1hz). BLE scans mostly happen at boot but occasionally the user will manually select them during run. 

Thanks for any help - I might have to send you another beer :)

doudar, 2021-01-06:
I changed (increased) my vTaskDelays() in the task that sends BLE notifications from 500ms to 1000ms. Webserver seems a little more responsive. I don't have signal/distance issues with mine, I'm starting to wonder if it has something to do with timing on wireless mesh networks.

h2zero, 2021-01-06:
Well, where do I start... lol. This can be a problematic issue that begins with the single antenna we have to work with. To cut the answer short here are some things to try:

- If scanning, try to keep the window time around 100ms or more greater than the interval time to give WiFi a chance to ping.
- If advertising, same idea, leave some antenna time for WiFi
- as a server try to request connection parameters with min and max intervals greater than 50ms, same trend here :)
- as a client use min and max intervals higher than 50ms and be conservative with the supervision timeout (2 seconds is good)

The RF frequencies of WiFi and BLE are in direct conflict at all times so we need to manage expectations here. Best results come from not mixing the 2 operations, instead opt for a 1 then the other sequence when possible. i.e get all the BLE data operations etc done then deal with WiFi stuff, repeat.

doudar, 2021-01-06:
h2zero - As always, thanks!

I'll give that a try.

h2zero, 2021-01-06:
It can be a lot of trial and error, what I do in my home automation project is scan > connect (if needed) > get data > disconnect > send to MQTT (reconnect WiFi first if needed). This is controlling my furnace etc. so I prioritize the BLE data and controlling things locally over sending the data to a server which loses connection occasionally (so my heat still works if the internet goes down 😄).
  
It's really stable once you get the settings sorted.

doudar, 2021-01-06:
Well, the good news is that even though all these things are going on (client, server and wifi), none of it is terribly fast. I can get away with 1000ms intervals on the BLE data per device. If I can negotiate it. I honestly think that advertising alone prior to device connection that may have been the source of my issues. I had left the interval at the default. MAX is now 260, min is 160.

h2zero, 2021-01-06:
It depends on the throughput you need vs packet loss expected. If the environment is noisy you'll need more forgiveness (longer timeout). I home that 70ms was a typo, much too small.

doudar, 2021-01-06:
pServer->updateConnParams(desc->conn_handle, 40, 50, 0, 70);

This is my current settings, I just followed some examples and increased them a bit. Any suggestions?

h2zero, 2021-01-06:
Oh, that's 700ms in your settings (*10, BLE spec), that's better then. If you still have issues try setting it to 200.

doudar, 2021-01-06:
Ahhh I see that now....I probably need to get some rest :) The logical part of my brain is getting mushy :)

doudar, 2021-01-06:
And just to clarify, when you say 200 you actually mean 20? Correct? (200ms)

doudar, 2021-01-06:
7

closed 2021-01-06 by doudar

h2zero, 2021-01-06:
Actually I meant 200 as in 2000ms, you'll have a better chance maintaining the connection with lots of stuff going on. Too short and you'll get lots of disconnections.