doudar/SmartSpin2k #60: Possible connection issue with iPhone

h2zero, 2021-01-06:
This may or may not come up in your testing but I have found that newer versions of iOS don't seem to like [updateConnParams](https://github.com/doudar/SmartSpin2k/blob/715743afa8b6395a000d38316b4952e424971ce3/src/BLE_Server.cpp#L308) in the server onConnect callback. In my experience lately nRFconnect will fail to connect until that line is removed, it could be due to the parameters I used or it could be a bug in nRFconnect but I thought I'd mention it in case you run into this.

doudar, 2021-01-06:
Nice...

Thanks for bringing that up. I do seem to remember encountering that issue previously and so I had commented it out until last night. Is it okay to move that line to a conditional test in my notify update loop?

doudar, 2021-01-06:
I'll test nRF connect before I make the change. The Zwift companion iPhone client already has some annoying issues.

h2zero, 2021-01-06:
Yes, you can call that line anytime after the connection, just seems to be an issue if it's in the callback.

doudar, 2021-01-06:
FYI, I just tested an up to date iphone and ipad with nrf connect (subscribed to notifications) and didn't see any issues. But I'll move it out of onConnect

doudar, 2021-01-06:
You know, I don't like setting flags for a one time run if statement where avoidable...Do you think I could get away with a simple vTaskDelay(100 / portTICK_PERIOD_MS); prior to updateConnParams() inside the onConnect callback?

h2zero, 2021-01-06:
Cool that it works for you, I'd need to play with it more to see if I can find out why it happens. Do no put any delays in callbacks, they will stall the stack. What you could do is trigger a task that delays and then makes the call instead.

PR #61 (2021-01-08): Moved updateConnParams out of onConnect
Fixes #60
closed 2021-01-08 by doudar