doudar/SmartSpin2k #62: Newest ESP32 Arduino Framework broke OTA

doudar, 2021-01-08:
It seems like the newest framework release has probably broken automatic updates. 

We haven't developed a fix yet but once it becomes available, existing units will have to preform a manual upgrade via the web interface. 

It has to do with how github redirects http & https traffic and how the ssl doesn't follow.

doudar, 2021-01-08:
I won't be back to work on this for a week. Existing firmwares should be okay (they are stable), just don't do a factory reset or you will lose the settings webpage until I roll out an update. 

I know exactly how to fix this (need to import a new ssl certificate) but I probably won't have time to implement it prior to leaving.

doudar, 2021-01-08:
@sspieser I'm pretty confident this is the issue you ran into. The 2.10 (newest) version of Arduino Framework for the Esp32 requires a local copy of the GitHub security certificate for automatic updates. 

You can still manually upload via the admin page or via usb. I'll get this fixed first thing on the 15th or so when I get back.

sspieser, 2021-01-08:
Thanks; I also can upload using VS Code / PIO I guess, so I always have the
last code :-)

Good Week end !

doudar, 2021-01-08:
It should be fixed in the latest master (I think I fixed it before I left).
Do a new clone and Give it a try, I’m excited to hear if it fixes your
issue.

closed 2021-01-09 by doudar

sspieser, 2021-01-11:
Hello Anthony,
Sorry but I'm very busy and it will last till the end of next week...

I can still say that I just received the hardware printed; I think I forget
the mounting part, and that I printed the from .STL file for the box...
Bottom_Case_PCB.stl
<https://github.com/doudar/SmartSpin2k/blob/master/Hardware/Bottom_Case_PCB.stl>
instead
of Bottom_Case.stl
<https://github.com/doudar/SmartSpin2k/blob/master/Hardware/Bottom_Case.stl>,
right ?

Last thing : I think the SB3 adapter is not correct, but maybe it will be
good enough to try using tape or rubbon or stuff like that ; I will try to
make a photo or video of the issue, in 2 words, it is not deep (height?)
enough and it is a bit too small : it will work if I put a lot of weight on
it to stick the knob.... anyway my english if too bad so you probably can't
understand what I mean, I'll make a photo/ video as soon as I can :-)

Otherwise I have all the electronic parts (except the motor), so I may
begin to solder stull (what I have also never done, I used to work with
breadboards only :-D)

Thanks a lot for your work and help anyway,

Have a nice day,
Sebastien

doudar, 2021-01-12:
Thanks for the feedback!

Yes, the PCB version is for the printed circuit board. It doesn’t have the
attachment points on the inside for the stepper driver or dc/dc converter.
Everything else is still the same. You can probably hot glue/tape your
boards inside if you do not want to print another version.

I can redesign the knob if it is too small. I have started making all of
them deeper since I designed yours too.

Let me know where you need help. I’m busy until Thursday and can provide
more support then!