doudar/SmartSpin2k #65: Testing ESP32 only : Zwift sees constant 100W / 90 rpm

sspieser, 2021-01-14:
Hello,
Just took 5 minutes to play with the ESP32... Paired to WIFI, paired to my Powertap P1 (seen Powertap P1.R, saved). Then Zwift sees the Smartspin2K : but power is 100 W ! 
I assume the ESP32 is not seeing the power from the powertap, or the ESP32 is not really associated, or... and in that case values are broadcasted?

Note:  I did not paired HR ; but HR is of no use for the smart trainer right ? And it's ANT+ anyway....

Thx, rgds, Seb

doudar, 2021-01-15:
I see you closed another similar issue. Were you able to get it working?

To pair and connect, select your power meter from the drop down, select save, then press scan/reconnect.   

If your power meter is connected correctly, In the green debugging window, you should see a string of numbers coming from your power meter that ends with CAD (or some other number) if your power meter is connected correctly.

What you are seeing currently is the result of the automatic HR->PWR function when a powermeter is not detected. 

If that doesn't work, post a couple screenshots of the debugging window and I'll track down what the issue is. 

You are correct. HR doesn't need to be connected, only for HR->PWR calculation if you don't have a power meter. I'm currently adding a feature to disable automatic HR->PWR if it is not wanted.

sspieser, 2021-01-15:
Sorry I opened a duplicate ! No I did not have time to test further, but thanks I hope to find some time to check again as you said... eventough as I remember that's what I did...
The debugging windows did not show anything usefull as I rember (repeatedly shown the URL.. but I will check again)

OK thx, I'll do it again and post screenshots :-)

doudar, 2021-01-15:
Thank you! I look forward to what you find out!

doudar, 2021-01-18:
Btw, in the latest master (I haven't pushed a OTA binary yet because I need more testing), it will now preform a scan/reconnect when you press Save Changes on the Bluetooth connection webpage.

sspieser, 2021-01-18:
Hello,
I've just tried again; I compiled the code from this morning (french
time)...
Not that much working for me...

Smartphone: unable to get the http://smartspin2k.local/ working; need to
enter the IP address, which fails when the autoscan refreshes...
From the computer :
[image: image.png]

[image: image.png]
Then while pedaling, nothing :

[image: image.png]

On the Zwift side, what do I have to get : pair the "smartspin2k" in the
Bluetooth Windows settings? Anything I tried both and Zwift do not see any
"controlable"...

Any help would be appreciated :-)

Note : I think using BLE you should pair both pedals, otherwise you will
get just half of the power... I know that using ANT+, the left pedal is the
hub for the right, but that not working this way in BLE... what do you
think ?

PS : I need to send pictures of the know part, which will not be working...
but to much work this week...

sspieser, 2021-01-18:
Another try, here are the log monitor, if it may help.. to help me :-)

I guess the power number is doubled to get from one side pedal to dual
sided pedal? Anyway I'm trying to understand if the ESP32 really gets power
numbers, or not...

Thx for your help!

Forming a connection to: c4:02:da:69:fb:d2
 - Created clientlld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
 - Connected to server
E NimBLEClient: "<< getRssi(): Not connected"
 - RSSI
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
Failed to find service:0x1818
disconnecting Client
Initiating Server Connection
trying to connect to PM
Forming a connection to: c4:02:da:69:fb:d2
 - Created clientlld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
 - Connected to server
E NimBLEClient: "<< getRssi(): Not connected"
 - RSSI
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
Failed to find service:0x1818
disconnecting Client
Served /bluetoothscanner.html
Initiating Server Connection
trying to connect to PM
Reusing Client
E NimBLEClient: "<< getRssi(): Not connected"
Client RSSI 0
device RSSI -95
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
Reconnect failed
9 left.
Served /style.css
Initiating Server Connection
trying to connect to PM
Reusing Client
E NimBLEClient: "<< getRssi(): Not connected"
Client RSSI 0
device RSSI -95
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
Reconnect failed
8 left.
Sending BLE device list to HTTP Client
Initiating Server Connection
trying to connect to PM
Reusing Client
E NimBLEClient: "<< getRssi(): Not connected"
Client RSSI 0
device RSSI -95
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
Reconnect failed
7 left.
Initiating Server Connection
trying to connect to PM
Reusing Client
E NimBLEClient: "<< getRssi(): Not connected"
[... 217 lines cut ...]

23 0 64 0 0 48 0 d0 bf <-- CPMC sent

23 0 64 0 0 49 0 7a c2 <-- CPMC sent

23 0 64 0 0 4a 0 24 c5 <-- CPMC sent

23 0 64 0 0 4b 0 ce c7 <-- CPMC sent

23 0 64 0 0 4c 0 78 ca <-- CPMC sent

23 0 64 0 0 4d 0 22 cd <-- CPMC sent

23 0 64 0 0 4e 0 cc cf <-- CPMC sent

23 0 64 0 0 4f 0 76 d2 <-- CPMC sent

23 0 64 0 0 50 0 20 d5 <-- CPMC sent

23 0 64 0 0 51 0 ca d7 <-- CPMC sent

23 0 64 0 0 52 0 74 da <-- CPMC sent

23 0 64 0 0 53 0 1e dd <-- CPMC sent
Bluetooth Client Disconnected!

doudar, 2021-01-18:
Starting from the bottom:

I see that Ray (dcrainmaker.com) posted this in his review about the powertap pedals:

`Next, on the Bluetooth Smart side, you can pair the pedals with units that support Bluetooth Smart power meters today.  Sorta.  In my case, my firmware on the units (v1.7) that shipped in early June did not have the Bluetooth Smart stack fully enabled. So while devices could pair to it, they weren’t producing power on my firmware version.`

Can you confirm that your pedals have > version 1.7 of the firmware?

I can work on supporting dual pedals, but it may be a little ways off. All of the pedals I've encountered so far (Garmin and Assioma) support single data from the left pedal. Typically there's a setting in the firmware app to select weather or not the data should be sent from the left or individually.

Make sure that the SmartSpin2K is **not** connected to bluetooth in the windows operating system bluetooth menu. If it is, unpair it or forget the device. The windows bluetooth pairing is only for headsets, mice and keyboards.

sspieser, 2021-01-18:
Yes the firmware should be ok, the DCR post is of 2015, now I am using the
last available:

doudar, 2021-01-18:
These lines:
23 0 64 0 0 52 0 74 da <-- CPMC sent

Mean that Zwift or another device is connected to the SS2K and it is sending it's data. 

These lines means that it found your powertap left pedal and tries to connect to it but it cannot find the cycling power service (1818) even though it was advertised. 

Bluetooth Client Found Devices: {"device 4":{"address":"c4:02:da:69:fb:d2","name":"PowerTap.L","UUID":"0x1818"}}
E NimBLEClient: "Error: Connection failed; status=13 "
Forming a connection to: c4:02:da:69:fb:d2
 - Created clientlld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
E NimBLEClient: "Disconnected, could not retrieve services -aborting"

My best guess currently is that your pedals need a firmware update to make them fully bluetooth compatible. Try that and report back with more information. 

Thank you!

doudar, 2021-01-18:
I'll send a message upstream to the awesome bluetooth guy @h2zero and see if he has any ideas about: 

`Created clientlld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)`

sspieser, 2021-01-18:
That's the last firmware available (cf. picture previous mail)... In this
test, I was testing using Zwift on my Android phone : it seems to work a
little better than with my PC...
With Zwift on my phone, I can use the pedals using BLE....

So for now, I stop my testing, and I'll try some the next WE I hope.
Thanks! Have a nice day!
S

doudar, 2021-01-18:
Thanks - good luck, we'll get this sorted out!

h2zero, 2021-01-18:
First, sorry for your troubles, hopefully the solution is simple.

>Bluetooth Client Found Devices: {"device 4":{"address":"c4:02:da:69:fb:d2","name":"PowerTap.L","UUID":"0x1818"}}
E NimBLEClient: "Error: Connection failed; status=13 "
Forming a connection to: c4:02:da:69:fb:d2
    Created client
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)

This is a repeating pattern in the logs and I should explain them. The line with status=13 means the operation timed out, in other words the device did not respond before the supervision timeout setting.
`lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)` is the controller message saying it did not receive the the PDU for connection channel data within the connection interval time, so the connection failed.

My thoughts on this are that the device does not support the connection parameters well and we may simply be able to change them to accommodate it's requirements. Perhaps try:  `pClient->setConnectionParams(80, 80, 0, 100); that will set a 100ms interval and a 1 second supervision timeout so the device can miss 9 sequential  intervals before the connection is terminated.

h2zero, 2021-01-19:
Now that I think about this more, if the device provides preferred connection parameters in it's advertisements I could provide a patch that will use those for testing purposes. However, if possible, give the above settings a try and let me know.

doudar, 2021-01-19:
@h2zero , thanks for the help!

currently it’s:

pClient->setConnectionParams(24, 24, 0, 102);

@sspieser , I’ll put this change into master tonight. I’ll look forward to your test!

h2zero, 2021-01-19:
Just be aware that will slow your data rate for all devices, not sure how sensitive these are to this or how often you care to get your data. Then again for the readings you're doing it's probably a non-issue, just thought I'd mention the potential of creating other issues.

doudar, 2021-01-19:
> Just be aware that will slow your data rate for all devices, not sure how sensitive these are to this or how often you care to get your data. Then again for the readings you're doing it's probably a non-issue, just thought I'd mention the potential of creating other issues.

Yes, you're correct, 1hz update time is adequate, 2hz would be perfect.

h2zero, 2021-01-19:
In that case setting the last parameter to 200 (2 seconds) would be a good option. Gives the device lots of time to get the info to you in a noisy environment.

doudar, 2021-01-19:
The connection parameter change is in Master. I'll test it tomorrow and make sure it's good with all my devices. a quick run through indicates it will be. 

@h2zero , I just noticed that I have pClient->setConnectTimeout(5); with a comment that the default is 30. Is there a chance that the connect timeout is too short? Surely 5 seconds is an eternity........

h2zero, 2021-01-19:
There doesn't seem to be a problem with that, should only be a factor when connecting to devices that aren't actively advertising, i.e might not exist.

sspieser, 2021-01-19:
Hello ! Just did some tries, but... this time I was not able to see
anything on the monitoring or on the debugging console :

Open http://SmartSpin2K.local/
Checking for newer firmware:
  -Server Ver 0.1.1.9
  -Current ver 0.1.1.10
Starting Arduino BLE Client application...
Starting BLE Server
Bluetooth Characteristic defined!
BLE Notify Task Started
Scanning for BLE servers and putting them into a list...
BLE Advertised Device found: Name: , Address: 2b:66:cf:5a:4a:6e,
manufacturer data:
0600010f2002ca47e063a0fb7aa4c16889fc9b8e6e4fc9258811d271dd
BLE Advertised Device found: Name: , Address: cc:6e:a4:ea:49:e9,
manufacturer data: 75004204018060cc6ea4ea49e9ce6ea4ea49e801000000000000
BLE Advertised Device found: Name: , Address: 53:52:12:98:ac:7e,
manufacturer data: e000000eca58b3c3, serviceUUID: 0xfe9f
Service Data:
UUID: 0xfe9f, Data:
BLE Advertised Device found: Name: , Address: 69:d0:ec:aa:b4:9b,
manufacturer data: e000000aca7436e3, serviceUUID: 0xfe9f
Service Data:
UUID: 0xfe9f, Data:
Bluetooth Client Found Devices: null
Scanning
PowerTap.L any
End BLE Setup
HTTP server started
Creating Shifter Interrupts
Served /style.css
Served /bluetoothscanner.html
Served /style.css
Sending BLE device list to HTTP Client
Scanning for BLE Devices
Scanning for BLE servers and putting them into a list...
BLE Advertised Device found: Name: , Address: 2b:66:cf:5a:4a:6e,
manufacturer data:
0600010f2002ca47e063a0fb7aa4c16889fc9b8e6e4fc9258811d271dd
BLE Advertised Device found: Name: , Address: cc:6e:a4:ea:49:e9,
manufacturer data: 75004204018060cc6ea4ea49e9ce6ea4ea49e801000000000000
BLE Advertised Device found: Name: , Address: 53:52:12:98:ac:7e,
manufacturer data: e000000eca58b3c3, serviceUUID: 0xfe9f
Service Data:
UUID: 0xfe9f, Data:
BLE Advertised Device found: Name: , Address: 69:d0:ec:aa:b4:9b,
manufacturer data: e000000aca7436e3, serviceUUID: 0xfe9f
Service Data:
UUID: 0xfe9f, Data:
Bluetooth Client Found Devices: null
Served /bluetoothscanner.html
Served /style.css
Sending BLE device list to HTTP Client
Served /style.css
Served /status.html
Served /style.css

loading
Served /status.html
Served /style.css

sspieser, 2021-01-19:
Sorry, I may have to get to the end and try with Zwift!
So that's what I did :
- start the ESP32
- start pedaling
- start Zwift

Zwift pairs but displays 100 W, see screenshots attached...

Here are the logs :

Connecting to: freebox_BPBXWW
.Connected to freebox_BPBXWW IP address: 192.168.0.36
Open http://SmartSpin2K.local/
Checking for newer firmware:
  -Server Ver 0.1.1.9
  -Current ver 0.1.1.10
Starting Arduino BLE Client application...
Starting BLE Server
Bluetooth Characteristic defined!
BLE Notify Task Started
Scanning for BLE servers and putting them into a list...
BLE Advertised Device found: Name: , Address: cc:6e:a4:ea:49:e9,
manufacturer data: 75004204018060cc6ea4ea49e9ce6ea4ea49e801000000000000
BLE Advertised Device found: Name: , Address: 64:89:5f:a8:d3:eb,
manufacturer data: e000000eca58b3c3, serviceUUID: 0xfe9f
Service Data:
UUID: 0xfe9f, Data:
BLE Advertised Device found: Name: , Address: 02:fd:d1:27:2d:56,
manufacturer data:
0600010f20020b7ebbbaf85803b0c98fc6af1ebcba027afa39f5d784f9
BLE Advertised Device found: Name: , Address: 60:94:6e:ec:f0:4c,
manufacturer data: e000000aca7436e3, serviceUUID: 0xfe9f
Service Data:
UUID: 0xfe9f, Data:
Bluetooth Client Found Devices: null
Scanning
PowerTap.L any
End BLE Setup
HTTP server started
Creating Shifter Interrupts
Served /style.css
Served /bluetoothscanner.html
Served /style.css
Sending BLE device list to HTTP Client
Scanning for BLE Devices
Scanning for BLE servers and putting them into a list...
BLE Advertised Device found: Name: , Address: cc:6e:a4:ea:49:e9,
manufacturer data: 75004204018060cc6ea4ea49e9ce6ea4ea49e801000000000000
BLE Advertised Device found: Name: , Address: 64:89:5f:a8:d3:eb,
manufacturer data: e000000eca58b3c3, serviceUUID: 0xfe9f
Service Data:
UUID: 0xfe9f, Data:
BLE Advertised Device found: Name: , Address: 39:64:b2:60:5a:0e,
manufacturer data:
0600010f20023e83fa9db5e0be82ecc2e9e4fb4051cd6c43c36d932ac0
BLE Advertised Device found: Name: , Address: 60:94:6e:ec:f0:4c,
manufacturer data: e000000aca7436e3, serviceUUID: 0xfe9f
Service Data:
UUID: 0xfe9f, Data:
Bluetooth Client Found Devices: null
Served /bluetoothscanner.html
Served /style.css
Sending BLE device list to HTTP Client
Scanning for BLE Devices
Scanning for BLE servers and putting them into a list...
BLE Advertised Device found: Name: , Address: 39:64:b2:60:5a:0e,
manufacturer data:
0600010f20023e83fa9db5e0be82ecc2e9e4fb4051cd6c43c36d932ac0
BLE Advertised Device found: Name: , Address: 64:89:5f:a8:d3:eb,
manufacturer data: e000000eca58b3c3, serviceUUID: 0xfe9f
[... 101 lines cut ...]
23 0 64 0 0 17 0 46 3d <-- CPMC sent
11 0 0 d7 0 28 33 <-- From APP  Target Incline: 2.15
11 0 0 d7 0 28 33 <-- From APP  Target Incline: 2.15

23 0 64 0 0 18 0 f0 3f <-- CPMC sent

23 0 64 0 0 19 0 9a 42 <-- CPMC sent
11 0 0 d7 0 28 33 <-- From APP  Target Incline: 2.15
11 0 0 d7 0 28 33 <-- From APP  Target Incline: 2.15

23 0 64 0 0 1a 0 44 45 <-- CPMC sent
11 0 0 d7 0 28 33 <-- From APP  Target Incline: 2.15
11 0 0 d7 0 28 33 <-- From APP  Target Incline: 2.15
11 0 0 d7 0 28 33 <-- From APP  Target Incline: 2.15

23 0 64 0 0 1b 0 ee 47 <-- CPMC sent
11 0 0 d7 0 28 33 <-- From APP  Target Incline: 2.15
11 0 0 d7 0 28 33 <-- From APP  Target Incline: 2.15

23 0 64 0 0 1c 0 98 4a <-- CPMC sent

23 0 64 0 0 1d 0 42 4d <-- CPMC sent

23 0 64 0 0 1e 0 ec 4f <-- CPMC sent
Bluetooth Client Disconnected!

doudar, 2021-01-19:
On the first try it didn’t see your power tap on the scan. That probably means the power tap was connected to something else or not on. 

More interestingly, on the second try, it finds your power tap but the power tap isn’t broadcasting its name. Since you have power tap L selected as your power meter, it’s looking for that device name and didn’t connect because it couldn’t see it. 

Try again with powermeter selected as “Any”

The connection to Zwift looks good.

sspieser, 2021-01-19:
Hello again,
Here using "any" : same from the Zwift side (100W) :

loading
23 0 64 0 0 f 0 f6 27 <-- CPMC sent

Served /status.html
Bluetooth Client Disconnected!
Served /style.css
Bluetooth Client Connected! 0
23 0 64 0 0 10 0 a0 2a <-- CPMC sent

23 0 64 0 0 11 0 4a 2d <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 12 0 f4 2f <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 13 0 9e 32 <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 14 0 48 35 <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 15 0 f2 37 <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 16 0 9c 3a <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 17 0 46 3d <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 18 0 f0 3f <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 19 0 9a 42 <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 1a 0 44 45 <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 1b 0 ee 47 <-- CPMC sent

23 0 64 0 0 1c 0 98 4a <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 1d 0 42 4d <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 1e 0 ec 4f <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 1f 0 96 52 <-- CPMC sent

23 0 64 0 0 20 0 40 55 <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 21 0 ea 57 <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 22 0 94 5a <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 23 0 3e 5d <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 24 0 e8 5f <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 25 0 92 62 <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 26 0 3c 65 <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 27 0 e6 67 <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 28 0 90 6a <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 29 0 3a 6d <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 2a 0 e4 6f <-- CPMC sent

23 0 64 0 0 2b 0 8e 72 <-- CPMC sent

23 0 64 0 0 2c 0 38 75 <-- CPMC sent

23 0 64 0 0 2d 0 e2 77 <-- CPMC sent

23 0 64 0 0 2e 0 8c 7a <-- CPMC sent

23 0 64 0 0 2f 0 36 7d <-- CPMC sent

23 0 64 0 0 30 0 e0 7f <-- CPMC sent

23 0 64 0 0 31 0 8a 82 <-- CPMC sent

23 0 64 0 0 32 0 34 85 <-- CPMC sent

23 0 64 0 0 33 0 de 87 <-- CPMC sent

23 0 64 0 0 34 0 88 8a <-- CPMC sent

23 0 64 0 0 35 0 32 8d <-- CPMC sent

23 0 64 0 0 36 0 dc 8f <-- CPMC sent

23 0 64 0 0 37 0 86 92 <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 38 0 30 95 <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 39 0 da 97 <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 3a 0 84 9a <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 3b 0 2e 9d <-- CPMC sent
11 0 0 a9 0 28 33 <-- From APP Target Incline: 1.69

23 0 64 0 0 3c 0 d8 9f <-- CPMC sent

23 0 64 0 0 3d 0 82 a2 <-- CPMC sent

Bluetooth Client Disconnected!

doudar, 2021-01-19:
Thanks for the info and log - I don't see that it tried to do a scan during that log. 

Can you hit scan/reconnect and paste the results here?  

Thanks for all your help!

sspieser, 2021-01-19:
OK I did another try, I hope the Rescan as been caught ?

loading
Served /status.html
Served /style.css
6 left.
Initiating Server Connection
trying to connect to PM
Reusing Client
Client RSSI 0
device RSSI -93
5 left.
Initiating Server Connection
trying to connect to PM
Reusing Client
Client RSSI 0
device RSSI -93
4 left.
Initiating Server Connection
trying to connect to PM
Reusing Client
Client RSSI 0
device RSSI -93
3 left.
Initiating Server Connection
trying to connect to PM
Reusing Client
Client RSSI 0
device RSSI -93
2 left.
Initiating Server Connection
trying to connect to PM
Reusing Client
Client RSSI 0
device RSSI -93
1 left.
Initiating Server Connection
trying to connect to PM
Reusing Client
Client RSSI 0
device RSSI -93
0 left.
Bluetooth Client Connected! 0
23 0 64 0 0 60 0 c0 ff <-- CPMC sent

23 0 64 0 0 61 0 6a 2 <-- CPMC sent

23 0 64 0 0 62 0 14 5 <-- CPMC sent

23 0 64 0 0 63 0 be 7 <-- CPMC sent

23 0 64 0 0 64 0 68 a <-- CPMC sent

23 0 64 0 0 65 0 12 d <-- CPMC sent

23 0 64 0 0 66 0 bc f <-- CPMC sent

23 0 64 0 0 67 0 66 12 <-- CPMC sent

23 0 64 0 0 68 0 10 15 <-- CPMC sent

23 0 64 0 0 69 0 ba 17 <-- CPMC sent

23 0 64 0 0 6a 0 64 1a <-- CPMC sent

23 0 64 0 0 6b 0 e 1d <-- CPMC sent

23 0 64 0 0 6c 0 b8 1f <-- CPMC sent

23 0 64 0 0 6d 0 62 22 <-- CPMC sent

23 0 64 0 0 6e 0 c 25 <-- CPMC sent

23 0 64 0 0 6f 0 b6 27 <-- CPMC sent

23 0 64 0 0 70 0 60 2a <-- CPMC sent

23 0 64 0 0 71 0 a 2d <-- CPMC sent

23 0 64 0 0 72 0 b4 2f <-- CPMC sent

23 0 64 0 0 73 0 5e 32 <-- CPMC sent

23 0 64 0 0 74 0 8 35 <-- CPMC sent

23 0 64 0 0 75 0 b2 37 <-- CPMC sent

23 0 64 0 0 76 0 5c 3a <-- CPMC sent

23 0 64 0 0 77 0 6 3d <-- CPMC sent

23 0 64 0 0 78 0 b0 3f <-- CPMC sent

23 0 64 0 0 79 0 5a 42 <-- CPMC sent

23 0 64 0 0 7a 0 4 45 <-- CPMC sent

23 0 64 0 0 7b 0 ae 47 <-- CPMC sent

23 0 64 0 0 7c 0 58 4a <-- CPMC sent

23 0 64 0 0 7d 0 2 4d <-- CPMC sent

23 0 64 0 0 7e 0 ac 4f <-- CPMC sent

23 0 64 0 0 7f 0 56 52 <-- CPMC sent

23 0 64 0 0 80 0 0 55 <-- CPMC sent

23 0 64 0 0 81 0 aa 57 <-- CPMC sent

23 0 64 0 0 82 0 54 5a <-- CPMC sent

23 0 64 0 0 83 0 fe 5c <-- CPMC sent

23 0 64 0 0 84 0 a8 5f <-- CPMC sent

23 0 64 0 0 85 0 52 62 <-- CPMC sent

23 0 64 0 0 86 0 fc 64 <-- CPMC sent

23 0 64 0 0 87 0 a6 67 <-- CPMC sent

23 0 64 0 0 88 0 50 6a <-- CPMC sent

23 0 64 0 0 89 0 fa 6c <-- CPMC sent

23 0 64 0 0 8a 0 a4 6f <-- CPMC sent

23 0 64 0 0 8b 0 4e 72 <-- CPMC sent

23 0 64 0 0 8c 0 f8 74 <-- CPMC sent

Bluetooth Client Disconnected!
Bluetooth Client Connected! 0
23 0 64 0 0 8d 0 a2 77 <-- CPMC sent

23 0 64 0 0 8e 0 4c 7a <-- CPMC sent

23 0 64 0 0 8f 0 f6 7c <-- CPMC sent

23 0 64 0 0 90 0 a0 7f <-- CPMC sent

23 0 64 0 0 91 0 4a 82 <-- CPMC sent

23 0 64 0 0 92 0 f4 84 <-- CPMC sent

23 0 64 0 0 93 0 9e 87 <-- CPMC sent

23 0 64 0 0 94 0 48 8a <-- CPMC sent

23 0 64 0 0 95 0 f2 8c <-- CPMC sent

23 0 64 0 0 96 0 9c 8f <-- CPMC sent

23 0 64 0 0 97 0 46 92 <-- CPMC sent

23 0 64 0 0 98 0 f0 94 <-- CPMC sent

23 0 64 0 0 99 0 9a 97 <-- CPMC sent

23 0 64 0 0 9a 0 44 9a <-- CPMC sent

Bluetooth Client Disconnected!

sspieser, 2021-01-19:
Another try, this time the ESP32 connected to the PC and VS Code; 1) rescan
2) pedaling 3) Zwift pairing 4) Zwift displays 100W / 50 RPM...

Contents of file: /userPWC.txt
{"session1HR":129,"session1Pwr":100,"session2HR":154,"session2Pwr":150,"hr2Pwr":false}
Writing File: /userPWC.txt
ion
 read:current=
ion
Setting up cpu Tasks
Connecting to: freebox_BPBXWW
.Connected to freebox_BPBXWW IP address: 192.168.0.36
Open http://SmartSpin2K.local/
Checking for newer firmware:
  -Server Ver 0.1.1.9
  -Current ver 0.1.1.10
Starting Arduino BLE Client application...
Starting BLE Server
Bluetooth Characteristic defined!
BLE Notify Task Started
Scanning for BLE servers and putting them into a list...
BLE Advertised Device found: Name: , Address: 10:ac:80:a1:59:e1,
manufacturer data:
0600010f2002e2448df132b41482649aee36e9a804977c350e73c72e9f
BLE Advertised Device found: Name: [TV] Samsung 7 Series (49), Address:
cc:6e:a4:ea:49:e9, manufacturer data:
7500420401016ecc6ea4ea49e9ce6ea4ea49e801000000000000
BLE Advertised Device found: Name: , Address: 62:79:54:50:b9:a0,
manufacturer data: e000000aca7436e3, serviceUUID: 0xfe9f
Service Data:
UUID: 0xfe9f, Data:
BLE Advertised Device found: Name: , Address: 4a:d4:1b:f0:fd:aa,
manufacturer data: e000000eca58b3c3, serviceUUID: 0xfe9f
Service Data:
UUID: 0xfe9f, Data:
Bluetooth Client Found Devices: null
Scanning
any any
End BLE Setup
HTTP server started
Creating Shifter Interrupts
Served /style.css
Served /bluetoothscanner.html
Served /style.css
Sending BLE device list to HTTP Client
Scanning for BLE Devices
Scanning for BLE servers and putting them into a list...
BLE Advertised Device found: Name: , Address: 62:79:54:50:b9:a0,
manufacturer data: e000000aca7436e3, serviceUUID: 0xfe9f
Service Data:
UUID: 0xfe9f, Data:
BLE Advertised Device found: Name: , Address: 10:ac:80:a1:59:e1,
manufacturer data:
0600010f2002e2448df132b41482649aee36e9a804977c350e73c72e9f
BLE Advertised Device found: Name: , Address: 4a:d4:1b:f0:fd:aa,
manufacturer data: e000000eca58b3c3, serviceUUID: 0xfe9f
Service Data:
UUID: 0xfe9f, Data:
Bluetooth Client Found Devices: {"device
4":{"address":"c4:02:da:69:fb:d2","UUID":"0x1818"}}
Served /bluetoothscanner.html
Served /style.css
Sending BLE device list to HTTP Client
Bluetooth Client Connected! 0

23 0 64 0 0 1 0 aa 2 <-- CPMC sent

23 0 64 0 0 2 0 54 5 <-- CPMC sent

23 0 64 0 0 3 0 fe 7 <-- CPMC sent

23 0 64 0 0 4 0 a8 a <-- CPMC sent

23 0 64 0 0 5 0 52 d <-- CPMC sent

23 0 64 0 0 6 0 fc f <-- CPMC sent

23 0 64 0 0 7 0 a6 12 <-- CPMC sent

23 0 64 0 0 8 0 50 15 <-- CPMC sent

23 0 64 0 0 9 0 fa 17 <-- CPMC sent

23 0 64 0 0 a 0 a4 1a <-- CPMC sent

23 0 64 0 0 b 0 4e 1d <-- CPMC sent
11 0 0 d4 0 28 33 <-- From APP  Target Incline: 2.12
11 0 0 d4 0 28 33 <-- From APP  Target Incline: 2.12

23 0 64 0 0 c 0 f8 1f <-- CPMC sent
11 0 0 d4 0 28 33 <-- From APP  Target Incline: 2.12
11 0 0 d4 0 28 33 <-- From APP  Target Incline: 2.12
11 0 0 d4 0 28 33 <-- From APP  Target Incline: 2.12

23 0 64 0 0 d 0 a2 22 <-- CPMC sent
11 0 0 d4 0 28 33 <-- From APP  Target Incline: 2.12

23 0 64 0 0 e 0 4c 25 <-- CPMC sent
11 0 0 d4 0 28 33 <-- From APP  Target Incline: 2.12
11 0 0 d4 0 28 33 <-- From APP  Target Incline: 2.12

23 0 64 0 0 f 0 f6 27 <-- CPMC sent
11 0 0 d4 0 28 33 <-- From APP  Target Incline: 2.12

23 0 64 0 0 10 0 a0 2a <-- CPMC sent
11 0 0 d4 0 28 33 <-- From APP  Target Incline: 2.12
11 0 0 d4 0 28 33 <-- From APP  Target Incline: 2.12

23 0 64 0 0 11 0 4a 2d <-- CPMC sent
11 0 0 d4 0 28 33 <-- From APP  Target Incline: 2.12
11 0 0 d4 0 28 33 <-- From APP  Target Incline: 2.12

23 0 64 0 0 12 0 f4 2f <-- CPMC sent
11 0 0 d4 0 28 33 <-- From APP  Target Incline: 2.12

23 0 64 0 0 13 0 9e 32 <-- CPMC sent

23 0 64 0 0 14 0 48 35 <-- CPMC sent

23 0 64 0 0 15 0 f2 37 <-- CPMC sent
Bluetooth Client Disconnected!

doudar, 2021-01-19:
Thanks for that. I’ll take a look into this tonight. For some reason it isn’t trying to connect to your power meter now. If you have Zwift Companion on any devices nearby, you could try to turn those off and do scan/reconnect again. Sometimes that has caused issues previously.

If you are successful in pairing your pedals, you’ll see a line of numbers in your log that ends with CAD (meaning cadence)

sspieser, 2021-01-19:
STOP ! It's worked !! I don't know what has changed on my side, I have the
feeling I did the same thing... but anyway, it worked !
Note : I get the last commit, but seems to be about de HTTP server...
Logs below ; I will try again later, and of course I still have to connect
the knob, or to provide you with the adjustments to make to the STL file so
I order a new part - it won't work as it is, pictures or video to come...

Connected to freebox_BPBXWW IP address: 192.168.0.36
Open http://SmartSpin2K.local/
Checking for newer firmware:
  -Server Ver 0.1.1.9
  -Current ver 0.1.1.10
Starting Arduino BLE Client application...
Starting BLE Server
Bluetooth Characteristic defined!
BLE Notify Task Started
Scanning for BLE servers and putting them into a list...
BLE Advertised Device found: Name: , Address: 5e:27:fb:b5:7f:f6,
manufacturer data: e000000aca7436e3, serviceUUID: 0xfe9f
Service Data:
UUID: 0xfe9f, Data:
BLE Advertised Device found: Name: , Address: 1d:5a:69:ea:0b:02,
manufacturer data:
0600010f20023d0337e98841a1fe5628300661b15118049f7c0516a70c
BLE Advertised Device found: Name: [TV] Samsung 7 Series (49), Address:
cc:6e:a4:ea:49:e9, manufacturer data:
7500420401016ecc6ea4ea49e9ce6ea4ea49e801000000000000
BLE Advertised Device found: Name: , Address: 6a:64:03:8c:59:74,
manufacturer data: e000000eca58b3c3, serviceUUID: 0xfe9f
Service Data:
UUID: 0xfe9f, Data:
Bluetooth Client Found Devices: null
Scanning
any any
End BLE Setup
HTTP server started
Creating Shifter Interrupts
Served /style.css
Sending BLE device list to HTTP Client
Scanning for BLE Devices
Scanning for BLE servers and putting them into a list...
BLE Advertised Device found: Name: , Address: 1d:5a:69:ea:0b:02,
manufacturer data:
0600010f20023d0337e98841a1fe5628300661b15118049f7c0516a70c
BLE Advertised Device found: Name: [TV] Samsung 7 Series (49), Address:
cc:6e:a4:ea:49:e9, manufacturer data:
7500420401016ecc6ea4ea49e9ce6ea4ea49e801000000000000
BLE Advertised Device found: Name: , Address: 6a:64:03:8c:59:74,
manufacturer data: e000000eca58b3c3, serviceUUID: 0xfe9f
Service Data:
UUID: 0xfe9f, Data:
BLE Advertised Device found: Name: , Address: 5e:27:fb:b5:7f:f6,
manufacturer data: e000000aca7436e3, serviceUUID: 0xfe9f
Service Data:
UUID: 0xfe9f, Data:
BLE Advertised Device found: Name: PowerTap.L, Address: c4:02:da:69:fb:d2,
serviceUUID: 0x1818
Initiating Server Connection
trying to connect to PM
Forming a connection to: c4:02:da:69:fb:d2
 - Created clientBluetooth Client Found Devices: {"device
5":{"address":"c4:02:da:69:fb:d2","name":"PowerTap.L","UUID":"0x1818"}}
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
 - Connected to server
E NimBLEClient: "<< getRssi(): Not connected"
 - RSSI
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
Failed to find service:0x1818
[... 334 lines cut ...]
11 0 0 bc 0 28 33 <-- From APP  Target Incline: 1.88

23 0 13 0 0 1e 0 d 1f <-- CPMC sent
20 0 38 0 8b 42 ef 99 <-- 0x2a63 CAD: 96.00
11 0 0 bc 0 28 33 <-- From APP  Target Incline: 1.88
11 0 0 bc 0 28 33 <-- From APP  Target Incline: 1.88

23 0 38 0 0 1f 0 8d 21 <-- CPMC sent
20 0 1a 0 8d 42 e3 9e <-- 0x2a63 CAD: 96.00

23 0 1a 0 0 20 0 d 24 <-- CPMC sent
11 0 0 bc 0 28 33 <-- From APP  Target Incline: 1.88
20 0 13 0 8e 42 70 a1 <-- 0x2a63 CAD: 94.00
11 0 0 bc 0 28 33 <-- From APP  Target Incline: 1.88
20 0 1b 0 90 42 7e a6 <-- 0x2a63 CAD: 94.00

23 0 1b 0 0 21 0 9a 26 <-- CPMC sent
11 0 0 bc 0 28 33 <-- From APP  Target Incline: 1.88
20 0 17 0 92 42 99 ab <-- 0x2a63 CAD: 94.00
11 0 0 bc 0 28 33 <-- From APP  Target Incline: 1.88

23 0 17 0 0 22 0 27 29 <-- CPMC sent
11 0 0 bc 0 28 33 <-- From APP  Target Incline: 1.88
11 0 0 bc 0 28 33 <-- From APP  Target Incline: 1.88
20 0 15 0 93 42 3c ae <-- 0x2a63 CAD: 91.00

sspieser, 2021-01-19:
Hello again,
Just for info : what is this change for ?

MDNS.addService("http", "_tcp", 80);

Thx,

doudar, 2021-01-19:
The mdns change is to hopefully make the url links work for you so you don’t have to use the IP address.

Glad it worked!

For some reason however, the connection doesn’t look very good to the device and that must be the cause of the issue we are having. Is your esp32 very close to your router or other strong wireless equipment?

doudar, 2021-01-19:
@h2zero , he was able to get his device to connect but it  doesn’t look happy about it. Check the log above. Connect timeout? Strong interference?

sspieser, 2021-01-19:
In fact, the mDNS works well with my laptop ; but does not work - and still
does not with this last commit - with my Android 10 Sony Xperia using
Chrome !?

For the last try with VS Code monitoring, the ESP32 was connected to my
laptop (near from the laptop so), and so 4 meters away from the pedals, and
4 meters away from the WIFI router as well...

fireyeti, 2021-01-19:
@sspieser can you confirm that the pedals are Powertap P1 Power Pedals? 

I only have a few comments.

1. Per the [owners manual](https://www.clevertraining.com/images/product_manuals/PowertapP1UserGuide.pdf), the pedals operate in a Master/Slave configuration with pairing occuring at the factory. The right pedal is the slave, transmitting to the left pedal, which is the master. Therefore your data should be accurate and two sided.
2. As mentioned, the Zwift companion can interfere. There is a setting to disable Bluetooth scanning on iOS. Make sure that is off (if it is indeed an option for android).
3. After the initial configuration of your pedals using the PowerTap mobile app, I recommend "Forget this Device" in your Bluetooth settings. I have Garmin Vector 3S pedals. Initial setup required pairing with my phone to calibrate and set crankarm length. The following attempt to use the pedals with the SmartSpin2k resulted in the pedals pairing to my phone, which made the Bluetooth unavailable to the SmartSpin2k. After i forgot the device I do not have that issue anymore.

Bon chance!

sspieser, 2021-01-19:
Hello,
Yes I have PowerTap P1 power pedals.

1) Yes the power number seems accurate; in fact my last try did show just
half of the power... so maybe the msartspin2k paired to the right one ?
2) I'm not using any Zwift companion so far (I use another old phone for
that purpose and it's off)
3) The Android Sram app does not required pairing ; anyway I checked on my
phone and no Powertap is in the list of know bluetooth devices.

thanks :-)

PS : I did some tests again, but failed differently : no 100W default
anymore, instead 0W / 0RPM .... I stop for today ;-)

Served /status.html
Served /style.css
Bluetooth Client Connected! 0
23 0 0 0 0 2f 0 83 9a <-- CPMC sent

23 0 0 0 0 2f 0 83 9a <-- CPMC sent

23 0 0 0 0 2f 0 83 9a <-- CPMC sent
11 0 0 cf 0 28 33 <-- From APP Target Incline: 2.07

23 0 0 0 0 2f 0 83 9a <-- CPMC sent
11 0 0 cf 0 28 33 <-- From APP Target Incline: 2.07
11 0 0 cf 0 28 33 <-- From APP Target Incline: 2.07

23 0 0 0 0 2f 0 83 9a <-- CPMC sent
11 0 0 cf 0 28 33 <-- From APP Target Incline: 2.07
11 0 0 cf 0 28 33 <-- From APP Target Incline: 2.07

23 0 0 0 0 2f 0 83 9a <-- CPMC sent
11 0 0 cf 0 28 33 <-- From APP Target Incline: 2.07
11 0 0 cf 0 28 33 <-- From APP Target Incline: 2.07

23 0 0 0 0 2f 0 83 9a <-- CPMC sent
11 0 0 cf 0 28 33 <-- From APP Target Incline: 2.07

23 0 0 0 0 2f 0 83 9a <-- CPMC sent
11 0 0 cf 0 28 33 <-- From APP Target Incline: 2.07

23 0 0 0 0 2f 0 83 9a <-- CPMC sent
11 0 0 cf 0 28 33 <-- From APP Target Incline: 2.07
11 0 0 cf 0 28 33 <-- From APP Target Incline: 2.07

23 0 0 0 0 2f 0 83 9a <-- CPMC sent
11 0 0 cf 0 28 33 <-- From APP Target Incline: 2.07
11 0 0 cf 0 28 33 <-- From APP Target Incline: 2.07

23 0 0 0 0 2f 0 83 9a <-- CPMC sent
11 0 0 cf 0 28 33 <-- From APP Target Incline: 2.07
11 0 0 cf 0 28 33 <-- From APP Target Incline: 2.07

23 0 0 0 0 2f 0 83 9a <-- CPMC sent
11 0 0 cf 0 28 33 <-- From APP Target Incline: 2.07

23 0 0 0 0 2f 0 83 9a <-- CPMC sent
11 0 0 cf 0 28 33 <-- From APP Target Incline: 2.07
11 0 0 cf 0 28 33 <-- From APP Target Incline: 2.07

23 0 0 0 0 2f 0 83 9a <-- CPMC sent
11 0 0 cf 0 28 33 <-- From APP Target Incline: 2.07
11 0 0 cf 0 28 33 <-- From APP Target Incline: 2.07

23 0 0 0 0 2f 0 83 9a <-- CPMC sent
11 0 0 cf 0 28 33 <-- From APP Target Incline: 2.07
11 0 0 cf 0 28 33 <-- From APP Target Incline: 2.07

[... 256 lines cut ...]
23 0 0 0 0 2f 0 83 9a <-- CPMC sent

23 0 0 0 0 2f 0 83 9a <-- CPMC sent

23 0 0 0 0 2f 0 83 9a <-- CPMC sent

23 0 0 0 0 2f 0 83 9a <-- CPMC sent

23 0 0 0 0 2f 0 83 9a <-- CPMC sent

23 0 0 0 0 2f 0 83 9a <-- CPMC sent

23 0 0 0 0 2f 0 83 9a <-- CPMC sent

23 0 0 0 0 2f 0 83 9a <-- CPMC sent

23 0 0 0 0 2f 0 83 9a <-- CPMC sent

23 0 0 0 0 2f 0 83 9a <-- CPMC sent

23 0 0 0 0 2f 0 83 9a <-- CPMC sent

23 0 0 0 0 2f 0 83 9a <-- CPMC sent

23 0 0 0 0 2f 0 83 9a <-- CPMC sent

doudar, 2021-01-19:
Thanks for the input @sspieser and @fireyeti 

Most probably to me is that you have a weak signal or interference. I’ll look into how I may be able to make it work better with a bad signal.

As a workaround In the mean time, you should be able to get it to connect by pressing the scan/reconnect button a few times (at least ten seconds apart) until your pedals connect. 

Thanks for your patience!

h2zero, 2021-01-19:
>BLE Advertised Device found: Name: PowerTap.L, Address: c4:02:da:69:fb:d2,
serviceUUID: 0x1818
Initiating Server Connection
trying to connect to PM
Forming a connection to: c4:02:da:69:fb:d2
 - Created clientBluetooth Client Found Devices: {"device
5":{"address":"c4:02:da:69:fb:d2","name":"PowerTap.L","UUID":"0x1818"}}
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
 - Connected to server
E NimBLEClient: "<< getRssi(): Not connected"
 - RSSI
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
Failed to find service:0x1818
disconnecting Client

This log looks strange, makes things hard to follow. https://github.com/doudar/SmartSpin2k/blob/add748782c97e65c2066565d77a4fc08bbcd53ee/src/BLE_Client.cpp#L296

Could use a return value check here so as not to pollute the logs 😄 

As for the trouble connecting, this looks like a normal occurrence and happens for various reasons. Could be noise, could the the esp32 coexist antenna sharing (if you're connected to the webserver at this time, this is expected). I wouldn't be too concerned unless it takes a very long time to actually connect. The connection parameter may still need to be tweaked as well, hard to say without seeing the device personally.

doudar, 2021-01-20:
Yay! @sspieser has confirmed that the recent timing changes have fixed his issues. (most likely increasing the scanning interval and window).

closed 2021-01-20 by doudar