doudar/SmartSpin2k #70: Some pedals output power individually.

doudar, 2021-01-20:
We need to add a "power doubling" option to the configuration as a short term fix for dual pedals that don't combine their data into one common stream. 

In the future we will support dual pedals paired individually however that is a more intensive project. 

Thanks to @sspeiser & @fireyeti for all their hard work in testing this and recent changes!

sspieser, 2021-01-20:
I'm wondering : maybe it depends on which pedal the ESP32 pairs with, is
that possible? I mean, maybe when it pairs to the left one I have both
left + right aggregate, but if it pairs with the other pedal, there's no
aggregation and I just get one side power ? Does that make sense as I had
to choose the option "any" on the dropdown menu...

fireyeti, 2021-01-20:
I dug around online and came across this snippet.

https://powertap.zendesk.com/hc/en-us/articles/360022629794-What-s-the-difference-between-how-ANT-and-Bluetooth-read-data-from-the-P1-Pedals-

"For BLE (Bluetooth Low Energy or Bluetooth Smart) each pedal is paired individually to the display device and it is up to the display to combine the two power values. If a display device does not have the ability to pair to two individual power sensors (most do not) a user will see ½ power. Unless that display device has some built in smarts to double the value when only one pedal based power meter is connected."=

doudar, 2021-01-20:
That's kind of funny we get the data that way. Most manufacturers provide one stream BLE (because there's built in L/R balance in Cycling Power and Cadence BLE spec) but split the data for ANT+ because it's a lower power easier profile. 

Some day we can support ANT+ but the current hardware doesn't support it. We'd have to totally switch hardware or add a daughter board.

doudar, 2021-01-20:
> I'm wondering : maybe it depends on which pedal the ESP32 pairs with, is that possible? I mean, maybe when it pairs to the left one I have both left + right aggregate, but if it pairs with the other [...]

You should be able to try selecting just the left pedal and save. Hopefully I've fixed your pairing problems....

Also, Have the HTTP redirects been fixed or are you still having to enter the IP address?

sspieser, 2021-01-21:
Hello !
I did 5 tries tonight...
The 4 first one, using "PowerTap.L" ; the last one using "PowerTap P1.R".
The first 3 : no chance to display any power number. Then I got power
number, but always divided by 2. Note : last try I see "ERR" for the
cadence on Zwift (looks like the numbers where huge...).
Attached the log files I got.

Have a nice day !

http://github.com/doudar/SmartSpin2k
Main Index

Status
Network SSID

freebox_BPBXWW
Watts

0
MDNS Name

SmartSpin2K
HR

0
Incline Multiplier

2
Cadence

0
Shift Step

400
Incline

216
Connected Heart Monitor

any
Connected Power Meter

PowerTap.L
Debugging Info:

loading 20 0 0 0 16 6e c3 2e <-- 0x2a63 CAD: 0.00
20 0 0 0 16 6e c3 2e <-- 0x2a63 CAD: 0.00

Served /status.html
Served /style.css20 0 0 0 16 6e c3 2e <-- 0x2a63 CAD: 0.00
20 0 0 0 16 6e c3 2e <-- 0x2a63 CAD: 0.00
20 0 0 0 16 6e c3 2e <-- 0x2a63 CAD: 0.00
20 0 0 0 16 6e c3 2e <-- 0x2a63 CAD: 0.00
20 0 0 0 16 6e c3 2e <-- 0x2a63 CAD: 0.00
20 0 0 0 16 6e c3 2e <-- 0x2a63 CAD: 0.00
20 0 0 0 16 6e c3 2e <-- 0x2a63 CAD: 0.00
20 0 0 0 16 6e c3 2e <-- 0x2a63 CAD: 0.00
20 0 0 0 16 6e c3 2e <-- 0x2a63 CAD: 0.00
20 0 0 0 16 6e c3 2e <-- 0x2a63 CAD: 0.00
20 0 0 0 16 6e c3 2e <-- 0x2a63 CAD: 0.00
20 0 0 0 16 6e c3 2e <-- 0x2a63 CAD: 0.00
20 0 0 0 16 6e c3 2e <-- 0x2a63 CAD: 0.00
20 0 0 0 16 6e c3 2e <-- 0x2a63 CAD: 0.00
20 0 48 0 17 6e 21 e9 <-- 0x2a63 CAD: 0.00
20 0 15 0 19 6e b4 ef <-- 0x2a63 CAD: 73.00
20 0 15 0 1a 6e 2f f3 <-- 0x2a63 CAD: 68.00
20 0 10 0 1b 6e c5 f6 <-- 0x2a63 CAD: 66.00
20 0 15 0 1c 6e 68 fa <-- 0x2a63 CAD: 65.00
20 0 15 0 1d 6e 29 fe <-- 0x2a63 CAD: 63.00
[... 1913 lines cut ...]
23 0 0 0 0 29 0 fc 15 <-- CPMC sent
20 0 0 0 69 69 ee 93 <-- 0x2a63 CAD: 14593.00
11 0 0 f7 0 28 33 <-- From APP Target Incline: 2.47
20 0 0 0 f9 6e 83 7c <-- 0x2a63 CAD: 14593.00
20 0 0 0 69 69 ee 93 <-- 0x2a63 CAD: 14593.00

23 0 0 0 0 2a 0 0 16 <-- CPMC sent
20 0 0 0 69 69 ee 93 <-- 0x2a63 CAD: 14593.00

23 0 0 0 0 2b 0 4 16 <-- CPMC sent
20 0 0 0 69 69 ee 93 <-- 0x2a63 CAD: 14593.00

23 0 0 0 0 2c 0 8 16 <-- CPMC sent
20 0 0 0 f9 6e 83 7c <-- 0x2a63 CAD: 14593.00
20 0 0 0 69 69 ee 93 <-- 0x2a63 CAD: 14593.00

23 0 0 0 0 2d 0 c 16 <-- CPMC sent

Bluetooth Client Disconnected! 20 0 0 0 69 69 ee 93 <-- 0x2a63 CAD: 14593.00
20 0 0 0 f9 6e 83 7c <-- 0x2a63 CAD: 14593.00
20 0 0 0 69 69 ee 93 <-- 0x2a63 CAD: 14593.00
20 0 0 0 f9 6e 83 7c <-- 0x2a63 CAD: 14593.00
20 0 0 0 69 69 ee 93 <-- 0x2a63 CAD: 14593.00
20 0 0 0 f9 6e 83 7c <-- 0x2a63 CAD: 14593.00
20 0 0 0 69 69 ee 93 <-- 0x2a63 CAD: 14593.00

doudar, 2021-01-21:
I *think* what may be going on here is when you switch between “any”, “Powertap L”, and “powertap R” (without a reboot) both power meters are getting paired and it’s reading the revolutions & time data from each pedal sequentially. 

I’ll look into it further. 

Thanks for the report!

sspieser, 2021-01-22:
Hello !
I tried to read the code, add some debugging log... but I did not get to
anywhere.

I did some tries again, but nothing really stable ; I have still the left
pedal stored, but either it did not show anything on Zwift, or after a
reboot it showed the SmartSpin2K, but with 0 numbers... and in the log it
looks like it took the right pedal !! how come...

Here are some log, again I can't read those so far  - is anyone else able
to use it, am I the only one with such troubles?

Client RSSI 0
device RSSI -92
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
Detected PM Disconnect. Trying rapid reconnect
Reconnect failed
7 left.
Initiating Server Connection
trying to connect to PM
Reusing Client
E NimBLEClient: "<< getRssi(): Not connected"
Client RSSI 0
device RSSI -92
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
Detected PM Disconnect. Trying rapid reconnect
Reconnect failed
6 left.
Initiating Server Connection
trying to connect to PM
Reusing Client
E NimBLEClient: "<< getRssi(): Not connected"
Client RSSI 0
device RSSI -92
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
Detected PM Disconnect. Trying rapid reconnect
Reconnect failed
5 left.
Initiating Server Connection
trying to connect to PM
Reusing Client
E NimBLEClient: "<< getRssi(): Not connected"
Client RSSI 0
device RSSI -92
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
Detected PM Disconnect. Trying rapid reconnect
Reconnect failed
4 left.
Initiating Server Connection
trying to connect to PM
Reusing Client
E NimBLEClient: "<< getRssi(): Not connected"
Client RSSI 0
device RSSI -92
Reconnected client
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLERemoteService: "characteristicDiscCB() rc=7 "
E NimBLERemoteService: "Could not retrieve characteristics"
ted, could not retrieve services -aborting"cetec
Detected PM Disconnect. Trying rapid reconnect
Initiating Server Connection
trying to connect to PM
Reusing Client
E NimBLEClient: "<< getRssi(): Not connected"
Client RSSI 0
device RSSI -92
[... 947 lines cut ...]

23 0 0 0 0 0 0 0 0 <-- CPMC sent
11 0 0 ce 0 28 33 <-- From APP  Target Incline: 2.06

23 0 0 0 0 0 0 0 0 <-- CPMC sent
11 0 0 ce 0 28 33 <-- From APP  Target Incline: 2.06
11 0 0 ce 0 28 33 <-- From APP  Target Incline: 2.06

23 0 0 0 0 0 0 0 0 <-- CPMC sent
11 0 0 ce 0 28 33 <-- From APP  Target Incline: 2.06
11 0 0 ce 0 28 33 <-- From APP  Target Incline: 2.06

23 0 0 0 0 0 0 0 0 <-- CPMC sent
11 0 0 ce 0 28 33 <-- From APP  Target Incline: 2.06
11 0 0 ce 0 28 33 <-- From APP  Target Incline: 2.06

23 0 0 0 0 0 0 0 0 <-- CPMC sent
11 0 0 ce 0 28 33 <-- From APP  Target Incline: 2.06

23 0 0 0 0 0 0 0 0 <-- CPMC sent

23 0 0 0 0 0 0 0 0 <-- CPMC sent

23 0 0 0 0 0 0 0 0 <-- CPMC sent
Bluetooth Client Disconnected!

doudar, 2021-01-22:
I'm testing a branch right now with this and a bunch of other fixes in it. Hopefully it will roll out later today

doudar, 2021-01-22:
@sspieser 

If you want, try the "https://github.com/doudar/SmartSpin2k/tree/Ignore-Second-Cadence" branch. It should solve most of your problems. 

You will also need to update your filesystem, or optionally only upload the new bluetoothscanner.html https://github.com/doudar/SmartSpin2k/blob/Ignore-Second-Cadence/data/bluetoothscanner.html

doudar, 2021-01-22:
Well, it should fix some of it. We have some signal issues there for sure. 

Is it possible that you have a strong radio transmitter close to your ESP32?

Also, are you able to post a picture of your ESP32 Devkit?

sspieser, 2021-01-22:
I can try... how to get this branch, create a new project, or.... clone ?

[image: image.png]

doudar, 2021-01-22:
@sspieser  , I just merged the branch and pushed an update. 

Give it a spin - a few of the problems you're having should be fixed. Hopefully it helps with your signal issues but I'm not sure I can do much more about it. 

By the way, your pictures aren't coming through for some reason. It only shows an image tag where they should be.

sspieser, 2021-01-22:
Another picture try...

sspieser, 2021-01-22:
[image]
[image]

doudar, 2021-01-22:
Now I can see it :)

That is a little different unit than what we're using. I wonder if one screw may be interfering with the antenna trace? Maybe try removing them and see if that helps your signal.

doudar, 2021-01-22:
Here is what the devkits we're using look like:
[image]

sspieser, 2021-01-22:
Hi again, it worked at my first try ! Great job!! And a second try also worked well (had to reboot the device twice but works fine then) ! Thx for your work!!

PS : OK I may remove the OLED screen... if that does not work better, but the tests I just did are OK so then, maybe I can keep it :-)

But I agree : once I get (hopefully...) everything working, I may get an ESP32 w/o OLED screen screwed on it, has it is not used....

doudar, 2021-01-22:
Great! Good Luck!

doudar, 2021-01-22:
#71 Fixed this

closed 2021-01-22 by doudar