doudar/SmartSpin2k #72: Knob cover for BH-SB3 spinbike issue

sspieser, 2021-01-24:
Hello,
The printed part for the SB3 knob does not fit ; here are some pictures : 
[https://photos.app.goo.gl/GM4VxLWre1n7vYzr9](url)
[https://photos.app.goo.gl/t7VuRxsENP42sbmN9](url)
[https://photos.app.goo.gl/yYytiPBgKsXemeVh9](url)

Do you think, you could change a bit the design, maybe some milimeters wider (inner) and 1 or 2 cm higher?

Thanks for your help!

doudar, 2021-01-24:
Sure - 3 mm wider and 20mm taller? I can do it later this evening.

sspieser, 2021-01-24:
Thanks : 4mm more in the whole diameter I guess, and in that case 10mm
taller (deeper I mean...) should be enough.

Another question : I miss the mounting part, which one do you advice,
Mounting_StrapV2.STL
<https://github.com/doudar/SmartSpin2k/blob/master/Hardware/Mounting_StrapV2.STL>
or
the HeadTubeStrap.stl
<https://github.com/doudar/SmartSpin2k/blob/master/Hardware/HeadTubeStrap.stl>
 ?

Oh, and of course there is no hurry at all :-)

Thx,
S

doudar, 2021-01-24:
Use strap V2. You might need to sand the barbs on it a very little bit so it fits perfectly.

I'll adjust the knob tonight.

doudar, 2021-01-25:
The new file is uploaded. Check out https://github.com/doudar/SmartSpin2k/blob/master/Hardware/Inserts/Inserts%20for%20V1%20Cup/BH%20SB3%20spinbike%20Insert%20v3.stl

doudar, 2021-01-25:
I should add that it's pretty easy to mod this one if you need. The .f3d file is:
https://github.com/doudar/SmartSpin2k/blob/master/Hardware/Inserts/Inserts%20for%20V1%20Cup/BH%20SB3%20spinbike%20%2BInsert%20v3.f3d

sspieser, 2021-01-25:
OK thanks, I'll have it printed again...
Which software do you use to edit that kind of file ?

doudar, 2021-01-25:
Fusion 360 on those parts. It's free for hobbyists. 

https://www.autodesk.com/products/fusion-360/personal

sspieser, 2021-01-30:
Hello !
I have just received the new part, and it will work fine I guess,, thank
you !
Next step : the electronic part...

Good WE,
S

doudar, 2021-02-01:
Great to hear!

Good luck!

sspieser, 2021-02-01:
Sorry, first question : what kind of wires should I use ? I only have very
tight wires from NodeMCU / I2C breadboard, and that may not be enough for a
stepper...

Thx,

doudar, 2021-02-06:
I'm guessing you mean small wires?  It doesn't take much. I think I'm using 22ga? Whatever regular RC servo wires are.

sspieser, 2021-02-06:
Thanks, I have found 24awg 🙂

closed 2021-02-18 by doudar