doudar/SmartSpin2k #73: Minor memory leak

h2zero, 2021-01-27:
Was looking through the code trying to come up with ways to cause the watchdog fault on BLE and noticed that https://github.com/doudar/SmartSpin2k/blob/cbb078a1d2ad8a99b45b071dd7692c5fead65fb0/src/BLE_Client.cpp#L471 could leak memory if called more than once from the webserver ui.

Not sure if it's dealt with elsewhere, just thought I'd give a heads up

doudar, 2021-01-28:
Thanks for the report @h2zero 

I'll look into it. I haven't seen the stack error since I setup the 3 step flag process of initiating a scan. 

I though AdvertisedDeviceCallBack only ran once for the current scan and was dismantled if a new scan (not a continuation) was initiated.

h2zero, 2021-01-28:
When you set the callback, the reference is stored in the NimBLEScan class member variable and is not destroyed unless you delete the scan. So each time your function is called it is creating a new instance of your callback class and  updating NimBLEScan with the new reference while the old one is lost.

It's a minor thing for sure as it's probably not often called and would only be a few bytes, but the possibility is there.

doudar, 2021-02-06:
I've scanned and scanned and scanned while watching free heap and it always seems to recover the memory.

Not saying you're wrong but I can't get it to cause a problem?

h2zero, 2021-02-06:
That's good, I wonder if it gets optimized out by the compiler since there is no member variables the the class? I would only expect a pointers worth of memory to leak at most.

doudar, 2021-03-17:
Fixed within #134

closed 2021-03-17 by doudar