doudar/SmartSpin2k #74: Stepper crossbar mod?

drz01, 2021-01-27:
[image]

Damage sustained during shipping, I suspect. Nevertheless before I order a reprint, any compatible mods to improve tensile strength (for example, increase thickness by 1mm)?

doudar, 2021-01-27:
Unfortunately you're the second one with this issue. I suspect it's from the box getting dropped hard on the side and the leverage of the stepper motor breaking the crossbar. It's cold here and that probably doesn't help either. 

I've since redesigned the cover so it has a rim that fits snugly over the top of the stepper for shipping. 

The crossbar strength is adequate for normal operations however so you could try a repair in whatever method you see fit. (Gosh that's a lot of pieces....)

Let me know if you need help getting local reprints or anything else. 

In my next run of those small parts, I'll try nylon which should help for shipping flexibility also.

drz01, 2021-01-27:
Ok thanks. On closer look it was too shattered, and with pieces missing, for repair. I shall order a rush print with the current spec.

closed 2021-01-27 by drz01

doudar, 2021-01-27:
Ugh....so sorry about that....