doudar/SmartSpin2k #75: New gear ratio available

doudar, 2021-01-29:
When I first started this project, I planned on it being used primarily for magnetic resistance spin bikes. As it turns out, almost everyone that's using SmartSpin2K currently is a friction bike user. 

While SmartSpin2K has enough torque to operate a friction spin bike, it would be good to have just a bit more so I designed some new gears that will fit into the existing gear case with just a minor amount of work. The new gearing increases torque by 27%.

To install the new gears, you'll have to remove the top cover, stepper and electronics, turn the unit upside down, use a quarter inch bolt as a punch and (as gently as possible) drive the old 26T gear out of the knob cup. Then remove the old 13T gear from the stepper motor by (cutting is probably best) or prying it off. I then wrap the old 26T gear in sandpaper and chuck it into a cordless drill to sand the gearbox shroud a tiny bit wider (the new 30T gear fits exactly inside) 

Installing is essentially the reverse, buff the new parts with sandpaper so that the shaft on the 30T almost goes into the knob cup (you still want some resistance so the layer lines lock it into place) and then place the knob cup onto a level surface, 30t gear into the bearings inside the case and press down evenly on the 30T gear to push it into the knob cup. Once that's done, install the new 11T onto the stepper motor. Make sure it's in the correct position so the gear face covers the narrower 30T gear. I use a small socket to press the gear into position. The rest is then just reassembly. 

https://github.com/doudar/SmartSpin2k/blob/master/Hardware/Spur%20gear%201M%2011T.STL
https://github.com/doudar/SmartSpin2k/blob/master/Hardware/Spur%20gear%201M%2030T.STL

fireyeti, 2021-02-01:
Sweet! I will be modifying. I think I will have luck printing at the library.

doudar, 2021-02-01:
Let me know if you need any pointers. The key is driving the old shaft out of the cup with a 1/4" bolt. The rest is pretty easy - sand to fit.

fireyeti, 2021-02-08:
Everything came together great. I used a drum sander attachment on a Dremel to increase the case size. I did have to cut the gear off the stepper.

The gears I had printed at the local library were PLA. Print thickness was thinner than the PETG, so the layers didn't lock quite as well. Hot glue worked well between the knob cup and knob drive shaft to pick up the difference between machines/plastics.

drz01, 2021-02-22:
In case anybody considering SLA over FDM, my reprint of the stepper crossbar came out great! Super smooth and unexpectedly sturdy. It was supposedly a "pro" resin (thought it was going to be Formlabs Grey Pro but no).

doudar, 2021-03-01:
SLA is some pretty cool stuff! 

I bet it works great for the crossbar. 

Thanks for the report!

closed 2021-03-01 by doudar