doudar/SmartSpin2k #77: Manual update page doesn't work if not connected to the internet.

doudar, 2021-02-06:
This is a little frustrating. 

The manual firmware update doesn't work offline (i.e. when connected to the SmartSpin2K SSID) because the page needs jQuery which is downloaded from a different website on the internet. 

I tried a bunch of workarounds this week to get it working but to no avail. I could host the file on the SPIFFS partition but currently that gets wiped on a "reset to defaults". I also tried shoehorning it into the program memory but it's much too big for that (even by splitting it up into 20 parts). 

I'll probably have to completely redesign the manual OTA update page so it doesn't use AJAX and jQuery but that may be a while.   

Another option would be individually downloading the contents of the data folder into SPIFFS during an update instead of replacing the whole SPIFFS but this leaves open the possibility of a bad file corrupting SPIFFS.

doudar, 2021-02-17:
@kadaan Do you have any ideas on this?

kadaan, 2021-02-17:
Haven’t tried manual update yet. Do you just upload a firmware file?

kadaan, 2021-02-17:
@doudar  Oh, what about the jquery dependency on https://github.com/doudar/SmartSpin2k/blob/6a01e1e853d0a06daa309f051c5e6cadbb90e1b1/include/Builtin_Pages.h#L69

if there is no network connection how can it get jquery?

doudar, 2021-02-17:
Yes, that's the issue.

Any ideas how to upload a file (with a progress bar so we know it's not hung) without jQuery?

kadaan, 2021-02-17:
Would something like this work? https://codepen.io/PerfectIsShit/pen/zogMXP

doudar, 2021-02-17:
That looks really promising!

doudar, 2021-04-24:
Fixed by #195

closed 2021-11-25 by doudar