doudar/SmartSpin2k #78: Wiring

sspieser, 2021-02-06:
Hello,
Do you have a wiring diagram ?

I'm wondering : what is the output of the LM2596, 5v or 12V? I guess 5V... but then, what is the TMC2225 :VMOT +12V ? An output or an input ?

Thanks for your help,
S

doudar, 2021-02-06:
For the non-PCB, these should work:
https://github.com/doudar/SmartSpin2k/blob/master/Pictures/SmartSpin2k_Esp32_Connections.jpg
https://github.com/doudar/SmartSpin2k/blob/master/Pictures/SmartSpin2k_TMC_Connections.jpg

Additionally you may find my assembly pictures helpful although I haven't had time to add documentation to them yet:
https://github.com/doudar/SmartSpin2k/tree/master/Pictures/Assembly

sspieser, 2021-02-06:
Thanks; I need two clarifications, one : VMOT 12V ln the TMC2225, where
does that go? If this is input, well, I have a 24V power... If output,
well, I do not see where to go...

Second point, the UART wiring: I need to adf a 1k before plugging both
ESP32 pins at the same TMC pin? Which one then?

Thanks a lot!

doudar, 2021-02-06:
The 1k ohm resistor goes between pin 12 and the uart pin on the tmc. Pin 14 no resistor, connect directly to the uart pin.

You are correct, VMOT is input power and can take up to +24v

sspieser, 2021-02-07:
So, VMOT : you plug it with the output of your SMartspin2K power supply ?

For the UART : whih one one the TMA2225 is the UART pin you wire : if I read the datasheet well, there's 2 UART pin, depending on the mode you use it... probable you wire the upper one ?

Thx again, Seb

doudar, 2021-02-07:
> So, VMOT : you plug it with the output of your SMartspin2K power supply ?
> 
> For the UART : whih one one the TMA2225 is the UART pin you wire : if I read the datasheet well, there's 2 UART pin, depending on the mode you use it... probable you wire the upper one ?
> 
> Thx again, Seb

VMOT is the power you're switching - yes, the voltage coming from your wall power supply. The other comes from your 5v power supply.  

Here's the PCB schematic if it helps:
[image]

The UART pin can differ depending on the configuration of the stepstick. The ones from BigTreeTech are configured as above.

sspieser, 2021-02-07:
Thanks a lot !
So I just began the wiring : I'm trying to power up the ESP32.... Plugged +5V, GND... but I cannot get the Web Server...

doudar, 2021-02-07:
Hard to say exactly. 

Does it work if you're plugged in via USB only?

Does the LED blink indicating it's on? A new version of SmartSpin2K was just released so it may be doing an OTA update (light flashing rapidly) which could take a couple minutes after powerup.

sspieser, 2021-02-07:
It was on with the red light, but not blinking at all... Sure I'l do it
again with USB, and with the PC... thanks, I'll keep you posted :-)

doudar, 2021-02-07:
Good luck! 

Yes, the blue light should start blinking while it's waiting for a client (Bluetooth) connection.

doudar, 2021-02-11:
Any Updates @sspieser ?

sspieser, 2021-02-12:
Hello, yes : it works, I can get to the ESP32 WEB server !
I have found that my ESP32 devkit (now I use this one: [https://www.az-delivery.de/fr/products/esp-32-dev-kit-c-v4](https://www.az-delivery.de/fr/products/esp-32-dev-kit-c-v4)) had a third GND pin... using this one works!

I noticed thus a weird thing : while using the WEB application from my phone, I often lose any WIFI signal on the phone and lose Internet connection... just as if the ESP32 was scrambling things...

But anyway, it works, I paired my pedals ; now I have to keep on wiring but I am not at home for some days; I'll keep you posted and put pictures as soon as I can :-)

sspieser, 2021-02-16:
Hello again, looking at the chart above, I'm wondering : TMC2225(1) pin 5 is linked to the ESP32 D27 pin ; but this link is missing from the diagram on the hme page : 
[image]

Do you confirm pin 5 and D27 must be linked also ?

doudar, 2021-02-16:
The PCB has extra connections on it in case I ever switch to a different stepper driver. 

I believe ms1, ms2 and SLP can all be omitted/unconnected.

sspieser, 2021-02-17:
Thanks.... not sure to understand your last answer ;-)
But I will keep on wiring stuff: +24V for the stepper, and the UART/ 1k stuff ; so far, waiting for wires...

sspieser, 2021-02-19:
I think I'm almost done, just the UART wiring missing ! But then, testing... so far, unable to get in touch with the WEB application...

[image]

doudar, 2021-02-19:
Starting to look good @sspieser ! 

UART is not required for it to work in a basic mode, so you should be able to test it just fine. 

I'm not certain what the issue may be with the webserver. It basically "just works" for me. The only thing I can think of is maybe a bad antenna on your board? 

Do you see the SmartSpin2K access point? If not it means it is successfully connected to your router. If you are able to do manual firmware update https://github.com/doudar/OTAUpdates/blob/main/firmware.bin on the unit and connect it to your local network, it may help me troubleshoot your issue. 

The easiest is to upload it via USB cable with (the brand new) https://github.com/doudar/SmartSpin2K-flasher/releases/download/0.0.1/SmartSpin2Kflasher.exe

closed 2021-02-24 by doudar