doudar/SmartSpin2k #80: Refactoring

MarkusSchneider, 2021-02-07:
Hi doudar,
I've taken a look into the implementation and I've a few suggestions to improve the code base.
- Create a Logger Class debugDirector and put the Telegram-Code into this class with a conditional compiler switch.
-- We can then build a binary with and without Telegram support.
- Move the Stepper Code into a class to handle all the stepper logic
- Splitt BLE and WebServer code into a protocol and use case layer/class e.g. FW Update or the handling of the ERG-Mode will by in the use case layer.
I've you like, I can do a few changes and send you a pull request. 
Please let me know.

Greetings
Markus

doudar, 2021-02-07:
I’d love to see any pushes you’re interested in!

Moving things into classes would be great - I started this with the BLE code but it’s far from finished. 

Let me know where you may have questions as I know there’s a fair bit of spaghetti.

doudar, 2021-02-08:
Markus, 

I just added some compiler definitions that should enable compiling without telegram_token.h 

The default #defines are in settings.h . It will still try (and fail) to send three messages with the default bot token (before it gives up) but it doesn't seem to cause any issue with operation.

closed 2021-02-10 by MarkusSchneider