doudar/SmartSpin2k #83: Tag 1.2.6 do not compile

sspieser, 2021-02-08:
commit be88fd74434ccefacc62cdcf25cf7f02fe25d5fa (HEAD -> master, tag: 1.2.6, origin/master, origin/HEAD)
Author: doudar <anthonydoud97@gmail.com>
Date:   Mon Feb 8 09:01:14 2021 -0600 : 

> Compiling .pio\build\esp32doit\lib1f6\NimBLE-Arduino\NimBLEScan.cpp.o
> src/HTTP_Server_Basic.cpp: In function 'void telegramUpdate(void*)':
> src/HTTP_Server_Basic.cpp:633:34: error: 'TELEGRAM_CHAT_ID' was not declared in this scope
>        bool rm = (bot.sendMessage(TELEGRAM_CHAT_ID, telegramMessage, ""));
>                                   ^
> Compiling .pio\build\esp32doit\lib1f6\NimBLE-Arduino\NimBLESecurity.cpp.o
> *** [.pio\build\esp32doit\src\HTTP_Server_Basic.cpp.o] Error 1
> =========================================================================== [FAILED] Took 14.83 seconds =========================================
> 
>

doudar, 2021-02-08:
I knew I forgot something!

This should now be fixed by be88fd7

Thanks for the report!!!

sspieser, 2021-02-09:
I may have done something wrong, but pull / build, and it's the same :

> Compiling .pio\build\esp32doit\lib1f6\NimBLE-Arduino\NimBLERemoteCharacteristic.cpp.o
> src/HTTP_Server_Basic.cpp: In function 'void telegramUpdate(void*)':
> src/HTTP_Server_Basic.cpp:638:34: error: 'TELEGRAM_CHAT_ID' was not declared in this scope
>        bool rm = (bot.sendMessage(TELEGRAM_CHAT_ID, telegramMessage, ""));
>

doudar, 2021-02-09:
Please try again. I had misnamed the variable in the default template file. 

You can download https://github.com/doudar/SmartSpin2k/blob/master/include/telegram_token.h manually and put it in your local include directory or delete your whole local repository and clone again. It won't update your local copy otherwise because I do not have that file tracked.

sspieser, 2021-02-09:
Thanks, compilation successfull :-)
For now I have not tested anything and still not solved my issue when powering without USB socket...

closed 2021-02-09 by sspieser