doudar/SmartSpin2k #85: Flywheel Branch Bike Needs Testing

doudar, 2021-02-11:
I've quickly added code to enable the Flywheel bike to act as a BLE power device for the SmartSpin2K into the https://github.com/doudar/SmartSpin2k/tree/Flywheel branch but I need some testers as I don't have direct access to the hardware. 

It might work on the first try as the changes needed weren't huge (but who am I kidding...)

You don'tneed to have a fully functioning SmartSpin2K but any old ESP32 devkit will work (such as https://www.amazon.com/MELIFE-Development-Dual-Mode-Microcontroller-Integrated/dp/B07Q576VWZ) 

Just install platformio into visual studio code and upload compile and upload to the ESP32. 

With the bike powered on first, the ESP32 should then convert the proprietary Flywheel BLE information into useable BLE cycling power and cadence BLE GATT spec.  

If you need help, I'll be glad to provide all the pointers you need. 

Mega thanks to @ptx2 for his amazing write up on the Flywheel, as well as his own awesome project over at https://github.com/ptx2/gymnasticon (and who knows, maybe he want's to take a look at this one as well)

ptx2, 2021-02-21:
Hey @doudar love your project! I will try to test this for you sometime next week. The Flywheel code looks right to me. When scanning for the Flywheel bike it might be worth filtering by advertising name as a lot of devices (e.g. Assioma pedals) use that same Nordic UART service uuid. I think all Flywheel bikes have the name “Flywheel 1”.

doudar, 2021-02-21:
Cool - thanks for the pointer! I hadn't thought to see if that uuid was common and just assumed it was proprietary Flywheel. I could test for the name easy enough.

Let me know before you build a binary as I think I've got a few more changes that should go into the branch. I'll put it on my priority list. 

I'm envious of your community over there - your group is really starting to get rolling!

ptx2, 2021-03-22:
Hey @doudar apologies for the delay! I just gave this a try...

I made a couple minor changes to get it to connect:

* Add hyphens to the service/characteristic UUIDs
* Match only if name of device is Flywheel 1 (I have another Nordic UART device within range)
* Also I think power/cadence offsets are swapped and maybe power bytes_to_u16 args need swapping

That got it to connect and start receiving data (!) but I was only seeing the second half of the Flywheel stats packet in the logs. The Flywheel bike sends two packets with no delay in between, e.g.:

```
ff 1f 0c 00 00 00 00 00 00 00 00 00 00 00 00 26 00 00 00 00  # <- power and cadence in here
00 00 00 00 00 00 00 0c 00 00 00 01 38 55  # <- nothing interesting in here (BLECommunications() only logs this part)
```

I think there is a race happening and BLECommunications() doesn't get to see the first half of the Flywheel stats message because it's overwritten so quickly by the second half. If so, one solution could be to pass a `notify_callback` to `NimBLERemoteCharacteristic::subscribe` and only decode the packet beginning with `0xff`?

Here's the full log:

```
Initiating Server Connection
trying to connect to Flywheel Bike
Forming a connection to:  ab:cd:ef:00:11:22
 - Created client - Connected to server
 - Found service:6e400001-b5a3-f393-e0a9-e50e24dcca9e
 - Found Characteristic:6e400003-b5a3-f393-e0a9-e50e24dcca9e
Sucessful 6e400003-b5a3-f393-e0a9-e50e24dcca9e subscription.
These did not match|ab:cd:ef:00:11:22|ab:cd:ef:00:11:22|
We are now connected to the BLE Server.
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
[... 5 lines cut ...]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
[... cut ...]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 01 00 00 00 00 12 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 02 00 00 00 00 21 55 <- 6e400001 | 6e400003 | Null:[ ]
00 00 00 00 00 00 00 03 00 00 00 00 20 55 <- 6e400001 | 6e400003 | Null:[ ]
```

doudar, 2021-03-22:
Cool - thank you very much for your report. 

If you only knew how long it took us to get rid of notify callbacks in the last release 🤣 

The good news is that we could probably add in a very small test during the notify to pass the data. It's a super easy hack but it might take a second to think about how we'd like to do that for code maintainability. 

We got rid of the notify callbacks because it was starting to take radio time and causing crashes as our decoding got bigger during the callback (isr time sensitive). 

Thinking out loud here, maybe we can use a notify callback to park the Characteristic data to be processed in a FIFO buffer. 

I'll mull all this over for a bit and get back to you. @kadaan , any thoughts?

Thanks again @ptx2

kadaan, 2021-03-22:
@ptx2 Can you share a branch with the name filtering, etc.?

kadaan, 2021-03-22:
@doudar Wonder if @h2zero has any ideas other than the NotifyCallback?

doudar, 2021-03-22:
> @doudar Wonder if @h2zero has any ideas other than the NotifyCallback for dealing with data being overwritten.

Possibly we could just decrease BLE_client_delay but that's not without the issues of spamming logging, breaking cadence calculations and pwr->hr.

kadaan, 2021-03-22:
@ptx2 Do the two events always come in pairs from the Flywheel bike?

@doudar Feels like we only care about the most recent X events.  If so, a LIFO queue which is bounded actually be better.

kadaan, 2021-03-22:
Decreasing `BLE_client_delay` might only get us so far.  WTF does flywheel send this???

kadaan, 2021-03-22:
Though, guess would have the same issue with FTMS data if it was large enough to span multiple messages...

doudar, 2021-03-22:
I'd build the buffers inside SpinBLEDevices

h2zero, 2021-03-22:
Not much for me to add to this other than the FIFO buffer queue, mentioned above, on notify callback seems like the best approach here as the data seems to be sent pretty quickly. So notify callback->store data->add to event queue->process queue in another task would be my approach.

ptx2, 2021-03-23:
@doudar No problem!

@kadaan Sure, I just pushed up #145 with those few changes as requested.

closed 2021-11-25 by doudar