doudar/SmartSpin2k #93: Need static IP setup

doudar, 2021-02-18:
We should have a static ip setup for Station mode.

This will require web front end adjustments (are we outgrowing a single settings page and need a network settings page?) , new server.on()’s and new userConfig values . 

If @kadaan wants to work on the front end, I’ll start work on the backend. 

Just shoot me a list of the new variables you think you’ll need.

EdKingscote, 2021-12-22:
Rather than go Static-IP on device, I went for configuring my router to issue the same IP via DHCP all the time. It would be convenient to have the UI output MAC address as part of a network configuration summary/settings screen too.

doudar, 2021-12-22:
@EdKingscote - That's a good idea, thanks!

closed 2022-01-28 by doudar