doudar/SmartSpin2k #98: Reject invalid uploads to SPIFFS via manual updates

doudar, 2021-02-18:
We need to validate manually uploaded files to determine if they are valid for use instead of blindly dumping them into spiffs if they are not named firmware.bin

Additional checks could be:
* Only accept firmware.bin or HTML files. 
* Check the free space in SPIFFS and don't accept anything that is too big. 
* Probably more......

The code to process this starts at:
https://github.com/doudar/SmartSpin2k/blob/663605769f40d7881cd8154dd3dd030436b49af6/src/HTTP_Server_Basic.cpp#L390

MarkusSchneider, 2022-01-01:
With PR #294 the upload dialog has a file mask (*.bin, *.html). Mask should prevent upload of wrong files like firmeware.elf, ...
There is no server side validation of the file or check of free space. 
File mask is an easy fix that should help user to pick the right files.

doudar, 2022-01-01:
That sounds like a great solution. Thanks Markus!

doudar, 2022-01-28:
Fixed & merged.

closed 2022-01-28 by doudar