doudar/SmartSpin2k #100: Fitness Machine Service Branch Needs Testing

doudar, 2021-02-20:
@kadaan , please test this latest binary from the Fitness Machine Service Branch

It might work (might not 😂) with your IC4. 

It won't work with the fitness machine service in general however because I haven't incorporated reading the bitflags yet. (It's hardcoded to the fields that are supposed to work for the IC4.  

[firmware.zip](https://github.com/doudar/SmartSpin2k/files/6013810/firmware.zip)

kadaan, 2021-02-20:
@doudar Interestingly I get this page when connecting to the device after flashing the firmware.  Is the SPIFFS missing int he firmware you uploaded?

[image]

kadaan, 2021-02-20:
@doudar. I was able to work around this by uploading the filesystem image from PlatformIO

doudar, 2021-02-21:
@kadaan  Any updates from the testing session last night after the bit field implementation?

kadaan, 2021-02-23:
FTMS decoding is in this branch and a big refactoring to abstract away different sensor data providers.

doudar, 2021-03-09:
Merged into develop-release-candidate

closed 2021-03-09 by doudar