doudar/SmartSpin2k #102: Peloton app interfacing - Pelomon

OmetKs, 2021-02-21:
I wonder if this could be expanded to listening peloton app, there is some work there, PeloMon that has the esp8266 ble function... Would like to have this function here...

doudar, 2021-02-21:
Is there a way to interface the Peloton bike's power and cadence data while riding via ble through the Peloton app now?

If so, that's huge and probably wouldn't be a big issue to implement. 

If we need to add a serial port to the SS2K, I'm okay with that also, it's just that I don't have access to a Peloton bike for testing, so someone else would have to help with that. (I do have access to a bike+ through a friend). It's an easy change to the hardware/PCB though and I've been contemplating at least breaking out a serial connection when I get a new batch of PCB's made.

OmetKs, 2021-02-21:
My intention is to read resistance value from the Peloton App, via Bluetooth, easy to add to esp8266.

https://github.com/ihaque/pelomon

Don't really know yet if we can send info back to the Peloton app

doudar, 2021-02-21:
It hasn't been tested but the peloton app should be able to get power and cadence from the SS2K if the app supports the BLE CPS or FTMS protocols (which I think it does.)

I see what you're talking about now, getting resistance levels from the peloton app so SS2K can control the knob. 

Thanks for the clarification- I'll look more into it tonight!

kadaan, 2021-02-23:
> My intention is to read resistance value from the Peloton App, via Bluetooth, easy to add to esp8266.
> 
> https://github.com/ihaque/pelomon
> 
> Don't really know yet if we can send info back to the Peloton app

Just to be clear, when you say "Peloton App" do you mean the Android App running on the Peloton Bike Screen, or something like the iOS Peloton App?

OmetKs, 2021-02-24:
> Peloton

I mean the Peloton App for android or iOS, trying to use an IC4 Schwinn bike with the Peloton App in a tablet, hoping to be able to read the resistance dictation with the SS2K to move the knob

kadaan, 2021-02-24:
I would like that is well. Not sure that it can send control signal, but it would be interesting to try to find out.

doudar, 2021-03-19:
I'm pretty sure that the Peloton app doesn't currently provide a wireless way of requesting resistance levels unfortunately. 

If we ever find a way, I'll re-open this issue.

closed 2021-03-19 by doudar