doudar/SmartSpin2k #103: Power field from FTMS as input?

drz01, 2021-02-22:
Finally resumed testing from the weekend; sorry for the long absence. It appears my bike console has undocumented FTMS implementation! It broadcasts itself as a bidirectional controller (which it definitely is not) and therefore not currently captured in SS2K. Both power and cadence can be isolated in Zwift.

I presume the power is being calculated rather than measured, as a function of cadence and resistance; though it appears to be reasonably accurate! I have even decided against a pedal meter for now. I shall try to look for FTMS documentation of the power field, though all this BLE talk is completely over my head...

Surely this cannot be an isolated case? I perceive widespread use of OEM generic parts...

doudar, 2021-02-22:
Awesome news!

FTMS support is something @kadaan and I are currently working on. Hopefully we have a stable binary ready for an OTA update in the next week.

We're probably also going to be implementing a calibration slider on the web page (where you can multiply the power by a %) because a lot of these bikes are pretty inaccurate without. I've got mixed feelings about the slider though because it makes cheating easy.

kadaan, 2021-02-23:
@drz01 What kind of bike?  Can you share the FTMS packet?

drz01, 2021-02-23:
@kadaan A Lifespan SM-400. Does this count?
[Zwift log BLE snippet.txt](https://github.com/doudar/SmartSpin2k/files/6029050/Zwift.log.BLE.snippet.txt)

kadaan, 2021-02-23:
Unfortunately those logs don’t contain the raw info needed to see if this will work. I thought you were trying with the FTMS branch, but sounds like you are not. When you are able to use that code, we will have a better idea what is going on.

doudar, 2021-02-23:
Once we have the new changes merged into main, I can post a binary here for @drz01 to try and hopefully it will "just work". Either way it should report his FTMS packet back through Telegram

drz01, 2021-02-24:
Flashed the last test build anyway *which didn't just work); but did it send a Telegram?

drz01, 2021-02-24:
[ESP32 log snippet.txt](https://github.com/doudar/SmartSpin2k/files/6033413/ESP32.log.snippet.txt)

Flashed the last FTMS build anyway; not quite there, I'll see if can get Telegram to send...

kadaan, 2021-02-24:
If you have an iPhone can you install nRF Connect and see if you can find your bike. If you locate it, you should be able to connect and see the services on the client and determine what BLE service it is exposing.

drz01, 2021-02-24:
[nRF log.txt](https://github.com/doudar/SmartSpin2k/files/6034236/nRF.log.txt)

kadaan, 2021-02-24:
By change was SmartSpin2k on when you ran this nRF scan?  I just want to be sure that the Fitness Machine Services are not SS2K

drz01, 2021-02-24:
The log was not universal, but specific to each manually/exclusively connected device in-app; the hostname "ZV-E31E6" remains constant in every encountered setting -- as was "ss2k", which I switched off at the wall.

Hmm is Telegram being blocked at a TCP port? Need a log for that too...

doudar, 2021-03-01:
Telegram shouldn't be getting blocked, however if you compiled SS2K from your own repository, it won't have the correct telegram secrets required for communication (unless you provided your own telegram bot info).

drz01, 2021-03-03:
This was from the first FTMS branch build; saw error messages on the console. Not getting my own test builds yet -- still battling through bunch of VSC gremlins!

doudar, 2021-03-19:
With the recent success of Echelon support, I'd really like to get more information . 

There have been new methods in NimBLE to request the preferred connection parameters from servers. I wonder if this is a case where that may need to be used as it seems like the connection isn't completing because of a connection parameter (timing) issue. 

You could try this binary (use the SS2K flash tool) to test some timings that are newish from the Echelon branch:
[firmware.zip](https://github.com/doudar/SmartSpin2k/files/6168403/firmware.zip)

drz01, 2021-03-19:
Any [closer?](https://pastebin.com/eBX2eWnW)

I think this was up to date, but I also built with the latest merges just to see.

Currently battling my new FAU with no output -- everything is so hard!

doudar, 2021-03-20:
Any chance you can connect to it with the NFR connect phone app and take pictures of all the services it provides? The best is if you can send us a screen shot of what the data string looks like after you subscribe to notifications (triple down arrow icon) of the "indoor bike data" characteristic - 0x1826.

drz01, 2021-03-20:
Was [this](https://github.com/doudar/SmartSpin2k/issues/103#issuecomment-784872801) not it?

doudar, 2021-03-20:
Ah okay - Yes that will work. I was expecting a screen shot. Thx

doudar, 2021-03-20:
The issue is that it isn't hosting one of the services we're looking for. I'll see if we can plot a workaround

closed 2021-04-24 by doudar