doudar/SmartSpin2k #106: [BUG] mDNS stops responding after multiple minutes

kadaan, 2021-02-23:
### Description

mDNS stops responding after multiple minutes. The web server is still available via IP address. 

### Steps to reproduce

1. Start SS2K
2. Connect to website via mDNS
3. Close website
4. Wait 10 minutes
5. Connect to website via mDNS

**Expected behavior**
mDNS should continue working. 

**Actual behavior**
mDNS stops working.

### Additional Information
None

doudar, 2021-02-23:
Possible reference issue:
https://github.com/espressif/arduino-esp32/issues/4406

kadaan, 2021-02-23:
Yes, I think one thing to try is to have SS2K resolve its own host against mDNS every minute.

doudar, 2021-02-24:
@kadaan i know you were working on this also. Did you pick up on my progress or should I continue my testing in the morning?

kadaan, 2021-02-24:
I'm testing and will continue in the morning

doudar, 2021-12-03:
Fixed

closed 2021-12-03 by doudar