doudar/SmartSpin2k #108: [BUG] Debugging output without FTMS device connected shows SensorData(Null):[ ]

doudar, 2021-02-24:
### Description

With a normal CPS BLE sensor connected, debug shows output like:

23 0 7e 0 0 a4 34 10 2a <-- 0x2a63 SensorData(Null):[ ]
 CAD: 83.00
23 0 7e 0 0 a5 34 f4 2c <-- 0x2a63 SensorData(Null):[ ]

The Sensordata(Null):[] is new and not desired. 

### Steps to reproduce

1. Turn on ESP-32
2. Connect CPS sensor (i.e. another Smartspin2k)
3. look at debugging output.

**Expected behavior**

When no FTMS device is attached, Sensordata(Null):[] should not be in the debug data.

**Actual behavior**

Sensordata(Null):[] is spammed on each BLE Notify Callback. 

### Additional Information

doudar, 2021-02-24:
The best fix would be adding the CPS service to SensorDataFactory 😅

kadaan, 2021-02-24:
I was hope you would add CPS there at some point ;)

doudar, 2021-02-24:
I probably can if you'd like me to work on it. 

I've read your code enough that I mostly understand what's going on and should be able to add in CPS. 

Besides that, it would be good for me. 

Let me know.

doudar, 2021-03-01:
This has been addressed in the upcoming release. Will close once testing is complete.

doudar, 2021-03-17:
Fixed within #134

closed 2021-03-17 by doudar