doudar/SmartSpin2k #114: [BUG] Powermeter dropdown : duplicates...

sspieser, 2021-03-02:
### Description

Dropdown list contains several times the same entries : 

[image]

### Steps to reproduce

1. [First Step]
git pull, build, upload & monitor

2. [Second Step]
http://192.168.0.41/bluetoothscanner.html
Scan devices
> spin bike pedals

3. [and so on...]

**Expected behavior**

**Actual behavior**

### Additional Information
* If you provide debug output and it is long, consider using [Pastebin](https://pastebin.com/) and paste a link in this issue to keep the issue easier to read. 
* Provide pictures or links to videos that clearly demonstrate the issue.

```
Flash: [=======   ]  69.4% (used 1364258 bytes from 1966080 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM9
Uploading .pio\build\esp32doit\firmware.bin
esptool.py v3.0
Serial port COM9
Connecting.....
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 8c:aa:b5:8c:14:30
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 17072 bytes to 11172...
Writing at 0x00001000... (100 %)
Wrote 17072 bytes (11172 compressed) at 0x00001000 in 0.1 seconds (effective 1011.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.0 seconds (effective 1852.9 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 7299.9 kbit/s)...
Hash of data verified.
Compressed 1364368 bytes to 793339...
Writing at 0x00010000... (2 %)
Writing at 0x00014000... (4 %)
Writing at 0x00018000... (6 %)
Writing at 0x0001c000... (8 %)
Writing at 0x00020000... (10 %)
Writing at 0x00024000... (12 %)
Writing at 0x00028000... (14 %)
Writing at 0x0002c000... (16 %)
Writing at 0x00030000... (18 %)
[... 710 lines cut ...]
Initiating Server Connection
Reusing Client
E NimBLEClient: "<< getRssi(): Not connected"
Client RSSI 0
device RSSI -63
E NimBLEClient: "Error: Connection failed; status=13 "
Reconnect failed
2 left.
Initiating Server Connection
Reusing Client
E NimBLEClient: "<< getRssi(): Not connected"
Client RSSI 0
device RSSI -63
E NimBLEClient: "Error: Connection failed; status=13 "
Reconnect failed
1 left.
Initiating Server Connection
Reusing Client
E NimBLEClient: "<< getRssi(): Not connected"
Client RSSI 0
device RSSI -63
E NimBLEClient: "Error: Connection failed; status=13 "
Reconnect failed
0 left.
```

doudar, 2021-03-03:
@sspieser, thanks for the report! 

I believe @kadaan is currently doing some frontend stuff maybe he wants to look into how to fix it?

I've seen this before and I think it may have something to do with multiple scans while the page is open.

kadaan, 2021-03-24:
@sspieser I've gotten pulled away from the UI.  Hope to get back to it soon.

doudar, 2021-12-31:
@sspieser , have you seen this recently with new builds?

closed 2022-09-06 by doudar