doudar/SmartSpin2k #123: [BUG] develop-release-candidate branch: parsePacket(): could not receive data: 9

doudar, 2021-03-08:
### Description

Continuous repeating string of:
`[E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 9`
If another unit with duplicate hostname is on the network. 

### Steps to reproduce

1. Use branch with #119 or #120
2. Program another unit with same PR and same hostname
3. Admire unusable wifi on one unit :)

**Expected behavior**

Error checking to not set similar hostname?

**Actual behavior**

Surprisingly it doesn't crash

### Additional Information
* If you provide debug output and it is long, consider using [Pastebin](https://pastebin.com/) and paste a link in this issue to keep the issue easier to read. 
* Provide pictures or links to videos that clearly demonstrate the issue.

doudar, 2021-03-19:
@kadaan , any updates on this? is it fixed or resolved that you know of?

kadaan, 2021-03-19:
I’ve seen it too. Not sure how to reproduce it.

doudar, 2021-07-02:
This seems to have been fixed.

closed 2021-07-02 by doudar