doudar/SmartSpin2k #129: [FR] Server characteristic faildown

doudar, 2021-03-11:
**Describe the solution you'd like**
There should be a faildown method for determining which services and or characteristics we want to subscribe to from a remote BLE server. I.e.

Is there FTMS Service? - Yes then Connect to that. 
No VV
Is there Cycling Power Service? - Yes, then Connect to that. 
No VV
Is There Cycling Speed and Cadence Service?? Yes, then connect to that. 

Currently, we're just taking the luck of the draw and **Assuming** that the first advertised charUUID is the best one the device broadcasts. 

https://www.youtube.com/watch?v=svkgOsr7pUc

kadaan, 2021-03-19:
This shouldn’t be hard to implement, but it will only have an effect if explicit pairing has not been selected.

doudar, 2021-03-19:
Adding more thoughts, 

My bike trainer separate FTMS, CPS and CSC services. It would be nice to be able to choose exactly what services I want but then use the faildown logic if "any" (or any's future likeness) is selected.

closed 2022-09-06 by doudar