doudar/SmartSpin2k #132: [FR] #define CONFIG_BT_NIMBLE_MAX_CONNECTIONS 6 somewhere other than in Library

doudar, 2021-03-13:
**Is your feature request related to a problem? Please describe.**
@kadaan, I can't figure out how to change a CPPDefine for a library in PlatformIO. 

**Describe the solution you'd like**
Specifically I'd like to #define CONFIG_BT_NIMBLE_MAX_CONNECTIONS 6 for the NimBLE library but if it's done in settings.h the compiler recognizes it after it's already built NimBLE. I've also tried using build flags in platformio.ini with the same effect. 

**Describe alternatives you've considered**
Tracking only .pio/libdeps/esp32doit/NimBLE/src/nimconfig.h but that seems to cause file lock issues on my local computer.
Currently I'm tracking the whole NimBLE library but that's suboptimal for getting upstream changes.  

**Additional context**
Wracked my brain into mush on this one today.

h2zero, 2021-03-31:
Doesn't adding it to the build flags work? i.e.
```
build_flags = !python git_tag_macro.py 
                    -D CONFIG_BT_NIMBLE_MAX_CONNECTIONS=6
```

Seems strange that you would have to import the whole library to make that change in platformio. If that isn't working I'll do some investigating.

kadaan, 2021-03-31:
@h2zero Seemed to work when I did it on Mac, but thought @doudar had troubles on Windows.  

That said, I saw it "working" only be debugging and seeing the correct value set, not in action.

h2zero, 2021-03-31:
I'll check it tomorrow, never seen an issue with this in the past. Possibly just needed the build folder deleted.

doudar, 2021-07-02:
I believe that this was a temporary issue with (my install of PIO).

closed 2021-11-25 by doudar