doudar/SmartSpin2k #149: [FR] Larger range of values for "Shift Step" setting

bau-sec, 2021-03-23:
**Is your feature request related to a problem? Please describe.**
My bike, the Joroto X2, seems to need more turns than other models to achieve the target change of 50W/shift step. I need 2-3 full turns of the knob to move up 50W (80W to 130W) w/ a cadence of 90. Based on the allowed range, I'm guessing other bikes need less than a full turn for this level of resistance.

**Describe the solution you'd like**
It seems like 1 turn requires a shift step of somewhere around 2250. What's the formula for degrees of ration for shift step? For 2 full turns I'd need a shift step of 4500. So ideally I could set the shift step to this value with the slider. Currently I've been just modifying the slider component w/ my browser to allow setting a higher value (the SS2K doesn't validate this value).

**Describe alternatives you've considered**
The alternative is just pressing the shift button a lot, which isn't a show stopper but it's not ideal.

**Additional context**
Possible suggestion for the setting: change this to setting "shift step" in degrees. Reading through the code I can see this value is microsteps on the stepper motor, but that changes depending on stepper motor settings. Most users won't have any context for what this value means, but would understand 360 degrees = one spin of the knob, 180 = half spin, etc. And if they're like me before having the SS2K they were doing erg mode manually during workouts so they probably have an idea of how many watts a turn corresponds to.

Probably a separate issue, but in general the SS2K ESP32 code should be validating the input it receives from the web forms and that they fall in an acceptible range. "shift step" might be harmless, but "stepper power" could be problematic.

kadaan, 2021-03-24:
Interesting idea.  To calibrate SS2K to a bike, the users would need to be able to figure out how much to turn the knob to get a ~50w increase.  On my spin bike there is no visible power output.  Once could use Kinetic app, or something similar, to figure out how many turns (or fraction of a turn) results in the desired wattage increase.

@doudar I'll defer to you on the TMC2225/stepper/gearing to see if this is something that can work.

@bau-sec Totally right on the data validation.  I created #151

kadaan, 2021-03-24:
@doudar Any reason to increase the slider max value as an immediate for for @bau-sec ?

doudar, 2021-03-24:
There's no downside to increasing the range of the slider other than loss of fine adjustability. This could be offset by implementing + - buttons on either side of the sliders. 

The incline multiplier should also be increased as that's essentially how much the computer turns the knob for incline amount. It functions similarly to the "difficulty slider" in Zwift.

The two settings are probably linearly correlated and may also be redundant but I found to error on the side of more adjustability is better atm. 

I've thought a bunch about the units and weather to simplify them somewhat but that's when I decided on the slider method vs just an input box. It also enabled a simple means for client side data validation (server validation is needed). With two separate gear ratios in use, I'm not certain degrees of rotation is appropriate but maybe we could add an approximation table in the wiki or something?

doudar, 2021-03-24:
Also, the stepper power is capped on the hardware side by the potentiometer. We can only obtain actual power values less than the hardware set limit.

doudar, 2021-03-24:
As to steps/revolution, it should be motor steps per revolution*microsteps*gear ratio. 

For the current setup that's:
200*4* 2.7272 = 2,181.76

bau-sec, 2021-03-24:
> The incline multiplier should also be increased as that's essentially how much the computer turns the knob for incline amount. It functions similarly to the "difficulty slider" in Zwift.

This is something I considered as well. I was thinking of making a separate issue depending on discussion here, but it really is the same core issue: num_turns / watt is different for different bikes. Currently in erg mode I have to "help" by shifting up or down on big watt target changes, otherwise it takes 30-60 seconds to move up on its own.

Maybe the settings stay separate; they stay sliders but with higher max values. I'd think the value on the "shift step" setting should be upped to 5000. Not sure on the equivalent max value for the multiplier. To better communicate to users you could have some javascript generated label/text under the sliders that says something like `Rotate X.X times per shift`. For the multiplier: `Rotate X.X times on big Incline/erg adjustments; Y.Y times for small adjustments` Then users could try to get the shift value to be equivalent to the label for big incline/erg adjustments. Then you have something tying the 2 values together while keeping them independent if users want that.

> I'm not certain degrees of rotation is appropriate but maybe we could add an approximation table in the wiki or something?

Yeah, I'm not sure how meaningful/conventional/appropriate degrees as a unit are either. My only other thought there is that degrees are independent of stepper motor settings. In the future there might be some reason where internally it made sense to use a bigger/smaller step, then your setting either has to be scaled or changed to the new number of microsteps. Same issue if you replaced the stepper motor in a future revision.

kadaan, 2021-03-24:
@bau-sec Do you have to "help" the shifting even if you up the shifter step?

bau-sec, 2021-03-24:
> 
> 
> @bau-sec Do you have to "help" the shifting even if you up the shifter step?

I have the *multiplier* set to `5` and the *shift step* set to `4500`. The helping is only during erg mode. Looking at the code I think erg would be fixed just injecting a higher *multiplier* from my browser, maybe a value of 10-15 if it scales the same as the shift step? I'd have to experiment.

doudar, 2021-03-24:
`injecting a higher multiplier from my browser`

Someone's discovered one of my secret testing methods :)

doudar, 2021-03-24:
> I have the _multiplier_ set to `5` and the _shift step_ set to `4500`. The helping is only during erg mode. Looking at the code I >think erg would be fixed just injecting a higher _multiplier_ from [...]

The incline multiplier will help. Ultimately what needs to be done is to make the multiple inside the ERG code user definable. The .5, 1 and 3 in the code below:

`void computeERG(int currentWatts, int setPoint)
{
  //cooldownTimer--;

  float incline = userConfig.getIncline();
  int cad = userConfig.getSimulatedCad();
  int newIncline = incline;
  int amountToChangeIncline = 0;

  if (cad > 20)
  {
    if (abs(currentWatts - setPoint) < 50)
    {
      amountToChangeIncline = (currentWatts - setPoint) * .5;
    }
    if (abs(currentWatts - setPoint) > 50)
    {
      amountToChangeIncline = amountToChangeIncline + ((currentWatts - setPoint)) * 1;
    }
    amountToChangeIncline = amountToChangeIncline / ((currentWatts / 100) + 1);
  }

  if (abs(amountToChangeIncline) > userConfig.getShiftStep() * 3)
  {
    if (amountToChangeIncline > 0)
    {
      amountToChangeIncline = userConfig.getShiftStep() * 3;
    }
    if (amountToChangeIncline < 0)
    {
      amountToChangeIncline = -(userConfig.getShiftStep() * 3);
    }
  }

  newIncline = incline - amountToChangeIncline; //  }
  userConfig.setIncline(newIncline);
}`

 The first block is a 50w "deadband". It could probably be smaller. The second block is a more aggressive band for when the power is outside the deadband. This should/could probably scale somewhat linearly with an absolute cap (as power spikes and power drops do happen).

kadaan, 2021-03-24:
Basically you are trying to have it scale down/up faster if the requested change is greater, right?

doudar, 2021-03-24:
> Basically you are trying to have it scale down/up faster if the requested change is greater, right?

Yes. And the deadband keeps it from oscillation at the set point - one of the things I hate about erg mode on my elite direto trainer.

closed 2021-05-18 by doudar