doudar/SmartSpin2k #151: [FR] Validate config settings

kadaan, 2021-03-24:
**Is your feature request related to a problem? Please describe.**
The UI does validates some config settings before submitting.  However, the web server does not validate config changes to ensure they are within allowed ranges.

**Describe the solution you'd like**
We should be validating all config settings on the server before persisting them.

**Describe alternatives you've considered**
No reasonable alternatives exist other than validating the config vaues.

**Additional context**
@bau-sec  Any context to add?

bau-sec, 2021-03-24:
@kadaan No additional context from me. Validating the input might be a little paranoid when there are endpoints for replacing the entire firmware image, but it's still good practice for security and maintaining consistent state. I imagine that eventually the SPIFFS endpoints and firmware endpoints will be locked down further, but locking them down at this point just makes development more tedious.

doudar, 2021-07-02:
At some point we should also check these in the BLE custom characteristic.

closed 2021-11-25 by doudar