doudar/SmartSpin2k #163: [BUG] Multiple Clients can control

doudar, 2021-03-29:
### Description

After the addition of multiple client connections, multiple clients can now attempt to control a single SS2K. 
### Steps to reproduce

1. Connect more than one Client to the SS2K with different apps running.
2. Watch the SS2K try to control the knob for two completely different requests. 
3. Enjoy!

**Expected behavior**

SS2K should ignore the second client attempt to control incline while still providing HR, Watts and Cadence to all remote connected clients. 

Additionally it would be nice to be able to either select which client gets the control or remove clients that are not actively controlling the resistance/incline/ERG Target. 

**Actual behavior**

SS2K allows multiple controlling clients and acts erratically. 

### Additional Information
[image]

h2zero, 2021-03-31:
Interesting issue here. It seems you would need priority levels for each client connected to determine which one in "in charge" in the current instance. What data would you base the priority level on? This could become a little complex algorithmically, assuming multiple clients are trying to access the same resource, it would make sense that the first successful write is top priority and the next is second etc. but then you need to deal with disconnect/timeout/reconnect and so on. Sounds like a fun puzzle if taken to that level, I'm interested to see what you all come up with.

MarkusSchneider, 2021-12-06:
You can also reject the second BLE connection with a profile who is requesting the incline control. As far as I know, BLE is organized in profiles. If it is possible to reject a connection request to a "controlling" profile after one client with control request is connected, we don't need priorities or must manage clients and there rights.

h2zero, 2021-12-07:
I'm not sure what you mean by profiles? 

When a BLE client connects to a server the server only knows the address and connection parameters of the client. 

The client then requests services from the server, at that time the server could reject the request if the resource is in use.

That's assuming there isn't some special protocol implemented.

doudar, 2021-12-08:
@h2zero , I think that’s what he means.

h2zero, 2021-12-08:
@doudar Yeah, that's what I assumed, the terminology got me though. Sort of sounded like a Bluetooth classic reference.

MarkusSchneider, 2021-12-08:
Sorry for the confusion. With profile I had FTMS in mind: https://www.noble-pro.com/information/what-is-ftms/

closed 2022-09-06 by doudar