doudar/SmartSpin2k #215: [BUG] Project will not build in PlatformIO due to linker errors

urb4nc4rl, 2021-05-16:
### Description

### Steps to reproduce

1. Install PlatformIO extension to Visual Studio Code
2. Pull the latest SS2K code (I used /develop branch)
3. Attempt to build via IDE

**Expected behavior**
Project should build and link

**Actual behavior**
Project fails to build with linker error:

`Linking .pio/build/esp32doit/firmware.elf
/Users/carl/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find crt1-sim.o: No such file or directory
/Users/carl/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find _vectors.o: No such file or directory
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32doit/firmware.elf] Error 1`

Full output from build at [this pastebin](https://pastebin.com/YfHN5sDr)

### Additional Information
I believe this may be an issue with using the wrong Espressif IDF version but am only just getting started with platformIO dev so it's not easy for me to tell.  Would be useful to have better installation steps if possible so we can understand what your build environment is and whether specific versions of IDEs and libraries need to be used.

doudar, 2021-05-17:
@kadaan , are you able to replicate?

kadaan, 2021-05-17:
FYI, I'm on macOS 11.2.3, VSCode 1.57.0-insider.

urb4nc4rl, 2021-05-17:
I'll try to poke around soon and see if I can figure out why it's failing to find these non-existing object files.

Might take me a few days to find some time though, these things can obviously take a while..

timo-r3, 2021-05-17:
Is it same as here? https://github.com/doudar/SmartSpin2k/discussions/201#discussion-3339992
Solution was to remove this line in the platformio.ini:
`platform_packages = framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#master`

kadaan, 2021-05-17:
I have:

1. Uninstalled platformIO
2. Renamed ~/.platformio -> ~/.platformio_BAK
3. Installed VSCode 1.56.2 
4. Installed PlatformIO extension
5. Restarted VSCode
6. Ran `git clone https://github.com/doudar/SmartSpin2k.git SS2K`
7. Opened SS2K folder in VSCode
8. Executed build task in UI
9. Reproduce error

@doudar had intentionally chosen to use master of arduinoespressif32.  He can speak to the reasons.  Ideally, we would pin to a stable version and upgrade intentionally.

Short of that changing:
```
platform_packages = framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#master
```
to
```
platform_packages = framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#master
                    toolchain-xtensa32@~2.80400.0
```
will fix the build.

kadaan, 2021-05-17:
Or pin framework-arduinoespressif32 to the latest 1.x release: 1.0.6:
```
platform_packages = framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#1.0.6
```

urb4nc4rl, 2021-05-17:
Thanks for the quick updates on this.

I can confirm that pinning the framework to version 1.0.6 fixes the issue and seems, to me, to be the correct approach for reproducible builds.

Shall I submit a pull request to fix this?

kadaan, 2021-05-17:
I like it, but @doudar needs to comment as to the desire for master.

urb4nc4rl, 2021-05-17:
Pull request for the config change is [here](https://github.com/doudar/SmartSpin2k/pull/216)

closed 2021-05-18 by doudar