doudar/SmartSpin2k #233: Zwift random motor movement.

doudar, 2021-06-29:
### Discussed in https://github.com/doudar/SmartSpin2k/discussions/223

<div type='discussions-op-text'>

<sup>Originally posted by **d-nx** June  8, 2021</sup>
I just finished my build the other day and for the most part it's been working well.

*SS2K connects to my Echelon Sport and Zwift pairs to SS2k and I get good data from bike to ZWIFT (Watts / Cadence)
*Shifters work and move my motor correctly (UP=more resistance, DOWN=Less resistance)

My only problem is, when riding in Zwift, I'm not getting accurate info. My motor moves very little, and in what feels like very random intervals. Sometimes a little, sometimes a lot. Other times it starts moving even when my avatar is parked and Im not pedaling.

I'm hoping someone can point me in the right direction as to how to solve this.

This is the motor I got:
[Usongshine Nema 17 Stepper Motor Bipolar 38mm Body 42N.cm (60oz.in) 1.5A 4-Lead with 1m Cable and Connector for 3D Printer/Extruder/DIY CNC(17HS4401S)](https://www.amazon.com/gp/product/B07TZ4DPK1/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)

I've also made sure I've adjusted the knob on my TMC2225 to the max position

Any help is greatly appreciated!

PS
I've got some cool octolapses of the prints if you want to use. 

</div>

doudar, 2021-06-29:
Confirmed issue. Cause unknown but other trainers seem to occasionally have the same issue. 

Cause - Zwift will occasionally start sending out random garbage incline settings in the FTMS message. Turning off the windows/mac BLE adaptor, then re-pairing, or switching to using the companion app for connections seems to fix the issue. I've only seen it once in Windows, another user has seen it on Mac. 

Possibly the incline range we are sending Zwift occasionally corrupts it's processing. We will mess around with different supported incline ranges and see if it goes away. Testing is the hard part because duplicating the bug can be difficult.

jkbaker4, 2021-07-01:
I am having the same problem with Sufferfest.  Since the last update random motor movement but also no watts or mph.  RPM seems to be working fine.

Below is a quick shot of my logfile..

N 202534 BLE_Server: "05 52 00 -> ERG Mode Target: 82 Current: 0 Incline: 94.800003"
N 203017 BLE_Common: "f0 d1 09 00 b1 00 00 00 6b 00 00 00 e6 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 203020 BLE_Server: "44 02 00 00 00 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(0.00) PW(0) SD(0.00) ]"
N 203022 BLE_Server: "23 00 00 00 00 69 00 27 d9 -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(0.00) PW(0) ]"
N 203024 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 207059 BLE_Common: "f0 d1 09 00 b5 00 00 00 6b 00 00 00 ea <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 207062 BLE_Server: "44 02 00 00 00 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(0.00) PW(0) SD(0.00) ]"
N 207064 BLE_Server: "23 00 00 00 00 69 00 27 d9 -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(0.00) PW(0) ]"
N 207067 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 207483 BLE_Server: "05 52 00 -> ERG Mode Target: 82 Current: 0 Incline: 94.800003"
N 208068 BLE_Common: "f0 d1 09 00 b6 00 00 00 6b 00 00 00 eb <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 208071 BLE_Server: "44 02 00 00 00 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(0.00) PW(0) SD(0.00) ]"
N 208073 BLE_Server: "23 00 00 00 00 69 00 27 d9 -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(0.00) PW(0) ]"
N 208074 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 210092 BLE_Server: "44 02 00 00 00 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(0.00) PW(0) SD(0.00) ]"
N 210098 BLE_Server: "23 00 00 00 00 69 00 27 d9 -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(0.00) PW(0) ]"
N 210101 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 212524 BLE_Server: "05 52 00 -> ERG Mode Target: 82 Current: 0 Incline: 94.800003"
N 213122 BLE_Common: "f0 d1 09 00 bb 00 00 00 6b 00 00 00 f0 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 213124 BLE_Server: "44 02 00 00 00 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(0.00) PW(0) SD(0.00) ]"
N 213126 BLE_Server: "23 00 00 00 00 69 00 27 d9 -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(0.00) PW(0) ]"
N 213128 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 214130 BLE_Common: "f0 d1 09 00 bc 00 00 00 6b 00 00 00 f1 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 214132 BLE_Server: "44 02 00 00 00 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(0.00) PW(0) SD(0.00) ]"
N 214134 BLE_Server: "23 00 00 00 00 69 00 27 d9 -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(0.00) PW(0) ]"
N 214135 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 215136 BLE_Common: "f0 d1 09 00 bd 00 00 00 6b 00 00 00 f2 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 215141 BLE_Server: "44 02 00 00 00 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(0.00) PW(0) SD(0.00) ]"
N 215142 BLE_Server: "23 00 00 00 00 69 00 27 d9 -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(0.00) PW(0) ]"
N 215144 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 216145 BLE_Common: "f0 d1 09 00 be 00 00 00 6b 00 00 00 f3 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 216147 BLE_Server: "44 02 00 00 00 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(0.00) PW(0) SD(0.00) ]"
N 216149 BLE_Server: "23 00 00 00 00 69 00 27 d9 -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(0.00) PW(0) ]"
N 216152 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 217155 BLE_Common: "f0 d1 09 00 bf 00 00 00 6b 00 00 00 f4 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 217157 BLE_Server: "44 02 00 00 00 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(0.00) PW(0) SD(0.00) ]"
N 217159 BLE_Server: "23 00 00 00 00 69 00 27 d9 -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(0.00) PW(0) ]"
N 217160 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 217593 BLE_Server: "05 52 00 -> ERG Mode Target: 82 Current: 0 Incline: 94.800003"
N 218162 BLE_Common: "f0 d1 09 00 c0 00 00 00 6b 00 00 00 f5 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 218165 BLE_Server: "44 02 00 00 00 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(0.00) PW(0) SD(0.00) ]"
N 218168 BLE_Server: "23 00 00 00 00 69 00 27 d9 -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(0.00) PW(0) ]"
N 218170 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 219172 BLE_Common: "f0 d1 09 00 c1 00 00 00 6b 00 00 00 f6 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 219176 BLE_Server: "44 02 00 00 00 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(0.00) PW(0) SD(0.00) ]"
N 219180 BLE_Server: "23 00 00 00 00 69 00 27 d9 -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(0.00) PW(0) ]"
N 219182 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 220184 BLE_Common: "f0 d1 09 00 c2 00 00 00 6b 00 00 00 f7 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 220186 BLE_Server: "44 02 00 00 00 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(0.00) PW(0) SD(0.00) ]"
N 220188 BLE_Server: "23 00 00 00 00 69 00 27 d9 -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(0.00) PW(0) ]"
N 220190 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 221191 BLE_Common: "f0 d1 09 00 c4 00 00 00 6b 00 00 00 f9 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
[... cut ...]
N 224224 BLE_Server: "23 00 00 00 00 6a 00 bc dc -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(67.00) PW(0) ]"
N 224226 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 225228 BLE_Common: "f0 d1 09 00 c8 00 00 00 6e 00 4a 00 4a <- 0bf669f1 | 0bf669f4 | ECH[ CD(74.00) PW(0) ]"
N 225231 BLE_Server: "44 02 d7 13 94 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(74.00) PW(0) SD(79.00) ]"
N 225233 BLE_Server: "23 00 00 00 00 6b 00 fa df -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(74.00) PW(0) ]"
N 225235 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"

doudar, 2021-07-02:
@jkbaker4  - Your issue is slightly different. Echelon sends out 2 data notifies in rapid succession and because of this we were mostly only receiving the last notification because earlier in development we had to discard using callbacks for processing because of crashes when the callback processing took too long.  

What we resorted to (to solve these crashes) was disregarding the notify and checking the received characteristic data on a predefined interval. This works well for most units but not for the Echelon because of the aforementioned data burst. 

We've had a branch mothballed for a while that addressed this by using the callbacks again but this time just putting the data in a buffer to be processed independent of the callback.  

I've brought this branch up to date with the newest Develop and it could probably be merged if we can get some testing done. Would you be able to flash this firmware.bin (you'll have to unzip it first) and report back your findings?  Thanks! 

[Notify-Buffer_firmware.bin.zip](https://github.com/doudar/SmartSpin2k/files/6755329/Notify-Buffer_firmware.bin.zip)

jkbaker4, 2021-07-02:
Hello,

I updated the firmware and I am still not seeing any wattage calculated.   

Below is a couple of lines from my log file...

N 24256 BLE_Client: "Successful 0bf669f4-45f2-11e7-9598-0800200c9a66 subscription."
N 24257 BLE_Client: "Registered PM on Connect"
N 24506 BLE_Common: "<- 0bf669f1 | 0bf669f4 | ECH[ ]"
N 24806 BLE_Client: "Activated Echelon callbacks."
N 24807 BLE_Client: "We are now connected to the BLE Server."
N 24971 HTTP_Server: "Served /status.html"
N 25210 HTTP_Server: "Served /style.css"
N 25506 BLE_Common: "f0 d2 01 0a cd <- 0bf669f1 | 0bf669f4 | ECH[ ]"
N 26508 BLE_Common: "f0 d1 09 00 00 00 00 00 00 00 00 00 ca <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 27510 BLE_Common: "f0 d1 09 00 01 00 00 00 00 00 00 00 cb <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 28511 BLE_Common: "f0 d1 09 00 02 00 00 00 00 00 00 00 cc <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 29512 BLE_Common: "f0 d1 09 00 03 00 00 00 00 00 00 00 cd <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 30512 BLE_Common: "f0 d1 09 00 04 00 00 00 00 00 00 00 ce <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 31513 BLE_Common: "f0 d1 09 00 05 00 00 00 00 00 00 00 cf <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 32515 BLE_Common: "f0 d1 09 00 06 00 00 00 00 00 00 00 d0 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 33518 BLE_Common: "f0 d1 09 00 07 00 00 00 00 00 00 00 d1 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 34519 BLE_Common: "f0 d1 09 00 08 00 00 00 00 00 00 00 d2 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 35519 BLE_Common: "f0 d1 09 00 09 00 00 00 00 00 00 00 d3 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 36521 BLE_Common: "f0 d1 09 00 0a 00 00 00 00 00 00 00 d4 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 37522 BLE_Common: "f0 d1 09 00 0b 00 00 00 00 00 00 00 d5 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 38523 BLE_Common: "f0 d1 09 00 0c 00 00 00 00 00 00 00 d6 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 39524 BLE_Common: "f0 d1 09 00 0d 00 00 00 00 00 00 00 d7 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 40032 HTTP_Server: "Telegram failed to send! -517073156"
N 40527 BLE_Common: "f0 d1 09 00 0e 00 00 00 00 00 00 00 d8 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 41527 BLE_Common: "f0 d1 09 00 0f 00 00 00 00 00 00 00 d9 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 42527 BLE_Common: "f0 d1 09 00 10 00 00 00 00 00 00 00 da <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 43527 BLE_Common: "f0 d1 09 00 11 00 00 00 00 00 00 00 db <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 44528 BLE_Common: "f0 d1 09 00 12 00 00 00 00 00 00 00 dc <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 45528 BLE_Common: "f0 d1 09 00 13 00 00 00 00 00 00 00 dd <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 46528 BLE_Common: "f0 d1 09 00 14 00 00 00 01 00 26 00 05 <- 0bf669f1 | 0bf669f4 | ECH[ CD(38.00) PW(0) ]"
N 47528 BLE_Common: "f0 d1 09 00 15 00 00 00 01 00 30 00 10 <- 0bf669f1 | 0bf669f4 | ECH[ CD(48.00) PW(0) ]"
N 48424 HTTP_Server: "Served /status.html"
N 48529 BLE_Common: "f0 d1 09 00 16 00 00 00 02 00 2d 00 0f <- 0bf669f1 | 0bf669f4 | ECH[ CD(45.00) PW(0) ]"
N 48736 HTTP_Server: "Served /style.css"
N 54051 BLE_Server: "Bluetooth Remote Client Connected: 4a:58:63:7d:ef:d6 Connected Clients: 1"
N 55645 BLE_Common: "f0 d1 09 00 1e 00 00 00 06 00 27 00 15 <- 0bf669f1 | 0bf669f4 | ECH[ CD(39.00) PW(0) ]"
N 55647 BLE_Server: "44 02 74 0a 4e 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(39.00) PW(0) SD(676.00) ]"
N 55649 BLE_Server: "23 00 00 00 00 02 00 4b 17 -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(39.00) PW(0) ]"
N 55650 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 56652 BLE_Common: "f0 d1 09 00 1f 00 00 00 07 00 2e 00 1e <- 0bf669f1 | 0bf669f4 | ECH[ CD(46.00) PW(0) ]"
N 56655 BLE_Server: "44 02 55 0c 5c 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(46.00) PW(0) SD(157.00) ]"
N 56656 BLE_Server: "23 00 00 00 00 03 00 82 1c -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(46.00) PW(0) ]"
N 56658 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 57245 BLE_Server: "App wrote nothing "
N 57246 BLE_Server: "assuming it's a Control request"
N 57308 BLE_Server: "05 52 00 -> ERG Mode Target: 82 Current: 0 Incline: 0.000000"
N 57660 BLE_Common: "f0 d1 09 00 20 00 00 00 07 00 2e 00 1f <- 0bf669f1 | 0bf669f4 | ECH[ CD(46.00) PW(0) ]"
N 57663 BLE_Server: "44 02 55 0c 5c 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(46.00) PW(0) SD(157.00) ]"
N 57666 BLE_Server: "23 00 00 00 00 04 00 b9 21 -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(46.00) PW(0) ]"
N 57668 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 58655 Main: "Shift DOWN: "
N 58670 BLE_Common: "f0 d1 09 00 21 00 00 00 08 00 2e 00 21 <- 0bf669f1 | 0bf669f4 | ECH[ CD(46.00) PW(0) ]"
N 58673 BLE_Server: "44 02 55 0c 5c 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(46.00) PW(0) SD(157.00) ]"
N 58675 BLE_Server: "23 00 00 00 00 05 00 f0 26 -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(46.00) PW(0) ]"
N 58677 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 59656 Main: "Shift DOWN: "
N 59678 BLE_Common: "f0 d1 09 00 22 00 00 00 09 00 2f 00 24 <- 0bf669f1 | 0bf669f4 | ECH[ CD(47.00) PW(0) ]"
N 59682 BLE_Server: "44 02 9a 0c 5e 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(47.00) PW(0) SD(226.00) ]"
[... cut ...]
N 71795 BLE_Common: "f0 d1 09 00 2e 00 00 00 16 00 3b 00 49 <- 0bf669f1 | 0bf669f4 | ECH[ CD(59.00) PW(0) ]"
N 71799 BLE_Server: "44 02 d1 0f 76 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(59.00) PW(0) SD(49.00) ]"
N 71803 BLE_Server: "23 00 00 00 00 12 00 ff 5b -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(59.00) PW(0) ]"
N 71805 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 71947 BLE_Server: "05 52 00 -> ERG Mode Target: 82 Current: 0 Incline: 12.300000"
N 72664 Main: "Shift DOWN: "

doudar, 2021-11-24:
> Hello,
>
> I updated the firmware and I am still not seeing any wattage calculated.
>
> Below is a couple of lines from my log file...
>
> N 24256 BLE_Client: "Successful 0bf669f4-45f2-11e7-9598-0800200c9a66 [...]

I know it’s been a very long while since you posted this but after having a “power issue “ myself yesterday, I noticed my power correction factor setting on the Bluetooth pairing screen was set at zero. Could this have been your issue?

doudar, 2021-11-30:
I’m fairly certain this has been solved (and previously caused) by the power factor slider at 0. Closing this issue.

closed 2021-11-30 by doudar