doudar/SmartSpin2k #237: Echelon[BUG] (Not receiving resistance regularly)

doudar, 2021-07-14:
### Description

We seem to only get the resistance number from Echelon whenever the knob turns. 

**Expected behavior**

We should be getting regular cadence and resistance updates to calculate wattage.

### Additional Information
We are currently using the calculation from https://github.com/snowzach/echbt but I've noticed there's some sort of polling code in https://github.com/cagnulein/qdomyos-zwift . I'll take a look in his issues and discussion to see if I can figure out how to implement it here, otherwise, maybe @cagnulein can comment here directly on the process?

doudar, 2021-07-15:
@jkbaker4 https://github.com/doudar/SmartSpin2k/issues/233#issuecomment-873255478

I think I found an issue in the sensor code for Echelon. Please try this binary (or build your own) from echelon_fixes branch.

It also has updated connect initialization and watt table from qdoymos_zwift. 

Remember to unzip the firmware and either use the flash tool or upload firmware.bin via http. Any other filename will clog up your spiffs and not get used. 
[firmware.zip](https://github.com/doudar/SmartSpin2k/files/6820174/firmware.zip)

doudar, 2021-07-15:
@marklogan17 , any chance you could also test the binary above using the SS2K connected directly to your Echelon as a power meter?

jkbaker4, 2021-07-15:
Hello,

I am at home today, so can assist with debugging (until 11:45 EST).  What is happening is its connecting then disconnecting and does not register anything on Sufferfest, below is the log..

N 3474 HTTP_Server: "Checking for newer firmware:"
N 5060 HTTP_Server: "  - Server version: 1.6.19"
N 5067 HTTP_Server: "  - Current Version: 1.6.20-echelon_fixes-78-g9f6bbb0"
N 5073 BLE_Setup: "Starting Arduino BLE Client application..."
N 5293 BLE_Server: "Starting BLE Server"
N 5301 BLE_Server: "Bluetooth Characteristic defined!"
N 5301 BLE_Setup: "BLE Notify Task Started"
N 5402 BLE_Setup: "Scanning"
N 5402 BLE_Setup: "any any"
N 5402 BLE_Setup: "End BLE Setup"
N 5412 HTTP_Server: "HTTP server started"
N 5413 Main: "Creating Shifter Interrupts"
N 6294 BLE_Client: "Initiating Scan from Client Task:"
N 6295 BLE_Client: "Scanning for BLE servers and putting them into a list..."
N 6339 BLE_Client: "BLE Advertised Device found: Name: , Address: 65:eb:bc:44:55:52, manufacturer data: 060001092002f25b0b34040f12b3ec593836142f93d71c5dfa3339858d"
N 7047 BLE_Client: "BLE Advertised Device found: Name: , Address: 5e:02:58:f1:83:39, manufacturer data: 4c0010050a18f2d99f, txPower: 12"
N 7092 BLE_Client: "BLE Advertised Device found: Name: , Address: 49:53:c1:81:da:88, manufacturer data: 4c001005471c01fcec, txPower: 12"
N 12400 BLE_Client: "BLE Advertised Device found: Name: , Address: 58:30:5e:48:58:c4, manufacturer data: 060001092002d0b76e1e4a5405179f7153b24eaeb1ed8fb88525254fec"
N 13385 BLE_Client: "BLE Advertised Device found: Name: ECH-SPORT-064725, Address: e5:2e:bc:d1:43:3a, serviceUUID: 0bf669f0-45f2-11e7-9598-0800200c9a66"
N 13387 BLE_Client: "Matching Device Name: ECH-SPORT-064725"
N 13388 BLE_Client: "PM String Matched Any"
N 13388 BLE_Client: "doConnect set on device: 0"
N 14040 BLE_Client: "BLE Advertised Device found: Name: , Address: 5e:02:58:f1:83:39, manufacturer data: 4c0010050a18f2d99f, txPower: 12"
N 16303 BLE_Client: "BLE Advertised Device found: Name: 1ABcn, Address: 90:e2:02:00:1f:9f"
N 16310 BLE_Client: "Bluetooth Client Found Devices: {"device 1":{"address":"e5:2e:bc:d1:43:3a","name":"ECH-SPORT-064725","UUID":"0bf669f0-45f2-11e7-9598-0800200c9a66"}}"
N 17312 BLE_Client: "Initiating Server Connection"
N 17313 BLE_Client: "Trying to connect to Echelon Bike"
N 17314 BLE_Client: "Forming a connection to:  e5:2e:bc:d1:43:3a"
N 17314 BLE_Client: " - Created client"
N 19263 BLE_Client: " - Found service: 0bf669f1-45f2-11e7-9598-0800200c9a66"
N 21264 BLE_Client: " - Found Characteristic: 0bf669f4-45f2-11e7-9598-0800200c9a66"
N 23270 BLE_Common: "f0 d1 09 00 00 00 00 00 00 00 00 00 ca <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 24271 BLE_Common: "f0 d1 09 00 00 00 00 00 00 00 00 00 ca <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 25274 BLE_Common: "f0 d0 01 01 c2 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 25276 BLE_Common: "f0 d2 01 0f d2 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 25277 BLE_Common: "f0 d1 09 00 00 00 00 00 00 00 00 00 ca <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 26278 BLE_Common: "f0 d1 09 00 01 00 00 00 00 00 00 00 cb <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 6171 BLE_Client: "Initiating Scan from Client Task:"
N 6172 BLE_Client: "Scanning for BLE servers and putting them into a list..."
N 6246 BLE_Client: "BLE Advertised Device found: Name: ECH-SPORT-064725, Address: e5:2e:bc:d1:43:3a, serviceUUID: 0bf669f0-45f2-11e7-9598-0800200c9a66"
N 6248 BLE_Client: "Matching Device Name: ECH-SPORT-064725"
N 6248 BLE_Client: "PM String Matched Any"
N 6249 BLE_Client: "doConnect set on device: 0"
N 6439 BLE_Client: "BLE Advertised Device found: Name: , Address: 49:53:c1:81:da:88, manufacturer data: 4c001005471c01fcec, txPower: 12"
N 7377 BLE_Client: "BLE Advertised Device found: Name: , Address: 5e:02:58:f1:83:39, manufacturer data: 4c0010050a18f2d99f, txPower: 12"
N 7989 BLE_Client: "BLE Advertised Device found: Name: , Address: 78:5a:a2:7b:13:c0, manufacturer data: 0600010920020f533883b50b638d2afd2943bdb048964159147d2ad79f"
N 8302 BLE_Client: "BLE Advertised Device found: Name: , Address: 65:eb:bc:44:55:52, manufacturer data: 060001092002f25b0b34040f12b3ec593836142f93d71c5dfa3339858d"
N 12911 BLE_Client: "BLE Advertised Device found: Name: ECH-SPORT-064725, Address: e5:2e:bc:d1:43:3a, serviceUUID: 0bf669f0-45f2-11e7-9598-0800200c9a66"
N 12913 BLE_Client: "Matching Device Name: ECH-SPORT-064725"
N 12914 BLE_Client: "PM String Matched Any"
N 12914 BLE_Client: "doConnect set on device: 0"
N 13125 BLE_Client: "BLE Advertised Device found: Name: , Address: 65:eb:bc:44:55:52, manufacturer data: 060001092002f25b0b34040f12b3ec593836142f93d71c5dfa3339858d"
N 13230 BLE_Client: "BLE Advertised Device found: Name: , Address: 49:53:c1:81:da:88, manufacturer data: 4c001005471c01fcec, txPower: 12"
N 15270 BLE_Client: "BLE Advertised Device found: Name: , Address: 5e:02:58:f1:83:39, manufacturer data: 4c0010050a18f2d99f, txPower: 12"
N 16180 BLE_Client: "BLE Advertised Device found: Name: 1ABcn, Address: 90:e2:02:00:1f:9f"
N 16183 BLE_Client: "Bluetooth Client Found Devices: null"
N 17184 BLE_Client: "Initiating Server Connection"
E 17185 BLE_Client: "Device has no Service UUID"
N 17185 BLE_Client: "Initiating Scan from Client Task:"
N 17185 BLE_Client: "Scanning for BLE servers and putting them into a list..."
N 17355 BLE_Client: "BLE Advertised Device found: Name: ECH-SPORT-064725, Address: e5:2e:bc:d1:43:3a, serviceUUID: 0bf669f0-45f2-11e7-9598-0800200c9a66"
[... cut ...]
N 36129 BLE_Common: "f0 d1 09 00 00 00 00 00 00 00 00 00 ca <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 36135 BLE_Common: "f0 d0 01 01 c2 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 36138 BLE_Common: "f0 d2 01 0f d2 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 38146 BLE_Common: "f0 d1 09 00 01 00 00 00 00 00 00 00 cb <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 6669 BLE_Client: "BLE Advertised Device found: Name: , Address: 6c:80:8b:ae:4c:85, manufacturer data: 4c0010062f1ed1e31b84, txPower: 8"

doudar, 2021-07-15:
Okay, tweaked some timings and found a size bug. Please give this a try. btw, thanks for your testing help!!
[firmware.zip](https://github.com/doudar/SmartSpin2k/files/6823527/firmware.zip)

jkbaker4, 2021-07-15:
Dumb Question,

How do I update using the web interface.  Asks for a userid and password.  No idea what that is…

Thank You
John Baker

doudar, 2021-07-15:
admin/admin

jkbaker4, 2021-07-15:
Hello,

Same results…Connects then disconnects not registering CD or PW

N 89473 BLE_Common: "f0 d1 09 00 00 00 00 00 00 00 00 00 ca <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 89476 BLE_Common: "f0 d2 01 14 d7 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 90478 BLE_Common: "f0 d2 01 14 d7 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 90480 BLE_Common: "f0 d2 01 15 d8 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 90481 BLE_Common: "f0 d1 09 00 00 00 00 00 00 00 00 00 ca <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 90483 BLE_Common: "f0 d2 01 14 d7 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
...
N 3465 HTTP_Server: "Checking for newer firmware:"
N 4886 HTTP_Server: "  - Server version: 1.6.19"
N 4892 HTTP_Server: "  - Current Version: 1.6.20-echelon_fixes-79-g64db89f"
N 4900 BLE_Setup: "Starting Arduino BLE Client application..."
N 5113 BLE_Server: "Starting BLE Server"
N 5121 BLE_Server: "Bluetooth Characteristic defined!"
N 5122 BLE_Setup: "BLE Notify Task Started"
N 5223 BLE_Setup: "Scanning"
N 5223 BLE_Setup: "any any"
N 5223 BLE_Setup: "End BLE Setup"
N 5234 HTTP_Server: "HTTP server started"
N 5236 Main: "Creating Shifter Interrupts"
N 6114 BLE_Client: "Initiating Scan from Client Task:"
N 6115 BLE_Client: "Scanning for BLE servers and putting them into a list..."
N 6210 BLE_Client: "BLE Advertised Device found: Name: ECH-SPORT-064725, Address: e5:2e:bc:d1:43:3a, serviceUUID: 0bf669f0-45f2-11e7-9598-0800200c9a66"
N 6212 BLE_Client: "Matching Device Name: ECH-SPORT-064725"
N 6212 BLE_Client: "PM String Matched Any"
N 6213 BLE_Client: "doConnect set on device: 0"
N 6803 BLE_Client: "BLE Advertised Device found: Name: , Address: 4d:e9:f5:40:8f:de, manufacturer data: 4c001005491cfc18f1, txPower: 12"
N 7074 Main: "Shift DOWN: "
N 7783 BLE_Client: "BLE Advertised Device found: Name: , Address: 7b:11:6a:5b:76:ea, manufacturer data: 4c0010050f18b86a67, txPower: 12"
N 7858 BLE_Client: "BLE Advertised Device found: Name: , Address: 57:3d:d9:5f:1d:0a, manufacturer data: 06000109200288de452f6b5c1d39850a5272c574c21f915fd8863e44d8"
N 7885 BLE_Client: "BLE Advertised Device found: Name: , Address: 42:6f:03:96:4b:82, manufacturer data: 0600010920022cd6b17c008e657fcc5dd3c1ad5d7fe237a7193c3b518a"
N 9037 BLE_Client: "BLE Advertised Device found: Name: , Address: 5d:ff:02:51:00:92, manufacturer data: 4c001006291e4fa9605d, txPower: 8"
N 10099 BLE_Client: "BLE Advertised Device found: Name: =<-njpC14M;*cQ|EdcN 12985 BLE_Client: "BLE Advertised Device found: Name: , Address: 57:3d:d9:5f:1d:0a, manufacturer data: 06000109200288de452f6b5c1d39850a5272c574c21f915fd8863e44d8"
N 12997 BLE_Client: "BLE Advertised Device found: Name: ECH-SPORT-064725, Address: e5:2e:bc:d1:43:3a, serviceUUID: 0bf669f0-45f2-11e7-9598-0800200c9a66"
N 12998 BLE_Client: "Matching Device Name: ECH-SPORT-064725"
N 12999 BLE_Client: "PM String Matched Any"
N 13000 BLE_Client: "doConnect set on device: 0"
N 13052 BLE_Client: "BLE Advertised Device found: Name: , Address: 4d:e9:f5:40:8f:de, manufacturer data: 4c001005491cfc18f1, txPower: 12"
N 13127 BLE_Client: "BLE Advertised Device found: Name: , Address: 42:6f:03:96:4b:82, manufacturer data: 0600010920022cd6b17c008e657fcc5dd3c1ad5d7fe237a7193c3b518a"
N 13164 BLE_Client: "BLE Advertised Device found: Name: , Address: 5d:ff:02:51:00:92, manufacturer data: 4c001006291e4fa9605d, txPower: 8"
N 13760 BLE_Client: "BLE Advertised Device found: Name: , Address: 7b:11:6a:5b:76:ea, manufacturer data: 4c0010050f18b86a67, txPower: 12"
N 15138 BLE_Client: "BLE Advertised Device found: Name: =<-njpC14M;*cQ|EdcN 16126 BLE_Client: "Bluetooth Client Found Devices: {"device 0":{"address":"e5:2e:bc:d1:43:3a","name":"ECH-SPORT-064725","UUID":"0bf669f0-45f2-11e7-9598-0800200c9a66"}}"
N 16802 Main: "Shift UP: "
N 17129 BLE_Client: "Initiating Server Connection"
N 17130 BLE_Client: "Trying to connect to Echelon Bike"
N 17130 BLE_Client: "Forming a connection to:  e5:2e:bc:d1:43:3a"
N 17131 BLE_Client: " - Created client"
N 17333 BLE_Client: " - Connected to server"
N 17333 BLE_Client: " - RSSI -112"
N 17839 Main: "Shift UP: "
N 18829 BLE_Client: " - Found service: 0bf669f1-45f2-11e7-9598-0800200c9a66"
N 19113 Main: "Shift UP: "
N 20552 Main: "Shift UP: "
N 21080 BLE_Client: " - Found Characteristic: 0bf669f4-45f2-11e7-9598-0800200c9a66"
N 22081 BLE_Client: "Successful 0bf669f4-45f2-11e7-9598-0800200c9a66 subscription."
N 22083 BLE_Client: "Registered PM on Connect"
N 23029 Main: "Shift DOWN: "
N 23090 BLE_Common: "f0 d1 09 00 00 00 00 00 00 00 00 00 ca <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 24092 BLE_Common: "f0 d1 09 00 00 00 00 00 00 00 00 00 ca <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 24191 BLE_Client: "Activated Echelon callbacks."
N 24192 BLE_Client: "We are now connected to the BLE Server."
N 25092 BLE_Common: "f0 d2 01 13 d6 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 25094 BLE_Common: "f0 d2 01 12 d5 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 25095 BLE_Common: "f0 d1 09 00 00 00 00 00 00 00 00 00 ca <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 26099 BLE_Common: "f0 d2 01 13 d6 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 26102 BLE_Common: "f0 d2 01 12 d5 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 26105 BLE_Common: "f0 d1 09 00 00 00 00 00 00 00 00 00 ca <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 26108 BLE_Common: "f0 d2 01 13 d6 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
[... cut ...]
N 30138 BLE_Common: "f0 d1 09 00 00 00 00 00 00 00 00 00 ca <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 31139 BLE_Common: "f0 d2 01 13 d6 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 31141 BLE_Common: "f0 d2 01 12 d5 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 31142 BLE_Common: "f0 d1 09 00 00 00 00 00 00 00 00 00 ca <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 31144 BLE_Common: "f0 d2 01 13 d6 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 32145 BLE_Common: "f0 d2 01 12 d5 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"

Thank You
John Baker

doudar, 2021-07-15:
So it looks like it caught it on the second go. and didn't try to rescan. I'm not seeing any cadence information coming through on the d1 lines. Are you pedaling? It should be here:

BLE_Common: "f0 d1 09 00 00 00 00 00 00 00 00 00 ca
                                                                           ^

jkbaker4, 2021-07-15:
Hello,

Yup I am pedaling no idea why its not registering

Thank You
John Baker

jkbaker4, 2021-07-15:
Hello,

Below is the latest…While 100% sure I was pedaling..

N 7384 BLE_Client: "BLE Advertised Device found: Name: , Address: 7a:49:a5:b3:57:f1, manufacturer data: 4c001006251eed37707f, txPower: 8"
N 7557 BLE_Client: "BLE Advertised Device found: Name: , Address: 51:77:27:72:85:e4, manufacturer data: 060001092002be6a683dba32f0c506befd507b5453263f79bcecdf518e"
N 14110 BLE_Client: "BLE Advertised Device found: Name: ECH-SPORT-064725, Address: e5:2e:bc:d1:43:3a, serviceUUID: 0bf669f0-45f2-11e7-9598-0800200c9a66"
N 14112 BLE_Client: "Matching Device Name: ECH-SPORT-064725"
N 14112 BLE_Client: "PM String Matched Any"
N 14113 BLE_Client: "doConnect set on device: 0"
N 14129 BLE_Client: "BLE Advertised Device found: Name: , Address: 49:1c:cd:76:60:93, manufacturer data: 0600010920026478b91d43d0b9a44cb5b222ad23e5e50caaf9b0f76cfe"
N 15046 BLE_Client: "BLE Advertised Device found: Name: , Address: 51:77:27:72:85:e4, manufacturer data: 060001092002be6a683dba32f0c506befd507b5453263f79bcecdf518e"
N 17109 BLE_Client: "BLE Advertised Device found: Name: 1ABcn, Address: 90:e2:02:00:1f:9f"
N 17112 BLE_Client: "Bluetooth Client Found Devices: null"
N 18113 BLE_Client: "Initiating Server Connection"
E 18114 BLE_Client: "Device has no Service UUID"
N 18114 BLE_Client: "Initiating Scan from Client Task:"
N 18114 BLE_Client: "Scanning for BLE servers and putting them into a list..."
N 18140 BLE_Client: "BLE Advertised Device found: Name: , Address: 7a:49:a5:b3:57:f1, manufacturer data: 4c001006251eed37707f, txPower: 8"
N 18215 BLE_Client: "BLE Advertised Device found: Name: ECH-SPORT-064725, Address: e5:2e:bc:d1:43:3a, serviceUUID: 0bf669f0-45f2-11e7-9598-0800200c9a66"
N 18216 BLE_Client: "Matching Device Name: ECH-SPORT-064725"
N 18217 BLE_Client: "PM String Matched Any"
N 18218 BLE_Client: "doConnect set on device: 0"
N 18266 BLE_Client: "BLE Advertised Device found: Name: , Address: 7b:11:6a:5b:76:ea, manufacturer data: 4c0010050f18b86a67, txPower: 12"
N 18336 BLE_Client: "BLE Advertised Device found: Name: , Address: 49:1c:cd:76:60:93, manufacturer data: 0600010920026478b91d43d0b9a44cb5b222ad23e5e50caaf9b0f76cfe"
N 22074 BLE_Client: "BLE Advertised Device found: Name: ECH-SPORT-064725, Address: e5:2e:bc:d1:43:3a, serviceUUID: 0bf669f0-45f2-11e7-9598-0800200c9a66"
N 22076 BLE_Client: "Matching Device Name: ECH-SPORT-064725"
N 22077 BLE_Client: "PM String Matched Any"
N 22077 BLE_Client: "doConnect set on device: 0"
N 22110 BLE_Client: "BLE Advertised Device found: Name: , Address: 49:1c:cd:76:60:93, manufacturer data: 0600010920026478b91d43d0b9a44cb5b222ad23e5e50caaf9b0f76cfe"
N 22121 BLE_Client: "BLE Advertised Device found: Name: , Address: 46:f3:35:c3:b3:50, manufacturer data: 4c000c0e00981ce3a641add8ca7c781ce9b61005491c08e64b"
N 22123 BLE_Client: "BLE Advertised Device found: Name: , Address: 51:77:27:72:85:e4, manufacturer data: 060001092002be6a683dba32f0c506befd507b5453263f79bcecdf518e"
N 29128 BLE_Client: "Initiating Server Connection"
N 29129 BLE_Client: "Trying to connect to Echelon Bike"
N 29130 BLE_Client: "Forming a connection to:  e5:2e:bc:d1:43:3a"
N 29130 BLE_Client: " - Created client"
N 31307 BLE_Client: " - Found service: 0bf669f1-45f2-11e7-9598-0800200c9a66"
N 33307 BLE_Client: " - Found Characteristic: 0bf669f4-45f2-11e7-9598-0800200c9a66"
N 36045 BLE_Common: "f0 d1 09 00 00 00 00 00 00 00 00 00 ca <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 36226 BLE_Client: "Activated Echelon callbacks."
N 36227 BLE_Client: "We are now connected to the BLE Server."
N 37047 BLE_Common: "f0 d0 01 01 c2 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 37051 BLE_Common: "f0 d2 01 14 d7 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 37054 BLE_Common: "f0 d1 09 00 00 00 00 00 00 00 00 00 ca <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 38058 BLE_Common: "f0 d1 09 00 01 00 00 00 00 00 00 00 cb <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 24354 BLE_Client: "Activated Echelon callbacks."
N 24355 BLE_Client: "We are now connected to the BLE Server."
N 25071 BLE_Common: "f0 d0 01 01 c2 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 25073 BLE_Common: "f0 d2 01 14 d7 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
...
N 3577 HTTP_Server: "Checking for newer firmware:"
N 4998 HTTP_Server: "  - Server version: 1.6.19"
N 5006 HTTP_Server: "  - Current Version: 1.6.20-echelon_fixes-79-g64db89f"
N 5012 BLE_Setup: "Starting Arduino BLE Client application..."
N 5232 BLE_Server: "Starting BLE Server"
N 5242 BLE_Server: "Bluetooth Characteristic defined!"
N 5243 BLE_Setup: "BLE Notify Task Started"
N 5343 BLE_Setup: "Scanning"
N 5343 BLE_Setup: "any any"
N 5343 BLE_Setup: "End BLE Setup"
N 5352 HTTP_Server: "HTTP server started"
N 5353 Main: "Creating Shifter Interrupts"
N 21952 BLE_Client: " - Found Characteristic: 0bf669f4-45f2-11e7-9598-0800200c9a66"
N 22957 BLE_Client: "Successful 0bf669f4-45f2-11e7-9598-0800200c9a66 subscription."
N 22959 BLE_Client: "Registered PM on Connect"

Thank You
John Baker

doudar, 2021-07-15:
Back to the original init where we were at least getting cadence info (but seemingly less d2's). Let me know. 
[firmware.zip](https://github.com/doudar/SmartSpin2k/files/6823737/firmware.zip)

jkbaker4, 2021-07-15:
Hello,

Same result…

N 11596 BLE_Client: "Matching Device Name: ECH-SPORT-064725"
N 11596 BLE_Client: "PM String Matched Any"
N 11597 BLE_Client: "doConnect set on device: 0"
N 12938 Main: "Shift UP: "
N 13110 BLE_Server: "44 02 00 00 00 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(0.00) PW(0) SD(0.00) ]"
N 13113 BLE_Server: "23 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(0.00) PW(0) ]"
N 13114 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 13589 BLE_Client: "BLE Advertised Device found: Name: ECH-SPORT-064725, Address: e5:2e:bc:d1:43:3a, serviceUUID: 0bf669f0-45f2-11e7-9598-0800200c9a66"
N 13591 BLE_Client: "Matching Device Name: ECH-SPORT-064725"
N 13592 BLE_Client: "PM String Matched Any"
N 13593 BLE_Client: "doConnect set on device: 0"
N 13605 BLE_Client: "BLE Advertised Device found: Name: , Address: 64:87:a5:ad:46:21, manufacturer data: 4c0010050318963fe1, txPower: 12"
N 13722 BLE_Client: "BLE Advertised Device found: Name: , Address: 74:db:61:bb:80:a6, manufacturer data: 4c001005441c1e9fa2, txPower: 12"
N 13808 BLE_Client: "BLE Advertised Device found: Name: , Address: 7a:49:a5:b3:57:f1, manufacturer data: 4c001006251eed37707f, txPower: 8"
N 13852 BLE_Client: "BLE Advertised Device found: Name: , Address: 4f:f8:35:b4:be:ca, manufacturer data: 0600010920024344a3450bb64b9d931875e2d49932d284fc7a1a92ff12"
N 14116 BLE_Server: "44 02 00 00 00 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(0.00) PW(0) SD(0.00) ]"
N 14118 BLE_Server: "23 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(0.00) PW(0) ]"
N 14119 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 14332 BLE_Client: "BLE Advertised Device found: Name: , Address: 6b:61:4c:99:30:cc, manufacturer data: 06000109200229dfabafbb5daa5db90dddcafbd2eae15ae6af42c5d635"
N 14377 Main: "Shift UP: "
N 15121 BLE_Server: "44 02 00 00 00 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(0.00) PW(0) SD(0.00) ]"
N 15123 BLE_Server: "23 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(0.00) PW(0) ]"
N 15124 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 15415 Main: "Shift DOWN: "
N 15962 BLE_Client: "BLE Advertised Device found: Name: 1ABcn, Address: 90:e2:02:00:1f:9f"
N 15967 BLE_Client: "Bluetooth Client Found Devices: {"device 0":{"address":"e5:2e:bc:d1:43:3a","name":"ECH-SPORT-064725","UUID":"0bf669f0-45f2-11e7-9598-0800200c9a66"}}"
N 16128 BLE_Server: "44 02 00 00 00 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(0.00) PW(0) SD(0.00) ]"
N 16129 BLE_Server: "23 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(0.00) PW(0) ]"
N 16131 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 16653 Main: "Shift DOWN: "
N 16969 BLE_Client: "Initiating Server Connection"
N 16970 BLE_Client: "Trying to connect to Echelon Bike"
N 16971 BLE_Client: "Forming a connection to:  e5:2e:bc:d1:43:3a"
N 16972 BLE_Client: " - Created client"
N 17134 BLE_Server: "44 02 00 00 00 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(0.00) PW(0) SD(0.00) ]"
N 17136 BLE_Server: "23 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(0.00) PW(0) ]"
N 17138 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 17174 BLE_Client: " - Connected to server"
N 17175 BLE_Client: " - RSSI -110"
N 18141 BLE_Server: "44 02 00 00 00 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(0.00) PW(0) SD(0.00) ]"
N 18143 BLE_Server: "23 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(0.00) PW(0) ]"
N 18145 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 18922 BLE_Client: " - Found service: 0bf669f1-45f2-11e7-9598-0800200c9a66"
N 19149 BLE_Server: "44 02 00 00 00 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(0.00) PW(0) SD(0.00) ]"
N 19150 BLE_Server: "23 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(0.00) PW(0) ]"
N 19153 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 20155 BLE_Server: "44 02 00 00 00 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(0.00) PW(0) SD(0.00) ]"
N 20157 BLE_Server: "23 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(0.00) PW(0) ]"
N 20159 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 20922 BLE_Client: " - Found Characteristic: 0bf669f4-45f2-11e7-9598-0800200c9a66"
N 21162 BLE_Server: "44 02 00 00 00 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(0.00) PW(0) SD(0.00) ]"
N 21163 BLE_Server: "23 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(0.00) PW(0) ]"
N 21165 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 22166 BLE_Server: "44 02 00 00 00 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(0.00) PW(0) SD(0.00) ]"
N 22168 BLE_Server: "23 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(0.00) PW(0) ]"
N 22169 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 22175 BLE_Client: "Successful 0bf669f4-45f2-11e7-9598-0800200c9a66 subscription."
N 22176 BLE_Client: "Registered PM on Connect"
N 23172 BLE_Server: "44 02 00 00 00 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(0.00) PW(0) SD(0.00) ]"
N 23174 BLE_Server: "23 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(0.00) PW(0) ]"
N 23175 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 24073 BLE_Client: "Activated Echelon callbacks."
[... cut ...]
N 9197 BLE_Server: "05 52 00 -> ERG Mode: No Power Meter Connected"
N 10106 BLE_Server: "44 02 00 00 00 00 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(0.00) PW(0) SD(0.00) ]"
N 10107 BLE_Server: "23 00 00 00 00 00 00 00 00 -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(0.00) PW(0) ]"
N 10109 BLE_Server: "00 00 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]"
N 10695 BLE_Client: "BLE Advertised Device found: Name: , Address: 6b:61:4c:99:30:cc, manufacturer data: 06000109200229dfabafbb5daa5db90dddcafbd2eae15ae6af42c5d635"
...

Thank You
John Baker

cagnulein, 2021-07-15:
@doudar resistance values are sent only when the knob changes. It's how echelon firmware works. in QZ i'm just keeping a cache value inside the code :)

jkbaker4, 2021-07-15:
Hello,

I turned off updates and went back to 1.2.17-Echelon-55-ga892a3d and its working.

23 00 00 00 00 23 00 e3 95 <-- CPMC sent
00 00 00 00 00 <-- HR sent
f0 d1 09 00 d5 00 00 00 37 00 31 00 07 <- 0bf669f1-45f2-11e7-9598-0800200c9a66 | 0bf669f4-45f2-11e7-9598-0800200c9a66 | ECH:[ CD(49.00) PW(66) ]
23 00 42 00 00 24 00 c8 9a <-- CPMC sent
00 00 00 00 00 <-- HR sent
f0 d1 09 00 d6 00 00 00 38 00 36 00 0e <- 0bf669f1-45f2-11e7-9598-0800200c9a66 | 0bf669f4-45f2-11e7-9598-0800200c9a66 | ECH:[ CD(54.00) PW(71) ]
23 00 47 00 00 25 00 39 9f <-- CPMC sent
00 00 00 00 00 <-- HR sent5 52 0 <-- From APP
ERG MODE Target: 82 Current: 71 Incline: 0.94
f0 d1 09 00 d7 00 00 00 39 00 37 00 11 <- 0bf669f1-45f2-11e7-9598-0800200c9a66 | 0bf669f4-45f2-11e7-9598-0800200c9a66 | ECH:[ CD(55.00) PW(72) ]
23 00 48 00 00 26 00 96 a3 <-- CPMC sent
00 00 00 00 00 <-- HR sent
f0 d2 01 11 d4 <- 0bf669f1-45f2-11e7-9598-0800200c9a66 | 0bf669f4-45f2-11e7-9598-0800200c9a66 | ECH:[ CD(55.00) PW(72) ]
23 00 48 00 00 27 00 f3 a7 <-- CPMC sent
00 00 00 00 00 <-- HR sent
f0 d1 09 00 d9 00 00 00 3b 00 42 00 20 <- 0bf669f1-45f2-11e7-9598-0800200c9a66 | 0bf669f4-45f2-11e7-9598-0800200c9a66 | ECH:[ CD(66.00) PW(85) ]
23 00 55 00 00 28 00 95 ab <-- CPMC sent
00 00 00 00 00 <-- HR sent
f0 d1 09 00 da 00 00 00 3d 00 45 00 26 <- 0bf669f1-45f2-11e7-9598-0800200c9a66 | 0bf669f4-45f2-11e7-9598-0800200c9a66 | ECH:[ CD(69.00) PW(89) ]

Thank You
John Baker

doudar, 2021-07-15:
> @doudar resistance values are sent only when the knob changes. It's how echelon firmware works. in QZ i'm just keeping a cache value inside the code :)

Good to know 😄  I'm not going crazy 👍 

Any idea why he's not getting Cadence now in spot 10? I'm currently only doing the initilization with one:
 uint8_t initData3[] = {0xf0, 0xb0, 0x01, 0x01, 0xa2};

As multiple init's like you have don't seem to make a difference. 

N 24177 BLE_Common: "f0 d1 09 00 00 00 00 00 00 00 00 00 ca <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"

@jkbaker4 , at the risk of sounding like cheesy tech support, have you tried rebooting your echelon?

doudar, 2021-07-15:
Well that's a good starting point!

doudar, 2021-07-15:
@jkbaker4 Here's another firmware. Even though it (the code) looks (mostly) the same I've rolled back the init process to the version that worked for you in this one. The only difference I see is type byte in the message vs uint_8t (which shouldn't matter but 🤷‍♂️ )

[firmware.zip](https://github.com/doudar/SmartSpin2k/files/6823964/firmware.zip)

jkbaker4, 2021-07-15:
Loaded the 1.6.20-echelon_fixes-79-g64db89f and back to the same result – no CD or PW and unable to hold a connection…

N 22840 BLE_Client: "Activated Echelon callbacks."
N 22842 BLE_Client: "We are now connected to the BLE Server."
N 33692 BLE_Common: "f0 d1 09 00 0a 00 00 00 00 00 00 00 d4 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 34693 BLE_Common: "f0 d1 09 00 0b 00 00 00 00 00 00 00 d5 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 35693 BLE_Common: "f0 d1 09 00 0c 00 00 00 00 00 00 00 d6 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 36693 BLE_Common: "f0 d1 09 00 0d 00 00 00 00 00 00 00 d7 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 37693 BLE_Common: "f0 d1 09 00 0e 00 00 00 00 00 00 00 d8 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 38693 BLE_Common: "f0 d1 09 00 0f 00 00 00 00 00 00 00 d9 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
...
N 3559 BLE_Setup: "Starting Arduino BLE Client application..."
N 3773 BLE_Server: "Starting BLE Server"
N 3782 BLE_Server: "Bluetooth Characteristic defined!"
N 3782 BLE_Setup: "BLE Notify Task Started"
N 3883 BLE_Setup: "Scanning"
N 3883 BLE_Setup: "any any"
N 3883 BLE_Setup: "End BLE Setup"
N 3893 HTTP_Server: "HTTP server started"
N 3894 Main: "Creating Shifter Interrupts"
N 6763 BLE_Client: "BLE Advertised Device found: Name: , Address: 4c:9f:76:eb:10:5a, manufacturer data: 0600010920022fe848e7b47160ecd85aca0bf8325d8abfecefa1e9d6cc"
...
N 3560 BLE_Setup: "Starting Arduino BLE Client application..."
N 3769 BLE_Server: "Starting BLE Server"
N 3778 BLE_Server: "Bluetooth Characteristic defined!"
N 3779 BLE_Setup: "BLE Notify Task Started"
N 3879 BLE_Setup: "Scanning"
N 3879 BLE_Setup: "any any"
N 3879 BLE_Setup: "End BLE Setup"
N 3888 HTTP_Server: "HTTP server started"
N 3889 Main: "Creating Shifter Interrupts"
N 12272 BLE_Client: "BLE Advertised Device found: Name: , Address: 40:0e:d5:3e:15:7e, manufacturer data: 4c001006271ecfaf9eac, txPower: 8"
N 16250 BLE_Client: " - Connected to server"
N 16251 BLE_Client: " - RSSI -107"
N 25759 BLE_Common: "f0 d1 09 00 02 00 00 00 00 00 00 00 cc <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 26763 BLE_Common: "f0 d1 09 00 03 00 00 00 00 00 00 00 cd <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 27765 BLE_Common: "f0 d1 09 00 04 00 00 00 00 00 00 00 ce <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 28766 BLE_Common: "f0 d1 09 00 05 00 00 00 00 00 00 00 cf <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 29767 BLE_Common: "f0 d1 09 00 06 00 00 00 00 00 00 00 d0 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 30768 BLE_Common: "f0 d1 09 00 07 00 00 00 00 00 00 00 d1 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 31768 BLE_Common: "f0 d1 09 00 08 00 00 00 00 00 00 00 d2 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 32768 BLE_Common: "f0 d1 09 00 09 00 00 00 00 00 00 00 d3 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 33768 BLE_Common: "f0 d1 09 00 0a 00 00 00 00 00 00 00 d4 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 34769 BLE_Common: "f0 d1 09 00 0b 00 00 00 00 00 00 00 d5 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 41777 BLE_Common: "f0 d1 09 00 12 00 00 00 00 00 00 00 dc <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 43779 BLE_Common: "f0 d1 09 00 14 00 00 00 00 00 00 00 de <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 44780 BLE_Common: "f0 d1 09 00 15 00 00 00 00 00 00 00 df <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 47783 BLE_Common: "f0 d1 09 00 18 00 00 00 00 00 00 00 e2 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 51788 BLE_Common: "f0 d1 09 00 1c 00 00 00 00 00 00 00 e6 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 54790 BLE_Common: "f0 d1 09 00 1f 00 00 00 00 00 00 00 e9 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 55791 BLE_Common: "f0 d1 09 00 20 00 00 00 00 00 00 00 ea <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 56791 BLE_Common: "f0 d1 09 00 21 00 00 00 00 00 00 00 eb <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 57791 BLE_Common: "f0 d1 09 00 22 00 00 00 00 00 00 00 ec <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 58791 BLE_Common: "f0 d1 09 00 23 00 00 00 00 00 00 00 ed <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
N 59792 BLE_Common: "f0 d1 09 00 24 00 00 00 00 00 00 00 ee <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) PW(0) ]"
...
N 3471 BLE_Setup: "Starting Arduino BLE Client application..."
N 3691 BLE_Server: "Starting BLE Server"
N 3699 BLE_Server: "Bluetooth Characteristic defined!"
N 3699 BLE_Setup: "BLE Notify Task Started"
N 3800 BLE_Setup: "Scanning"
N 3800 BLE_Setup: "any any"
N 3800 BLE_Setup: "End BLE Setup"
N 3812 HTTP_Server: "HTTP server started"
N 3813 Main: "Creating Shifter Interrupts"

Thank You
John Baker

doudar, 2021-07-15:
Okay. I've got to head out to do some other things. I'll think about this issue though and get back to you later today. If you can check, does the binary from this release work for you: https://github.com/doudar/SmartSpin2k/releases/tag/1.3.21

marklogan17, 2021-07-16:
Sorry I was rather busy at work today.  Let me know when you have the next build to test.  I should get a little bit of time tomorrow afternoon to run a couple of tests in between meetings.

doudar, 2021-07-16:
> Sorry I was rather busy at work today. Let me know when you have the next build to test. I should get a little bit of time tomorrow afternoon to run a couple of tests in between meetings.

If you could, please test the firmware.bin in that release above. You might have to upload the raw status.html via the web updater also in order to get the webpages working. That will give me a base as to where things may have broken.

marklogan17, 2021-07-17:
Here's my debug log from the 1.3.21 firmware...

Served /status.html
Served /style.css
 - Found Characteristic:0bf669f4-45f2-11e7-9598-0800200c9a66 
Sucessful 0bf669f4-45f2-11e7-9598-0800200c9a66 subscription.
Registered PM on Connect
<- 0bf669f1 | 0bf669f4 | ECH:[ ] 
f0 d1 09 00 00 00 00 00 00 00 00 00 ca <- 0bf669f1 | 0bf669f4 | ECH:[ CD(0.00) ] 
Activated Echelon callbacks.
We are now connected to the BLE Server.
f0 d2 01 13 d6 <- 0bf669f1 | 0bf669f4 | ECH:[ CD(0.00) PW(0) ] 
f0 d1 09 00 00 00 00 00 00 00 00 00 ca <- 0bf669f1 | 0bf669f4 | ECH:[ CD(0.00) PW(0) ] 
f0 d1 09 00 01 00 00 00 00 00 00 00 cb <- 0bf669f1 | 0bf669f4 | ECH:[ CD(0.00) PW(0) ] 
f0 d1 09 00 02 00 00 00 00 00 00 00 cc <- 0bf669f1 | 0bf669f4 | ECH:[ CD(0.00) PW(0) ] 
f0 d1 09 00 03 00 00 00 00 00 00 00 cd <- 0bf669f1 | 0bf669f4 | ECH:[ CD(0.00) PW(0) ] 
f0 d1 09 00 04 00 00 00 00 00 00 00 ce <- 0bf669f1 | 0bf669f4 | ECH:[ CD(0.00) PW(0) ] 
f0 d1 09 00 05 00 00 00 00 00 00 00 cf <- 0bf669f1 | 0bf669f4 | ECH:[ CD(0.00) PW(0) ] 
f0 d1 09 00 06 00 00 00 00 00 00 00 d0 <- 0bf669f1 | 0bf669f4 | ECH:[ CD(0.00) PW(0) ] 
f0 d1 09 00 07 00 00 00 00 00 00 00 d1 <- 0bf669f1 | 0bf669f4 | ECH:[ CD(0.00) PW(0) ] 
f0 d1 09 00 08 00 00 00 00 00 0f 00 e1 <- 0bf669f1 | 0bf669f4 | ECH:[ CD(15.00) PW(46) ] 
f0 d1 09 00 09 00 00 00 01 00 0f 00 e3 <- 0bf669f1 | 0bf669f4 | ECH:[ CD(15.00) PW(46) ] 
f0 d1 09 00 0a 00 00 00 02 00 31 00 07 <- 0bf669f1 | 0bf669f4 | ECH:[ CD(49.00) PW(78) ] 
f0 d1 09 00 0b 00 00 00 03 00 3a 00 12 <- 0bf669f1 | 0bf669f4 | ECH:[ CD(58.00) PW(90) ] 
f0 d1 09 00 0c 00 00 00 04 00 3e 00 18 <- 0bf669f1 | 0bf669f4 | ECH:[ CD(62.00) PW(95) ] 
f0 d1 09 00 0d 00 00 00 06 00 4b 00 28 <- 0bf669f1 | 0bf669f4 | ECH:[ CD(75.00) PW(116) ] 
f0 d1 09 00 0e 00 00 00 07 00 51 00 30 <- 0bf669f1 | 0bf669f4 | ECH:[ CD(81.00) PW(127) ] 
f0 d1 09 00 0f 00 00 00 08 00 53 00 34 <- 0bf669f1 | 0bf669f4 | ECH:[ CD(83.00) PW(131) ] 
f0 d1 09 00 10 00 00 00 0a 00 4f 00 33 <- 0bf669f1 | 0bf669f4 | ECH:[ CD(79.00) PW(124) ]
Shift UP: 900 
f0 d2 01 15 d8 <- 0bf669f1 | 0bf669f4 | ECH:[ CD(79.00) PW(147) ] 
f0 d1 09 00 12 00 00 00 0c 00 48 00 30 <- 0bf669f1 | 0bf669f4 | ECH:[ CD(72.00) PW(132) ] 
f0 d1 09 00 13 00 00 00 0d 00 43 00 2d <- 0bf669f1 | 0bf669f4 | ECH:[ CD(67.00) PW(122) ] 
f0 d1 09 00 14 00 00 00 0e 00 47 00 33 <- 0bf669f1 | 0bf669f4 | ECH:[ CD(71.00) PW(130) ] 
f0 d1 09 00 15 00 00 00 10 00 4a 00 39 <- 0bf669f1 | 0bf669f4 | ECH:[ CD(74.00) PW(136) ] 
f0 d1 09 00 16 00 00 00 10 00 45 00 35 <- 0bf669f1 | 0bf669f4 | ECH:[ CD(69.00) PW(126) ] 
f0 d1 09 00 17 00 00 00 10 00 45 00 36 <- 0bf669f1 | 0bf669f4 | ECH:[ CD(69.00) PW(126) ] 
f0 d1 09 00 18 00 00 00 10 00 45 00 37 <- 0bf669f1 | 0bf669f4 | ECH:[ CD(69.00) PW(126) ]

marklogan17, 2021-07-17:
@doudar seems my config is a little corrupt at this point.  How do I completely reset the ESP board/storage?  Here's the log when I try and save settings:
[14:54:10]N 146783 Main: "Stepper power is now 1500"
[14:54:10]N 146797 Main: "Stealthchop is now 1"
[14:54:10]N 146806 HTTP_Server: "Config Updated From Web"
[14:54:10][E][vfs_api.cpp:135] remove(): /config.txt does not exists or is directory
[14:54:10]N 146816 Config: "Writing File: /config.txt"
[14:54:11][E][vfs_api.cpp:270] VFSFileImpl(): fopen(/spiffs/config.txt) failed
[14:54:11]E 146978 Config: "Failed to create file"
[14:54:11]N 146979 Config: "Contents of file: /config.txt"
[14:54:11][E][vfs_api.cpp:135] remove(): /userPWC.txt does not exists or is directory
[14:54:11]N 147004 Config: "Writing File: /userPWC.txt"
[14:54:11][E][vfs_api.cpp:270] VFSFileImpl(): fopen(/spiffs/userPWC.txt) failed
[14:54:11]N 147153 BLE_Common: "f0 d1 09 00 79 00 00 00 00 00 00 00 43 <- 0bf669f1 | 0bf669f4 | ECH[ CD(0.00) ]"
[14:54:11]E 147153 Config: "Failed to create file"
[14:54:11]N 147154 Config: "Contents of file: /userPWC.txt"

doudar, 2021-07-17:
If you can’t “reset to defaults “ in the settings page, you can hold both shifters while you connect power and it will erase spiffs and then (on reboot) prompt you for your network info so it can download the files from GitHub.

marklogan17, 2021-07-17:
Do I need to be on an proper release for that?  I got a 404 after prompted for network info.

marklogan17, 2021-07-17:
> Do I need to be on an proper release for that? I got a 404 after prompted for network info.

Maybe I can just download the latest spiffs.bin and flash after flashing my firmware?  Firmware is built off tree from about a week ago.

doudar, 2021-07-17:
> Do I need to be on an proper release for that? I got a 404 after prompted for network info.

It should be back on your network and downloading the newest release after the reset. If for some reason that doesn’t work, the best way is to plug it in via usb in visual studio with this project open and select PlatformIO->upload filesystem image

marklogan17, 2021-07-17:
It wasn't downloading, even with latest firmware release.  Not sure what I screwed up.  Anyway, in my build directory I ran pio run -t uploadfs and it built spiffs and uploaded it by the looks of it.  Will test more but think I'm good now.

closed 2021-11-25 by doudar