doudar/SmartSpin2k #265: [BUG] (Manual firmware updates are slow)

doudar, 2021-12-03:
### Description

Manual firmware updates from the web interface are slower after UDP Logger implementation

### Additional Information
The UDP logging or erroneous messages from the BLE server being shut down should be turned off during firmware updates. This isn't critical because updates still complete, just slower than normal.

MarkusSchneider, 2021-12-03:
I can implement a setting to disable / enable the upd logger. A 'normal' user is maybe not interested in the logs. Default is disable.
With an option in the settings dialog the upd logging could be enable.
During FW update you can disable logging by using the setting.
Setting is evaluated her: https://github.com/doudar/SmartSpin2k/blob/develop/src/SS2KLog.cpp#L73 
before delegating the message to the UPD_Logger.

doudar, 2021-12-03:
It should be an easy setting to toggle wherever we put it. It's possible we could just add another config option to the userconfig class but there are a bunch of cascading things to take into account whenever anything is modified there (webpages, JSON sending, etc....) 

It honestly might just be easier removing the repeating spam messages occurring when you put the unit into firmware upload mode via the web page (just go to the firmware upload page and look at the log). Those errors are because if the BLE radio isn't turned off during a firmware update, it fails. 

Thoughts?

MarkusSchneider, 2021-12-04:
You are right, there are a lot of things to do to add a new setting. If it is not necessary, lets suppress the "spam" messages :)

MarkusSchneider, 2021-12-23:
Should be fixed with if ERG Branch is merged. [See] (https://github.com/MarkusSchneider/SmartSpin2k/blob/develop/src/HTTP_Server_Basic.cpp#L307)

doudar, 2021-12-23:
Thanks for pointing this out. Indeed they are faster now when I’ve tried on that branch.

doudar, 2021-12-27:
Fixed with #282

closed 2021-12-27 by doudar