doudar/SmartSpin2k #268: [FR] (ERG Mode needs TLC (and math....lots of math...)

doudar, 2021-12-03:
**Is your feature request related to a problem? Please describe.**
A great ERG mode has been fighting me for a long time due to the variability (and sometimes completely nowhere near linear resistance) of spin bikes. Because of this, I've only managed meh ERG mode. 

**Describe the solution you'd like**
I think it may need some type of learning algorithm where is builds a small lookup table that says "I'm at 90 rpm at X power and need to get to Y setpoint - last time that took z steps, so I'll move z steps." The issues are the previously mentioned non-linear response, variable pedaling cadence and total system delay from when a set point is received to when the new power is read from the power meter.  

**Describe alternatives you've considered**
It's possible a proper tuned PID loop may work but the last time I tried PID here I failed miserably. 

**Additional context**
ERG code starts around line 210 in BLE_Server.cpp

MarkusSchneider, 2021-12-03:
I also failed ;) tried it last winter.
The main problem is that we need the power measurement more frequent.

A cycle time of 1 sec is to slow. 
I think the current algorithm is not so bad. A spinning bike has an nonlinear curve when cadence changes.
With higher cadence the wheel is spinning faster, produce more friction and a change of incline has a bigger impact.

My recommendation is to decouple the ERG calculation cycle from the notification cycle from Zwift and increase the cycle time of power measurement.
1) Create a new vTask with a 100 ms cycle Time. Calculate incline with current algorithm. 
2) Take cadence in account. We can do this by dividing a constant e.g 70 by cadence. With cadence 100 we get then a factor of 0.7 and we can use this factor [her](https://github.com/doudar/SmartSpin2k/blob/develop/src/BLE_Server.cpp#L244), likely we use the ERG Sensitivity.
3) We must read out Power Meter or Heart Rate more frequent by increasing the BLE_CLIENT_DELAY, I guess

The reaction time of the whole loop could be a problem. But this time depends on the setup of the bike, stepper, power meter.
But if we read power more frequent and compare them with the Setpoint we can correct them quick.
We should than introduce a hysteresis, a band around the Setpoint where no Incline change is made.

doudar, 2021-12-03:
If you wanted to add another task, try moving everything out of the loop() in main.cpp except add vTaskDelete(NULL);. That way we should be able to free up some resources that aren't really being used for anything much. I'm guessing the default loop task allocates a pretty big stack as well so it's been on my list of things to ditch and this would be a perfect opportunity.

The only things in loop() right now are; checking to see if a shift happened and logging it (Has to happen outside the interrupt because it takes too long). And loop counter/timer that checks driver temperature and if the shifters have been held down for 4 consecutive loops (seconds). There's no reason that either couldn't be moved to any other loop and just check millis(). Every time the CheckDriverTemperature() is run it will log a message if the driver is too hot so we definitely don't want that getting called too often.    \

Then you can spawn your new task from setup() and run it as fast as you'd like. I'm sure there's probably already plenty of power left to add a new task without removing loop() but I'd like to try to keep things as balanced as possible moving forward because we're asking quite a lot out of this microprocessor (and were nowhere near running out of ideas yet.) 

I'm on vacation starting tomorrow, so hopefully you don't hear much out of me this upcoming week.

MarkusSchneider, 2021-12-05:
Hi, I did some research to find out how my spinning bike behaves.

The relation between cadence and power is almost linear:
[image]

Between the series I changed the resistance manually. Higher series No -> higher resistance.

Relation between Incline (Shifter Step) and Power is exponential when cadence stays same:
[image]

Maybe someone can do the same research on his bike, then we can compare the curves.
For ERG computation we must use an exponential function to guess the next incline and it would be good to use cadence change as an additional factor.

To calculate the next incline we need a triple of Stepper Position, Cadence, Power. Ideally all at the same time.
Cadance and Power should be created at the same time but this depends on the power meter. To get the Position of the stepper the [getCurrentPosition](https://github.com/gin66/FastAccelStepper/blob/master/src/FastAccelStepper.h#L142) can be used.

It could be a problem when power measurement takes long, the stepper reached the final position and the final position is not producing the requested power. 
The whole system gets then unstable. We get than is curve with inclines with huge under and overflows.
[image]

If we can compute incline frequently, a correction of the first computation can be made before the stepper reaches the final position and with every cycle we getting closer the the target.

doudar, 2021-12-06:
Great information! I’ll read it over a few times and think it over this week.

MarkusSchneider, 2021-12-07:
Hi,
I tried to speed up the BLE notification cycle of my Power-Meter Pedals with no luck. 
Assioma sends every ~1 Second one Cadence and Power value via Notification.
I've opened a support request to assioma to clarify if it is possible to get the values more frequent.
But I believe this would not be possible.  

If we can not get power values more frequent the only way to make ERG faster is to compute target incline better.
Then we need viewer cycles to adjust the correct value and hopefully will not get oscillations.

As you can see in my upper comment we need ideally the target-cadence and the target-power.
The target-power value we get from Zwift.  As far as I know do we not get the target-cadence from zwift.
So we missing one (linear) factor for exact computation.

With power value we can use an exponential function to get the target-incline. 
The parameters of this function may depend on the bike configuration, we have to verify this.
First approach is to implement fix parameters and make a test ride.

I'll keep you up to date :)

MarkusSchneider, 2021-12-07:
Tried it with no effect. Seems that this Setting (60, 200, ...) only impacts the connection interval and notifications is only sent when value changes, new value is collected. I guess that the power meter collects only one value per seconds (more than enough for power measurement but not sufficient for controlling :))

doudar, 2021-12-07:
Oh, I gotcha. You’re able to receive at intervals less than one second but the pedals only recalculate the value every second.

Does the value update faster if your cadence is higher? Maybe it takes one or two full crank revolutions for the pedals to calculate power?

MarkusSchneider, 2021-12-07:
mmh... getting power measurement faster than one revolution will not make sense.

MarkusSchneider, 2021-12-07:
Getting power measurements more frequent will not make so much sense as we need a complete revolution to get the power of the complete cycle. It will not be fast enough to create an algorithm to correct the stepper before stepper reach the final position. So we must try the calculate the position as good as possible.

doudar, 2021-12-21:
PowerTable::lookup ended up being pretty complex. I’m not looking forward to testing (debugging) it. 

PowerTable::newEntry should be complete as well. 

Now I just need to add these into the ERG loop and start testing. My plan is to PowerTable::newEntry anytime CAD>50 && power is stable (within 25w for ~3 seconds) then use PowerTable::lookup in ERG mode when a new power target is received. 

Todo are PowerTable::load and PowerTable::save so it’s persistent between boots, PowerTable::toLog to display the contents in the log, and then a new class for homing the stepper and userConfig variable to hold the valid range of stepper travel.

MarkusSchneider, 2021-12-22:
Hi Anthony,

I had time to take a look at your changes.
- As far as I understand you've created a table with the size of 20. 
   Into the table a tuple of cadence/power values is added and the target stepper position.
  - target stepper position will be fine if stepper is no longer moving, but we can also use the current stepper position  (rtConf.getCurrentIncline). Could be more precise. 
- The table stores chunks of power-values. A new Chunk is created if abs(power-values) > 50 Watts.
  - I think we can create the chunks based on the Watts requested by Zwift (TagetWatts/newSetPoint) e.g if Zwift requests 180W, check if current-power is around 180W, if value is between the range, add. Zwift slightly changes the target-watts. We must also us a range here.
    - Advantage: we must not build 50W chunks and we have the exact set-point for the request power.
    - Disadvantage: saving of table makes no sense because table depends on workout.
  - For the chunks we must also check the cadence e.g. create new chunk if abs(cadence) > 10% or we must normalize the power to a cadence (e.g 85)
- For every chunk we can add a timestamp and use the timestamp for an expiration policy. If no space left in the table, oldest entry will be skipped. Advantage: Can build smaller chunks, no need to skip high power-values. 

I'll try to create unit test. I've a lot of data from my last workouts.

Greetings
Markus

doudar, 2021-12-22:
@MarkusSchneider yes, you’ve got most of it correct.

An ERG lookup request finds the closest two chunks and generates an interpolation between the two which is then corrected for cadence. 

The table building (and averaging new inputs) happens continuously anytime cadence is above 50rpm and in any mode, not just ERG. 

There is a 5 item pre-buffer where all inputs have to be within that chunk. If an input value falls outside the chunk, the pre-buffer gets dumped and we start looking again.

That’s where I think it should be built off of stepper position. I agree about target position, We should expose and store the actual stepper position instead of target position.

Input data is capped to 10 averages per chunk.

I really like the idea of building the table the whole time while riding as I think we can then use this data in other modes as well. Perhaps as a way to detect end stops without homing 

(I.e. treat 50w stepper position as minimum and have the user input the maximum wattage their bike can handle as the maximum range)

doudar, 2021-12-22:
@MarkusSchneider I made an SS2K class really quickly to wrap up some globals that were on their own. Those should probably be moved into rtConfig instead.

MarkusSchneider, 2021-12-22:
> @MarkusSchneider yes, you’ve got most of it correct.
> 
> An ERG lookup request finds the closest two chunks and generates an interpolation between the two which is then corrected for cadence.

Good idea 👍
> The table building (and averaging new inputs) happens continuously anytime cadence is above 50rpm and in any mode, not just ERG.
>
> There is a 5 item pre-buffer where all inputs have to be within that [...]

Should we also check here for cadence and skip the pre-buffer if cadence outside a certain range e.g. 10%?
 
> That’s where I think it should be built off of stepper position. I agree about target position, We should expose and store the actual stepper position instead of target position.
> 
> Input data is capped to 10 averages per chunk.

👍
> I really like the idea of building the table the whole time while riding as I think we can then use this data in other modes as well. Perhaps as a way to detect end stops without homing.
>
> (I.e. [...]

doudar, 2021-12-22:
> Good idea 👍
>
>
> Should we also check here for cadence and skip the pre-buffer if cadence outside a certain range e.g. 10%?

Yes, I think that’s a great idea. Also, I think the current watt criteria is too wide. 25w (vs 50w) would probably be better.

> 👍

doudar, 2021-12-22:
If you get any debugging time today, I think the erroneous table inputs come from when the pre-buffer is dumped. Also, there’s a fixed size char in the powertable logging function that overflows if the target steps goes over 9999 easy fix, I just didn’t get time. I have to work so no programming for me today

MarkusSchneider, 2021-12-22:
I've change logging from 4 to 8 digits.
Found a problem if powerBuffer is not completely  filled (powerEntry[i] reading == 0), but this would not explain why power is to high...
Make a test run now 😉

doudar, 2021-12-22:
> I've change logging from 4 to 8 digits.
> Found a problem if powerBuffer is not completely filled (powerEntry[i] reading == 0), but this would not explain why power is to high...
> Make a test run now 😉

Awesome! 

It’s kind of fun to watch it build the table while you’re riding isn’t it?

doudar, 2021-12-22:
@MarkusSchneider . I was able to get a few minutes to test your changes. You fixed the erroneous table input and the table input seems to be working very well!!!

This is going to be great - Overall it's working very nice but maybe it isn't picking up the correct pairs of results? The output jumped nicely on ERG target changes. From high target watts to low target watts it was pretty much spot on. From low target watts to high target watts, the table result seemed a little too low. Still not bad, just a touch too low. And this is on a *very* difficult to control felt resistance bike. 

Very impressive so far! 

Tons of data in the log for you vv  
https://pastebin.com/VwLmWreD

doudar, 2021-12-22:
Thinking about this a bit, I’m guessing the cadence correction is probably a bit too much for my bike. @MarkusSchneider , should we decrease it or do you think there’s a way to generate it dynamically via the powertable data. Maybe look for similar inputs in the pre-buffer and check their watt/cadence/target relationship?

doudar, 2021-12-23:
I changed my mind. After doing another test ride, the power table lookup function seems to be spot on with the numbers it’s producing. Need more testing though

MarkusSchneider, 2021-12-23:
I think we must understand the spinning bikes better. I like to create a function to collection bike characteristics. To create a chart like:
[image]

For the first version I have a simple logging in mind [see](https://github.com/MarkusSchneider/SmartSpin2k/blob/develop/src/ERG_Mode.cpp#L85) enabled by an option in the Bluetooth Simulator page.
You can then create different statistics e.g.
- Keep Stepper steps constant, change cadence -> We get the CAD/POW factor
- Increase Stepper steps, keep cadence constant -> We can check if higher friction behaves linear or not

doudar, 2021-12-23:
After putting the SS2K on my magnetic resistance bike, I figured out where the issue is with not setting the correct resistance after powertable::lookup. 

If lookup commanded a move that took longer than erg_task delay, the regular erg task overrode the lookup move partway through it. 

To solve this, I added a while(rtConfig.getTargetIncline!=rtConfig.getCurrentIncline){vTaskDelay(100/portTICK_PERIOD_MS);}

The output was spot on and it’s starting to look pretty good!!!! The response was absolutely as good as my wheel off bike trainer. 

[image]

MarkusSchneider, 2021-12-23:
Awesome, good work Anthony 👍
I also think that we need calibration and it should not be a problem.
Let's try the new erg mode and wait for feedback. If you like, we can create an option to switch between old an new ERG-Mode.

In future we can create an automatism to determine erg-sensitivity but for now I'm absolutely happy with our result 😊

doudar, 2021-12-23:
> Awesome, good work Anthony 👍 I also think that we need calibration and it should not be a problem. Let's try the new erg mode and wait for feedback. If you like, we can create an option to switch [...]

@MarkusSchneider, One more bug needs to be fixed on it before merging. If I pause a workout in Zwift, 50% of the time I’m getting “ERG Request received but no PM connected”. I don’t have it in a log currently for you, I’ll see if I can catch it in a log later.

MarkusSchneider, 2021-12-24:
Ok, I'll debug it. Looks like erg_mode is disabled if you pause the workout an not enabled after resuming.

doudar, 2021-12-24:
When you press pause, I think it just switches to Sim mode, then back to ERG mode when unpause/resume.

MarkusSchneider, 2021-12-24:
It's maybe only an erroneous log because when you pause, erg_mode is disabled but targetWatts not reset to 0. 
Can you try to set targetWatts to 0 if ErgMode is disable:
https://github.com/MarkusSchneider/SmartSpin2k/blob/develop/include/SmartSpin_parameters.h#L80
void setERGMode(bool erg) { ERGMode = erg;  if(!erg) {targetWatts = 0;}}

After resuming all should start again with the first sending of targetWatts by zwift.
I'll try it tomorrow. 

I also like to do same smaller changes:
- Only run ERG if powerValue > 0 or >10 like you do in PowerTable.
- Update PowerEntry only if a new Measurement is present by comparing the timestamp of last and current powerValue

Happy Christmas 
Markus

doudar, 2021-12-24:
Merry Christmas to you too Markus! 

I found the issue and corrected it on my end. If ERG mode was turned off and rtConfig.getTargetWatts>0 , it would send that message. I just added (line 73) to setTargetWatts(0); if that is the case.

doudar, 2021-12-27:
Very happy to close this one. Fixed with #282

closed 2021-12-27 by doudar