doudar/SmartSpin2k #273: [FR] RGT Cycling support

pupdeg, 2021-12-12:
Is RGT Cycling fully supported? I'm just using the program for the virtual power meter (works great in Zwift for me), but in RGT Cycling the SS2K device doesn't seem to pass through the heart rate information. It's not a huge deal, it passes through the derived power no problem, but it'd be useful if it showed heart rate too.

doudar, 2021-12-14:
We worked on this a bunch last year and both RGT and Fulgaz seemingly will not accept Heart Rate service from a device that has cycling power service, cycling speed and cadence service or Fitness Machine Service. 

I think we’d need to be able to present a second (fake) BLE device with only the Heart Rate Service, but it’s currently not possible with the ESP32.

doudar, 2021-12-14:
@h2zero , do you think there’s any way to create a second NIMBLE_DEVICE on the same esp32? 

The issue we have is a certain Clients that will not accept the Heart Rate Service when it’s combined with other services (see above) - it could be worked around if we could (spoof) another Device.

h2zero, 2021-12-14:
That wouldn't really be possible. What you can do though is dynamically change the services of the server. What is the client rejecting? I mean does it just disconnect if the server has other services? Or is it refusing to connect if the advertisement doesn't match something?

doudar, 2021-12-14:
It’s refusing to connect if it’s connected to one of the other services (from what I remember)

EdKingscote, 2022-01-02:
I disabled device filtering in Fulgaz and the HRM showed up on the SS2K to connect and gave me data that looked reasonable.

closed 2022-07-09 by doudar