doudar/SmartSpin2k #317: Strange behavior (debug messages "no data in powertable" and "temperature now under control")

holger8080, 2022-01-29:
Hi all, 

I switched back to Zwift and used my indoor bike more often again recently.
While last fall Zwift and the smartspin with the old firmware worked flawlessly, I'm now having a lot of problems.
I have a Schwinn IC4, Assioma Unos, and updated the smartspin2k to the latest firmware "1.12.30-develop".

Problems include:

- if I use smartspin without Zwift or any connection to any app or device and I just want to use the up or down buttons, there will be "random" behavior after a while. I don't press any button, yet the resistance is going up or down. There is no ERG connected, so why is the resistance changing? 
- There are lots of debug messages saying that "no data found in powertable" and "temperature is now under control". Looking at the main.cpp code (but not understanding it fully ;-)  ), that debug message should only appear in pairs with a previous "overtemp" debug message, shouldn't it? But they don't.
- listening to the noise the stepper motor makes, it seems to hit a torque limit sometimes. I never had these problems with the v1.16 firmware (and downgraded to confirm it's no HW issue) with the same "stepper motor power" settings. Changing them didn't have much of an effect with the 1.12.30 firmware. Could there be a link to the "overtemp" reset? 

Debug log excerpt when pressing up/down buttons: 
> N 45764 ERG_Mode: "No data found in powertable."
> N 45771 ERG_Mode: "No data found in powertable."
> N 58216 Main: "Shift DOWN: "
> N 59020 Main: "Shifters Held 2"
> N 59038 Main: "Temperature is now under control. Driver current reset."
> N 59642 Main: "Shift DOWN: "
> N 64078 Main: "Temperature is now under control. Driver current reset."
> 73959 Main: "Shift UP: "
> 74176 Main: "Temperature is now under control. Driver current reset."
> 81475 ERG_Mode: "No data found in powertable."

When I downgraded to the old v1.16 firmware, all these problems were gone (so it doesn't seem to be a faulty HW), but of course Zwift doesn't work anymore.

Could there be a memory leak or sth like that? How else could the debug message "Temperature is now under control. Driver current reset." appear without ever showing "Overtemp! Driver is throttleing down!" ?

Is there any short-term workaround (downgrade etc) I can do to get Zwift to work but don't run into these problems?

Would appreciate some help. 
And: thanks for the project, using smartspin2k on my Schwinn with Zwift really helps me clear my mind, so your work is highly appreciated. :-)

MarkusSchneider, 2022-01-29:
Hi Holger,
- random resistance changes: I looked over the code but could not find the problem. I'll do a test today to see if I can reproduce it. If you are not connected to zwift, resistance is controlled by shifters and there are two way to set shifter 
   1) via shifter butten, 
   2) via html shifter settings page.  
- Log-Message: 
  - "Overtemp!" is written with log level warning
(https://github.com/doudar/SmartSpin2k/blob/a6220ed6808148208abfc6ff671c702c26b77718/src/Main.cpp#L372)
my be the reason why it is not present in log output.
  - ERG_Mode: "No data found in powertable.". You can ignore this message. If you are interested in details [see](https://github.com/doudar/SmartSpin2k/issues/268#issuecomment-999652893)
- stepper noise: Have you tried to increase / decrease stepper current, enabled/disabled stealthchop? I don't have any noise from stepper with the defaults. 
[image]

you can merge: https://github.com/doudar/SmartSpin2k/blob/a6220ed6808148208abfc6ff671c702c26b77718/src/BLE_Server.cpp#L341-L349
down to the latest version that worked for you. Zwift should then work again.

doudar, 2022-01-29:
It’s interesting that you don’t have the driver temp messages with the older firmware. They’re typically displayed when a motor driver with no/bad/incompatible uart interface is used. @eMadman discovered fystec had batch of 2208’s where the uart doesn’t work. Symptoms are the driver temp warning and not being able to set the driver current. 

I’ll check and see what differences there are between 1.16 and the current version with regard to driver code and shifter denounce and get back to you later today. 

Thanks for the report!

MarkusSchneider, 2022-01-29:
Did a quick test with a duration ~10 Min. I don't have any problems with resistance changing. @holger8080 can you check your logs if you have "shifter message" in the log when you have this problem (Shift DOWN or Shift UP).
Just to verify that it is not a problem with the micro switches or wires.

@doudar I'm seeing  the "Temperature is now under control. Driver current reset." log message too.
Were cycling for ~ 1h and had this message. "Overtemp" message was not logged.

> N 584099 BLE_Common: "10 97 97 01 94 01 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(151) ]"
> N 584102 BLE_Common: "23 00 72 00 66 ac 11 fe 49 <- 0x1818 | 0x2a63 | CPS[ CD(72.80) PW(114) ]"
> N 584108 BLE_Server: "44 02 4e 13 90 00 72 00 97 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(151) CD(72.00) PW(114) SD(942.00) ]"
> N 584114 BLE_Server: "23 00 72 00 00 f1 02 23 ac -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(72.00) PW(114) ]"
> N 584117 BLE_Server: "00 97 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(151) ]"
> N 584545 ERG_Mode_CSV: "8;295.00;335.00;130;130;113;114;66;72"
> **N 584699 Main: "Temperature is now under control. Driver current reset."**
> N 584820 BLE_Common: "10 97 96 01 98 01 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(151) ]"
> N 584824 BLE_Common: "23 00 5d 00 60 ad 11 2e 4d <- 0x1818 | 0x2a63 | CPS[ CD(75.29) PW(93) ]"
> N 584830 BLE_Server: "44 02 1c 14 96 00 5d 00 97 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(151) CD(75.00) PW(93) SD(148.00) ]"
> N 584834 BLE_Server: "23 00 5d 00 00 f2 02 56 af -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(75.00) PW(93) ]"
> N 584837 BLE_Server: "00 97 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(151) ]"
> N 585257 ERG_Mode_CSV: "9;335.00;427.50;130;130;114;93;72;75"
> N 585540 BLE_Common: "10 96 9a 01 95 01 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(150) ]"

doudar, 2022-01-29:
> Did a quick test with a duration ~10 Min. I don't have any problems with resistance changing. @holger8080 can you check your logs if you see "shifter messages" in the log when you have this problem (Shift DOWN or Shift UP). Just to verify that it is not a problem with the micro-switches, wires or plug.
> 
> @doudar I'm seeing the "Temperature is now under control. Driver current reset." log message too. Were cycling for ~ 1h and had this message. "Overtemp" message was not logged.
> 
> > N 584099 BLE_Common: "10 97 97 01 94 01 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(151) ]"
> > N 584102 BLE_Common: "23 00 72 00 66 ac 11 fe 49 <- 0x1818 | 0x2a63 | CPS[ CD(72.80) PW(114) ]"
> > N 584108 BLE_Server: "44 02 4e 13 90 00 72 00 97 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(151) CD(72.00) PW(114) SD(942.00) ]"
> > N 584114 BLE_Server: "23 00 72 00 00 f1 02 23 ac -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(72.00) PW(114) ]"
> > N 584117 BLE_Server: "00 97 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(151) ]"
> > N 584545 ERG_Mode_CSV: "8;295.00;335.00;130;130;113;114;66;72"
> > **N 584699 Main: "Temperature is now under control. Driver current reset."**
> > N 584820 BLE_Common: "10 97 96 01 98 01 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(151) ]"
> > N 584824 BLE_Common: "23 00 5d 00 60 ad 11 2e 4d <- 0x1818 | 0x2a63 | CPS[ CD(75.29) PW(93) ]"
> > N 584830 BLE_Server: "44 02 1c 14 96 00 5d 00 97 00 00 00 00 00 -> 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(151) CD(75.00) PW(93) SD(148.00) ]"
> > N 584834 BLE_Server: "23 00 5d 00 00 f2 02 56 af -> 0x1826 | 0x2ad2 | CPS(CPM)[ CD(75.00) PW(93) ]"
> > N 584837 BLE_Server: "00 97 -> 0x180d | 0x2a37 | HRS(HRM)[ HR(151) ]"
> > N 585257 ERG_Mode_CSV: "9;335.00;427.50;130;130;114;93;72;75"
> > N 585540 BLE_Common: "10 96 9a 01 95 01 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(150) ]"

I guess we should change that LOGW to a regular LOG so that we can get more information (The actual reported driver temperature is logged there). Do you think there's a chance that your driver maybe did get hot since you've not seen this message before?

MarkusSchneider, 2022-01-29:
> I guess we should change that LOGW to a regular LOG so that we can get more information (The actual reported driver temperature is logged there). Do you think there's a chance that your driver maybe [...]

Maybe. Did a workout with ERG for around 1h.

holger8080, 2022-01-29:
> I guess we should change that LOGW to a regular LOG so that we can get more information (The actual reported driver temperature is logged there). Do you think there's a chance that your driver maybe [...]

Thanks so much for your fast response! 
A few more details:
- I got the overheat message a few seconds after I switched the smartspin2k on. Consistently, when I press Shift up or down, the knob makes some small move and the "temperature now under control" message appears.
- I also had problems 30 mins in 
- the torque is so low (at any(!) setting of stepper motor power) that I can easily stop the knob with one single finger; with the old firmware there was plenty of power even at tow torque settings

@MarkusSchneider: Shifter messages do appear:
> N 74644 Main: "Shift UP: "
> N 77068 Main: "Temperature is now under control. Driver current reset."
> N 81331 ERG_Mode: "No data found in powertable."
> N 81340 ERG_Mode: "No data found in powertable."

I did try to merge the "case FitnessMachineControlPointProcedure::Reset:" case to the old firmware, but I keep getting some error messages while building (some platformio build issues) and didn't have the time to get it fixed.

MarkusSchneider, 2022-01-29:
@holger8080 I can try to compile for you what version is the one that worked for you? I can not find a version 1.16.
Please see here: https://github.com/doudar/SmartSpin2k/tags for a list of release versions.

holger8080, 2022-01-29:
> 

I 

> @holger8080 I can try to compile for you a new binary. What version is the one that worked for you? I can not find a version 1.16. Please see here: https://github.com/doudar/SmartSpin2k/tags for a list of release versions.

I meant this one here:
https://github.com/doudar/SmartSpin2k/releases/tag/1.6.19

😬 almost "1.16" 😅, sorry for that. And I'd really appreciate your help! :-)

MarkusSchneider, 2022-01-30:
@holger8080 Merged Reset Command into BLE Server and created a new Binary. I've removed automatic firmware update to avoid update with new version. You have to do this manually.

To update go to http://smartspin2k.local/ -> Update Firmware upload firmware.bin
I've tested connection with Zwift with erg and it was working. 

[firmware.zip](https://github.com/doudar/SmartSpin2k/files/7965747/firmware.zip)
[spiffs.zip](https://github.com/doudar/SmartSpin2k/files/7965748/spiffs.zip)

bdattilo, 2022-01-31:
@MarkusSchneider I have this same issue - what is the username and password on the ESP32 login page?

Also - have the same issue as this when flashing via the SmartSpin2kFlasher: https://github.com/doudar/SmartSpin2k/discussions/306

doudar, 2022-01-31:
> @MarkusSchneider I have this same issue - what is the username and password on the ESP32 login page?
> 
> 
> 
> Also - have the same issue as this when flashing via the SmartSpin2kFlasher: https://github.com/doudar/SmartSpin2k/discussions/306

admin/admin .

Not saving is most likely due to a full spiffs partition. Try a reset. You'll also have to manually upload all of the files  one at a time from your versions /data folder. You cannot upload spiffs.bin yet.

closed 2022-09-06 by doudar