doudar/SmartSpin2k #318: [BUG] Driver Overtemp not logged (only reset)

doudar, 2022-01-29:
### Description

Driver current reset message is displayed with no logging of the overtemp that caused it. 

### Steps to reproduce

**Expected behavior**
Driver overtemp should be on the same logging level as the current reset.

Fix is easy and needed here:
https://github.com/doudar/SmartSpin2k/blob/a6220ed6808148208abfc6ff671c702c26b77718/src/Main.cpp#L372

Created this issue so that it isn't forgotten, as we're working on a few other things currently and don't want to forget.

eMadman, 2022-01-29:
I'll do my next handful of rides using BigTreeTech TMC2209 V1.2's 

As discussed... there's a flaw with the Fysetc TMC2209 v3.0 boards and they don't support UART.  Info here: https://theres-probably-a-better-way-to-do-it.readthedocs.io/en/latest/pages/electronics/skr_1.3_and_fysetc2209v3.html - also reported here: https://forum.fysetc.com/d/199-tmc2209-v3-0-uart

The modifications required can be a bit daunting.  I was hoping it was an edge case but it looks like they're still selling the things and for less cost than the updated revisions.  It still works, but without UART... for the cost, it's worth looking at cheaper drivers or shop around for a different brand - BigTreeTech has been good, Waterott's great

eMadman, 2022-02-07:
Tested a bunch of 2209's from BTT over the weekend for my build queue - each one got several minutes on the desk with shifting up/down - didn't experience an overtemp issue.  Perhaps its a 2208 thing?

PR #354 (2022-07-09): Power table bug fix
Fixes #318
closed 2022-07-09 by doudar