doudar/SmartSpin2k #320: [BUG] Doesn't automatically reconnect device after disconnect

doudar, 2022-01-31:
### Description

When a server disconnects due to intermittent signal, it isn't automatically reconnected.

### Steps to reproduce

1. Ride until a server disconnects
2. Watch the log and it doesn't reconnect until reconnect tries hits zero and it preforms a scan. 

**Expected behavior**
It should automatically reconnect. Probably on the first try

**Actual behavior**
It doesn't reconnect until it preforms a scan

### Additional Information

	20719	N 2733287 BLE_Client: "Disconnect Called"
	20720	N 2733288 BLE_Client: "This disconnected client Address 3c:61:05:30:2a:66"
	20721	N 2733289 BLE_Client: "Detected 0x1818 Disconnect"
	20722	N 2733289 BLE_Client: "Deregistered PM on Disconnect"
	20723	N 2733713 BLE_Client: "Initiating Server Connection"
	20724	N 2733714 BLE_Client: "trying to connect to PM"
	20725	N 2733714 BLE_Client: "Forming a connection to:  3c:61:05:30:2a:66"
	20726	N 2733715 BLE_Client: " - Created client"
	20727	N 2738720 BLE_Client: " - Connected to server"
	20728	N 2738721 BLE_Client: "Failed to find service: 0x1818"
	20729	N 2738721 BLE_Client: "disconnecting Client"
	20730	N 2739721 BLE_Client: "Initiating Server Connection"
	20731	N 2739722 BLE_Client: "trying to connect to PM"
	20732	N 2739722 BLE_Client: "Reusing Client"
	20733	N 2739722 BLE_Client: "Client RSSI 0"
	20734	N 2739722 BLE_Client: "device RSSI -88"
	20735	N 2744828 BLE_Client: "Reconnect failed "
	20736	N 2744828 BLE_Client: "8 left."
	20737	N 2745829 BLE_Client: "Initiating Server Connection"
	20738	N 2745830 BLE_Client: "trying to connect to PM"
	20739	N 2745830 BLE_Client: "Reusing Client"
	20740	N 2745830 BLE_Client: "Client RSSI 0"
	20741	N 2745830 BLE_Client: "device RSSI -88"
	20742	N 2750935 BLE_Client: "Reconnect failed "
	20743	N 2750935 BLE_Client: "7 left."

If @h2zero could give my client code a look to see why it isn't reconnecting I'd love it! (Thank you in advance!)

My guess is something isn't getting done right prior to the "failed to find service" message. That server should always have that service. 

https://github.com/doudar/SmartSpin2k/blob/a6220ed6808148208abfc6ff671c702c26b77718/src/BLE_Client.cpp#L216-L220

h2zero, 2022-01-31:
>My guess is something isn't getting done right prior to the "failed to find service" message. That server should always have that service.

I think the issue regarding this is that you do not check if the connection was successful here:
https://github.com/doudar/SmartSpin2k/blob/a6220ed6808148208abfc6ff671c702c26b77718/src/BLE_Client.cpp#L212

As for why the connection is failing it's hard to know, I would need to see the NimBLE logs for the error code.

doudar, 2022-01-31:
> I think the issue regarding this is that you do not check if the connection was successful here: [...]

Good point. It's not getting the service most likely because the connection didn't complete...

I'll see if I can get some nimble logs tomorrow. It's a bit tougher to get those because I can't ride easily with a serial connection so it will take a bit of setup time.

MarkusSchneider, 2022-01-31:
Is it possible to collect logs of library by intercepting esp_log with [esp_log_set_vprintf](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/log.html#_CPPv419esp_log_set_vprintf14vprintf_like_t) ?

h2zero, 2022-01-31:
I don't think that will work as NimBLE doesn't use the ESP_LOGx macros.

doudar, 2022-01-31:
> I don't think that will work as NimBLE doesn't use the ESP_LOGx macros.

You should write a NimBLE, BLE logging service 😅

J/K,the more I think about it, I think I'll be able to get the logs via bench mounted esp32 and just route the BLE traffic through it. 

Looking at the logs further, it appears that NimBLE stopped communicating entirely (both server and client) for 10 seconds prior to the initial disconnect and then when it found the powermeter on the subsequent scan, the address was different which is probably why it couldn't reconnect or reuse the existing client.

doudar, 2022-02-01:
@h2zero , I've collected a few more logs. What really seems to be causing problems is that when one remote server times out and disconnects, all other remote servers get disconnected as well. These logs are generated from this branch:

https://github.com/doudar/SmartSpin2k/blob/5dda29d17f2c6987f152f8d70609b794b397bf08/src/BLE_Client.cpp#L73

[ 18316][E](BLE_Client): Initiating Server Connection
[ 18317][E](BLE_Client): Connecting device 0 Address a0:9e:1a:58:d0:27
[ 18317][E](BLE_Client): Trying to connect to HRM
[ 18320][E](BLE_Client): Reusing Client
[ 18320][E](BLE_Client): No Previous client found
[ 18320][E](BLE_Client): Forming a connection to:  a0:9e:1a:58:d0:27
[ 18323][E](BLE_Client):  - Created client
I NimBLEClient: "Connected event"
I NimBLEClient: "mtu update event; conn_handle=0 mtu=232"
I NimBLEClient: "Connection established"
[ 22270][E](BLE_Client):  - Connected to server
[ 27268][E](BLE_Client):  - Found service: 0x180d
[ 32269][E](BLE_Client):  - Found Characteristic: 0x2a37
[ 36272][E](BLE_Client): Successful 0x2a37 subscription.
[ 36273][E](BLE_Client): Registered HRM on Connect
[ 36273][E](BLE_Client): We are now connected to the BLE Server.
[ 36276][E](BLE_Client): Initiating Server Connection
[ 36276][E](BLE_Client): Connecting device 1 Address 3c:61:05:30:2a:66
[ 36276][E](BLE_Client): trying to connect to PM
[ 36279][E](BLE_Client): Reusing Client
[ 36279][E](BLE_Client): No Previous client found
[ 36279][E](BLE_Client): Forming a connection to:  3c:61:05:30:2a:66
[ 36282][E](BLE_Client):  - Created client
I NimBLEClient: "Connected event"
[ 39910][E](BLE_Common): 06 00 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]
[ 39911][E](BLE_Common): 06 00 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]
[ 40611][E](BLE_Common): 06 00 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]
[ 41312][E](BLE_Common): 06 00 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "
[ 42309][E](BLE_Client):  - Connected to server
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
[ 42309][E](BLE_Client): Failed to find service: 0x1818
[ 42714][E](BLE_Common): 06 00 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]
[ 42814][E](ERG_Mode): No data found in powertable.
[ 42815][E](ERG_Mode): No data found in powertable.
[ 43312][E](BLE_Client): Initiating Server Connection
[ 43313][E](BLE_Client): Connecting device 1 Address 3c:61:05:30:2a:66
[ 43313][E](BLE_Client): trying to connect to PM
[ 43313][E](BLE_Client): Reusing Client
[ 43414][E](BLE_Common): 06 00 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]
scan_evt timeout
I NimBLEClient: "Connected event"
[ 44816][E](BLE_Common): 06 00 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]
[ 45516][E](BLE_Common): 06 00 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]
[ 46917][E](BLE_Common): 06 00 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]
[ 47618][E](BLE_Common): 06 00 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]
[ 48320][E](BLE_Common): 06 00 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
E NimBLEClient: "Connection failed; status=574 "
[ 49710][E](BLE_Client): Reconnect failed
[ 49711][E](BLE_Client): 8 left.
[ 49720][E](BLE_Common): 06 00 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]
[ 50421][E](BLE_Common): 06 00 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]
[ 50711][E](BLE_Client): Initiating Server Connection
[ 50711][E](BLE_Client): Connecting device 1 Address 3c:61:05:30:2a:66
[ 50712][E](BLE_Client): trying to connect to PM
[ 50714][E](BLE_Client): Reusing Client
I NimBLEClient: "Connected event"
[ 51823][E](BLE_Common): 06 00 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]
[ 52523][E](BLE_Common): 06 00 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]
I NimBLEClient: "mtu update event; conn_handle=1 mtu=255"
I NimBLEClient: "Connection established"
[ 53273][E](BLE_Client): Reconnecting client
[ 53925][E](BLE_Common): 06 00 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]
E NimBLEClient: "characteristicDiscCB() rc=7 "
E NimBLEClient: "Could not retrieve services"
CRIT NimBLEClient: "Disconnect - host reset, rc=19"
CRIT NimBLEDevice: "Resetting s[tate; reason=19, "
I NimBLEClient: "disconnect; reason=19, "
 55286][E](BLE_Client): Couldn't find Service
[ 55290][E](BLE_Client): Disconnect Called
[ 55291][E](BLE_Client): This disconnected client Address 3c:61:05:30:2a:66
[ 55292][E](BLE_Client): Detected 0x0000 Disconnect
CRIT NimBLEClient: "Disconnect - host reset, rc=19"
I NimBLEClient: "disconnect; reason=19, "
[ 55295][E](BLE_Client): Disconnect Called
[ 55295][E](BLE_Client): This disconnected client Address a0:9e:1a:58:d0:27
[ 55298][E](BLE_Client): Detected 0x180d Disconnect
[ 55298][E](BLE_Client): Deregistered HR on Disconnect
ram_lld_pdu_tx_flush HCI packet count mismatch (2, 3)
I NimBLEDevice: "NimBle host synced."
[ 55325][E](BLE_Common): Starting Advertising From Communication Loop
E NimBLEAdvertising: "Advertisement data too long"
[ 56291][E](BLE_Client): Initiating Server Connection
[ 56292][E](BLE_Client): Connecting device 0 Address a0:9e:1a:58:d0:27
[ 56292][E](BLE_Client): Trying to connect to HRM
[ 56292][E](BLE_Client): Reusing Client
E NimBLEClient: "Connection failed; status=13 "
[ 66401][E](BLE_Client): Reconnect failed
[ 66402][E](BLE_Client): 6 left.
[ 66402][E](BLE_Client): Initiating Server Connection
[ 66404][E](BLE_Client): Connecting device 0 Address a0:9e:1a:58:d0:27
[ 66405][E](BLE_Client): Trying to connect to HRM
[ 66405][E](BLE_Client): Reusing Client
I NimBLEClient: "Connected event"
E NimBLEClient: "Connect timeout - no response"
[ 77516][E](BLE_Client): Reconnect failed
[ 77516][E](BLE_Client): 5 left.

h2zero, 2022-02-02:
The interesting bit is this:

> I NimBLEClient: "Connection established"
[ 53273]E: Reconnecting client
[ 53925]E: 06 00 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]
E NimBLEClient: "characteristicDiscCB() rc=7 "
E NimBLEClient: "Could not retrieve services"
CRIT NimBLEClient: "Disconnect - host reset, rc=19"

It looks like the controller is too busy to handle the command and times out. Maybe too much going on at once or interrupted for wifi stuff? Hard to say for sure. I'll have a better look later and see if I can think of anything.

At least the host resets are being handled well lol.
That's the reason everything gets disconnected, the stack is being reset there.

doudar, 2022-02-02:
> The interesting bit is this:
>
>
> It looks like the controller is too busy to handle the command and times out. Maybe too much going on at once or interrupted for wifi stuff? Hard to say for sure. I'll [...]

Very weirdly if the device labeled "PM" in the above log connects first, followed by the device "HRM", it connects okay.

h2zero, 2022-02-02:
Interesting, I don't know why that would produce a different result. The only thing that could be different between connections is the MTU, connection parameters, and security. 

If the server device requests connection parameter changes does your code accept or reject them?

That request could come late and if accepted it could cause some timing trouble.

doudar, 2022-02-02:
> Interesting, I don't know why that would produce a different result. The only thing that could be different between connections is the MTU, connection parameters, and security.
>
> If the server device [...]

My client code doesn’t have hooks for remote connection parameter requests.

h2zero, 2022-02-02:
>Would a higher logging level help?

Wouldn't hurt, not sure if any debug level logs would show much though.

I noticed this now that I looked at it better:
```
I NimBLEClient: "Connection established"
[ 53273]E: Reconnecting client
[ 53925]E: 06 00 <- 0x180d | 0x2a37 | HRS(HRM)[ HR(0) ]   << is this a notification?
E NimBLEClient: "characteristicDiscCB() rc=7 "
```

If that is a notification from the first device, it occurs while trying to discover the second device and right after that we have the error. Possible hang up from the notification handler?

doudar, 2022-02-02:
Half way solved...

When the disconnect callback is wrapped into another class like:
https://github.com/doudar/SmartSpin2k/blob/7d2975d10010b5bb7e704254f0cb9b8e215bd6cb/include/BLE_Common.h#L184-L228

A disconnect of any device disconnects all devices. 

Removing the callbacks from that class solved the multiple disconnect issue.

doudar, 2022-02-05:
@h2zero, here's the enhanced logging. If it connects to the heart rate monitor first, it triggers a disconnect on both devices when it tries to connect to the power meter. If it connects to the power meter first, It connects successfully on both devices. The only significant differences between the two is that the PM contains more services than the HR, and also contains the same service as the HR(but it isn't used in the program). I've also tried to add vTaskDelays to quiet down the BLE traffic during a connection but it doesn't seem to help. 

Here are a couple logs:
https://pastebin.com/yR0yWHt5
https://pastebin.com/YHAZ3uuW

And to reply to your post above, yes, it's a notification. WiFi usage was minimal during these tests. No logging was being sent out and I didn't try to load any webpages. 

Thanks!!!!!

h2zero, 2022-02-05:
Had a brief look at the logs. Looks like your notification callback may be holding up the stack. That's just my initial and previous assessment. If a notification arrives while processing the connection to another device I can see a scenario where the stack is waiting for the notification callback to return as well as all the connection stuff to complete creating a race condition that stalls the stack. 

That's just my initial thoughts, I don't know what all is going on in your callbacks but it seems they may be taking too much time? 

Very difficult for me to reproduce, so I can only speculate. What I would try, for verification, is to set a flag when connection is in progress and ignore notifications while it's set and see what happens.

doudar, 2022-02-05:
I’ll completely comment out notification subscriptions and see if that solves the issue. If it does, I’ll add a check to unsubscribe from notifications while a connection is in process.

doudar, 2022-02-05:
Also, I'm getting a exception was unhandled somewhat routinely on line 100 of the following code:

https://github.com/doudar/SmartSpin2k/blob/5d1dff6c7528659a63706e0d336329cadaa5141e/src/BLE_Client.cpp#L95-L101

PC: 0x400dfe38: NimBLEAdvertisedDevice::findAdvField(unsigned char, unsigned char, unsigned char*) at .pio/libdeps/release/NimBLE-Arduino/src/NimBLEAdvertisedDevice.cpp line 599
EXCVADDR: 0x000000cb

Decoding stack results

0x400dfe35: NimBLEAdvertisedDevice::findAdvField(unsigned char, unsigned char, unsigned char*) at .pio/libdeps/release/NimBLE-Arduino/src/NimBLEAdvertisedDevice.cpp line 592

0x400dff8b: NimBLEAdvertisedDevice::getServiceUUIDCount() at .pio/libdeps/release/NimBLE Arduino/src/NimBLEAdvertisedDevice.cpp line 439

0x400dffe9: NimBLEAdvertisedDevice::haveServiceUUID() at .pio/libdeps/release/NimBLE-Arduino/src/NimBLEAdvertisedDevice.cpp line 573

0x400d3fa3: SpinBLEClient::connectToServer() at src/BLE_Client.cpp line 100

0x400d5b91: bleClientTask(void*) at src/BLE_Client.cpp line 64

FWIW, this is with a testing branch and framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.2 and https://github.com/h2zero/NimBLE-Arduino/archive/refs/tags/1.3.4.zip

h2zero, 2022-02-05:
Something interesting I see in the log now:
```
D NimBLEClient: ">> getService: uuid: 0x1818"  **<<  First call**
D NimBLEClient: ">> retrieveServices"
[ 86171][D](BLE_Common): Address: (3c:61:05:30:2a:66) Client ID: (0) SerUUID: (0x1818) CharUUID: (0x2a63) HRM: (false) PM: (false) CSC: (false) CT: (false) doConnect: (true)
[ 86174][D](BLE_Common): Address: (a0:9e:1a:58:d0:27) Client ID: (1) SerUUID: (0x180d) CharUUID: (0x2a37) HRM: (false) PM: (false) CSC: (false) CT: (false) doConnect: (false)
D NimBLEClient: ">> getService: uuid: 0x180d"  **<<  Second call but first has not returned **
D NimBLEClient: "<< getService: found the service with uuid: 0x180d"
D NimBLERemoteService: ">> getCharacteristic: uuid: 0x2a37"
D NimBLERemoteService: "<< getCharacteristic: found the characteristic with uuid: 0x2a37"
[ 86881][D](BLE_Common): Address: (3c:61:05:30:2a:66) Client ID: (0) SerUUID: (0x1818) CharUUID: (0x2a63) HRM: (false) PM: (false) CSC: (false) CT: (false) doConnect: (true)
[ 86884][D](BLE_Common): Address: (a0:9e:1a:58:d0:27) Client ID: (1) SerUUID: (0x180d) CharUUID: (0x2a37) HRM: (false) PM: (false) CSC: (false) CT: (false) doConnect: (false)
D NimBLEClient: ">> getService: uuid: 0x180d"
D NimBLEClient: "<< getService: found the service with uuid: 0x180d"
D NimBLERemoteService: ">> getCharacteristic: uuid: 0x2a37"
D NimBLERemoteService: "<< getCharacteristic: found the characteristic with uuid: 0x2a37"
[ 87591][D](BLE_Common): Address: (3c:61:05:30:2a:66) Client ID: (0) SerUUID: (0x1818) CharUUID: (0x2a63) HRM: (false) PM: (false) CSC: (false) CT: (false) doConnect: (true)
[ 87594][D](BLE_Common): Address: (a0:9e:1a:58:d0:27) Client ID: (1) SerUUID: (0x180d) CharUUID: (0x2a37) HRM: (false) PM: (false) CSC: (false) CT: (false) doConnect: (false)
D NimBLEClient: ">> getService: uuid: 0x180d"
D NimBLEClient: "<< getService: found the service with uuid: 0x180d"
D NimBLERemoteService: ">> getCharacteristic: uuid: 0x2a37"
D NimBLERemoteService: "<< getCharacteristic: found the characteristic with uuid: 0x2a37"
D NimBLEClient: "Got Client event "
D NimBLEClient: "Notify Recieved for handle: 40"
D NimBLEClient: "checking service 0x180d for handle: 40"
D NimBLEClient: "Got Notification for characteristic Characteristic: uuid: 0x2a37, handle: 40 0x0028, props:  0x10
Descriptor: uuid: 0x2902, handle: 41"
D NimBLEClient: "Invoking callback for notification on characteristic Characteristic: uuid: 0x2a37, handle: 40 0x0028, props:  0x10
Descriptor: uuid: 0x2902, handle: 41"
D NimBLEClient: "Got Client event "
D NimBLEClient: "Notify Recieved for handle: 40"
D NimBLEClient: "checking service 0x180d for handle: 40"
D NimBLEClient: "Got Notification for characteristic Characteristic: uuid: 0x2a37, handle: 40 0x0028, props:  0x10
Descriptor: uuid: 0x2902, handle: 41"
D NimBLEClient: "Invoking callback for notification on characteristic Characteristic: uuid: 0x2a37, handle: 40 0x0028, props:  0x10
Descriptor: uuid: 0x2902, handle: 41"
D NimBLEClient: "Service Discovered >> status: 7 handle: -1" ** << First call now returns**
E NimBLEClient: "characteristicDiscCB() rc=7 "
D NimBLEClient: "<< << Service Discovered"
E NimBLEClient: "Could not retrieve services"
D NimBLEClient: "Got Client event "
D NimBLEClient: "<< getService: not found"
CRIT NimBLEClient: "Disconnect - host reset, rc=19"  ** << host/controller out of sync now**
CRIT NimBLEDevice: "Resetting state; reason=19, "
I NimBLEClient: "disconnect; reason=19, "
```

This appears to be a repeat of the FreeRTOS bug that causes task wait calls to not block... grrr
Do you use MDNS? if so, for testing please try disabling that and see what happens.

doudar, 2022-02-05:
> Something interesting I see in the log now:
>
>
> This appears to be a repeat of the FreeRTOS bug that causes task wait calls to not block... grrr Do you use MDNS? if so, for testing please try [...]

Yes, using MDNS as well. I'll try it. Thanks!

doudar, 2022-02-05:
Disabling MDNS (commented out everything including the library) didn't seem to make any difference. I think this is all new since I updated to ESP32 Arduino 2.0.0+ (idf 4.4).

In my testing I also just got another (seemingly random) exception unhandled on 
if(myDevice != nullptr){ myDevice->haveServiceUUID(); } 

Here's the log with MDNS turned off:
https://pastebin.com/6gG2C7vA

doudar, 2022-02-05:
To add, I wonder if maybe the getServices() has more to do with it as well. It checking services seems to be the point that starts the cascade:

(BLE_Client):  - Connected to server
D NimBLEClient: ">> getService: uuid: 0x1818"
D NimBLEClient: ">> retrieveServices"
 NimBLEClient: "Disconnected, could not retrieve services -aborting"
D NimBLEClient: "<< getService: not found"
(BLE_Client): Disconnect Called
(BLE_Client): Couldn't find Service
(BLE_Client): This disconnected client Address 3c:61:05:30:2a:66
(BLE_Client): Detected 0x0000 Disconnect
D NimBLEClient: "Got Client event "
CRIT NimBLEClient: "Disconnect - host reset, rc=19"
I NimBLEClient: "disconnect; reason=19, "
(BLE_Client): Disconnect Called

h2zero, 2022-02-05:
Something is really out of sync here, the logs are a real mess. 

>D NimBLERemoteService: "<< getCharacteristic: found the characteristic with uuid: 0x2a37"
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
D NimBLEClient: "Got Client event "
E NimBLEClient: "Connection failed; status=574 "
[ 29612][E](BLE_Client):  - Connected to server
D NimBLEClient: ">> getService: uuid: 0x1818"
D NimBLEClient: ">> retrieveServices"
E NimBLEClient: "Disconnected, could not retrieve services -aborting"
D NimBLEClient: "<< getService: not found"
[ 29616][E](BLE_Client): Failed to find service: 0x1818
[ 29618][E](BLE_Client): disconnecting Client
[ 30230][D](BLE_Common): Address: (a0:9e:1a:58:d0:27) Client ID: (0) SerUUID: (0x180d) CharUUID: (0x2a37) HRM: (false) PM: (false) CSC: (false) CT: (false) doConnect: (false)
D NimBLEClient: ">> getService: uuid: 0x180d"
D NimBLEClient: "<< getService: found the service with uuid: 0x180d"
D NimBLERemoteService: ">> getCharacteristic: uuid: 0x2a37"
D NimBLERemoteService: "<< getCharacteristic: found the characteristic with uuid: 0x2a37"
[ 30618][E](BLE_Client): Initiating Server Connection

That doesn't look right.

doudar, 2022-02-05:
> Something is really out of sync here, the logs are a real mess.
> 
> > D NimBLERemoteService: "<< getCharacteristic: found the characteristic with uuid: 0x2a37"
> > lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
> > D NimBLEClient: "Got Client event "
> > E NimBLEClient: "Connection failed; status=574 "
> > [ 29612][E](BLE_Client):  - Connected to server
> > D NimBLEClient: ">> getService: uuid: 0x1818"
> > D NimBLEClient: ">> retrieveServices"
> > E NimBLEClient: "Disconnected, could not retrieve services -aborting"
> > D NimBLEClient: "<< getService: not found"
> > [ 29616][E](BLE_Client): Failed to find service: 0x1818
> > [ 29618][E](BLE_Client): disconnecting Client
> > [ 30230][D](BLE_Common): Address: (a0:9e:1a:58:d0:27) Client ID: (0) SerUUID: (0x180d) CharUUID: (0x2a37) HRM: (false) PM: (false) CSC: (false) CT: (false) doConnect: (false)
> > D NimBLEClient: ">> getService: uuid: 0x180d"
> > D NimBLEClient: "<< getService: found the service with uuid: 0x180d"
> > D NimBLERemoteService: ">> getCharacteristic: uuid: 0x2a37"
> > D NimBLERemoteService: "<< getCharacteristic: found the characteristic with uuid: 0x2a37"
> > [ 30618][E](BLE_Client): Initiating Server Connection
> 
> That doesn't look right.
> 
> > I think this is all new since I updated to ESP32 Arduino 2.0.0+ (idf 4.4).
> 
> Yep, bugs galore. I don't recommend using it unless necessary. I have found and reported 2 in IDF that have gone unfixed so far.

Some of the confusion in the logs is that there is another task asking NimBLE for services with specific characteristic ID's (the task that unloads the buffer). For troubleshooting, I put a huge vTaskDelay into that task as well (to make sure it wasn't causing interruptions), and it didn't help anything. 

Re:4.4, yes, I may have to roll back :( 

(4.4 has so much better of a memory footprint however...)

h2zero, 2022-02-05:
I don't have a lot of suggestions left but here is one:

https://github.com/doudar/SmartSpin2k/blob/5dda29d17f2c6987f152f8d70609b794b397bf08/src/BLE_Client.cpp#L36

Try changing that to 0

h2zero, 2022-02-05:
I don't have a lot of suggestions left at this point but here is one:

https://github.com/doudar/SmartSpin2k/blob/5dda29d17f2c6987f152f8d70609b794b397bf08/src/BLE_Client.cpp#L38

Try changing that to 0

doudar, 2022-02-06:
I set all my tasks to core 0 for kicks and, bummer. same/same. And it's so weird. If the PM connects first everything works correctly until the PM disconnects, then the NimBLE host resets and disconnects both remote servers.  

[ 31849][E](BLE_Client): Successful 0x2a37 subscription.
[ 31853][E](BLE_Client): Registered HRM on Connect
[ 31853][E](BLE_Client): We are now connected to the BLE Server.
[ 36199][D](BLE_Common): Address: (a0:9e:1a:58:d0:27) Client ID: (0) SerUUID: (0x180d) CharUUID: (0x2a37) HRM: (false) PM: (false) CSC: (false) CT: (false) doConnect: (false)
D NimBLEClient: ">> getService: uuid: 0x180d"
D NimBLEClient: "<< getService: found the service with uuid: 0x180d"
D NimBLERemoteService: ">> getCharacteristic: uuid: 0x2a37"
D NimBLERemoteService: "<< getCharacteristic: found the characteristic with uuid: 0x2a37"
[ 36854][E](BLE_Client): Initiating Server Connection
[ 36856][E](BLE_Client): trying to connect to PM
[ 36856][E](BLE_Client): Forming a connection to:  3c:61:05:30:2a:66
[ 36856][E](BLE_Client):  - Created client
D NimBLEClient: ">> connect(3c:61:05:30:2a:66)"
D NimBLEClient: "Got Client event "
I NimBLEClient: "Connected event"
D NimBLEClient: "Got Client event "
I NimBLEClient: "mtu update event; conn_handle=1 mtu=255"
D NimBLEClient: "Got Client event "
D NimBLEClient: "Peer requesting to update connection parameters"
D NimBLEClient: "MinInterval: 20, MaxInterval: 100, Latency: 0, Timeout: 2000"
D NimBLEClientCallbacks: "onConnParamsUpdateRequest: default"
D NimBLEClient: "Accepted peer params"
I NimBLEClient: "Connection established"
D NimBLEClient: ">> deleteServices"
D NimBLEClient: "<< deleteServices"
D NimBLEClient: "<< connect()"
[ 40258][E](BLE_Client):  - Connected to server
D NimBLEClient: ">> getService: uuid: 0x1818"
D NimBLEClient: ">> retrieveServices"
D NimBLEClient: "Got Client event "
CRIT NimBLEClient: "Disconnect - host reset, rc=19"
CRIT NimBLEDevice: "Resetting state; reason=19, "
I NimBLEClient: "disconnect; reason=19, "
[ 42263][E](BLE_Client): Disconnect Called
[ 42264][E](BLE_ClientE NimBLEClient: "ble_gattc_disc_all_svcs: rc=7 "
D NimBLEClient: "<< getService: not found"
): This disconnected client Address 3c:61:05:30:2a:66
[ 42269][E](BLE_Client): Detected 0x0000 Disconnect
D NimBLEClient: "Got Client event "
CRIT NimBLEClient: "Disconnect - host reset, rc=19"
I NimBLEClient: "disconnect; reason=19, "
[ 42272][E](BLE_Client): Disconnect Called
[ 42275][E](BLE_Client): This disconnected client Address a0:9e:1a:58:d0:27
[ 42275][E](BLE_Client): Detected 0x180d Disconnect
[ 42275][E](BLE_Client): Deregistered HR on Disconnect
D NimBLEServer: ">> handleGapEvent: "
D NimBLEServer: "Advertising Complete"
[ 42278][E](BLE_Client): Failed to find service: 0x1818
[ 42289][E](BLE_Client): disconnecting Client
I NimBLEDevice: "NimBle host synced."
D NimBLEAdvertising: "Host re-synced"
D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
D NimBLEAdvertising: "<< Advertising start"

h2zero, 2022-02-06:
That's a much cleaner log.

>D NimBL[E](https://github.com/doudar/SmartSpin2k/issues/BLE_Client)Client: "<< connect()"
[ 40258][E](https://github.com/doudar/SmartSpin2k/issues/BLE_Client): - Connected to server
D NimBLEClient: ">> getService: uuid: 0x1818"
D NimBLEClient: ">> retrieveServices"
D NimBLEClient: "Got Client event "
CRIT NimBLEClient: "Disconnect - host reset, rc=19"
CRIT NimBLEDevice: "Resetting state; reason=19, "
I NimBLEClient: "disconnect; reason=19, "
[ 42263]E: Disconnect Called

Here I can see that the controller became non-responsive to the command. Basically we requested the controller to fetch the services from the device and it did not respond within the 2 seconds that the host waits for a command response, which triggers the reset. Why this happens, I am at a loss to explain. If this was a more common problem that I could devise a reproduction from I may be able to offer a solution, unfortunately this is not the case here. 

You said you have multiple tasks calling into the BLE stack? That could be the source of the issue, if more than 1 command is sent to the controller at nearly the same time, this issue could be a symptom of that (stack limitation). In this case a mutex may be required to block the other tasks while the connection task is working, or some other measure. 

Other than that, I'm all out of ammo to kill this bug. Maybe I'll wake up tomorrow with some ideas, I just hope it's not a controller bug (closed source).

doudar, 2022-02-06:
Okay, after going single threaded and a bunch of other things with no luck, I ended up line by line rebasing the connect code off of the NimBLE example and low and behold, it works!! 

Not exactly sure what was causing the issue as they were already pretty similar but 🤷‍♂️ and it works! 🥳

doudar, 2022-02-06:
fixed in #321

closed 2022-02-06 by doudar

h2zero, 2022-02-13:
>Also, I'm getting a exception was unhandled somewhat routinely on line 100 of the following code:

[SmartSpin2k/src/BLE_Client.cpp](https://github.com/doudar/SmartSpin2k/blob/5d1dff6c7528659a63706e0d336329cadaa5141e/src/BLE_Client.cpp#L95-L101)

Lines 95 to 101 in [5d1dff6](https://github.com/doudar/SmartSpin2k/commit/5d1dff6c7528659a63706e0d336329cadaa5141e)
 if (myDevice == nullptr) { 
   SS2K_LOG(BLE_CLIENT_LOG_TAG, "No Device Found to Connect"); 
   return false; 
 } 
 // FUTURE - Iterate through an array of UUID's we support instead of all the if checks. 
 if (myDevice->haveServiceUUID()) { 
   if (myDevice->isAdvertisingService(FLYWHEEL_UART_SERVICE_UUID) && (myDevice->getName() == FLYWHEEL_BLE_NAME)) { 

PC: 0x400dfe38: NimBLEAdvertisedDevice::findAdvField(unsigned char, unsigned char, unsigned char*) at .pio/libdeps/release/NimBLE-Arduino/src/NimBLEAdvertisedDevice.cpp line 599
EXCVADDR: 0x000000cb

Decoding stack results

0x400dfe35: NimBLEAdvertisedDevice::findAdvField(unsigned char, unsigned char, unsigned char*) at .pio/libdeps/release/NimBLE-Arduino/src/NimBLEAdvertisedDevice.cpp line 592

0x400dff8b: NimBLEAdvertisedDevice::getServiceUUIDCount() at .pio/libdeps/release/NimBLE Arduino/src/NimBLEAdvertisedDevice.cpp line 439

0x400dffe9: NimBLEAdvertisedDevice::haveServiceUUID() at .pio/libdeps/release/NimBLE-Arduino/src/NimBLEAdvertisedDevice.cpp line 573

0x400d3fa3: SpinBLEClient::connectToServer() at src/BLE_Client.cpp line 100

0x400d5b91: bleClientTask(void*) at src/BLE_Client.cpp line 64

Have you had any more occurrences of this? I've had reports of similar crashes but I can't reproduce it.

doudar, 2022-02-14:
I haven't seen it since I refactored my connect/reconnect code to the latest example. If you wanted to dig into the changes between the two, there lies your answer but again, I'm not certain what the difference was other than my reconnect and connect code wasn't consolidated.

h2zero, 2022-02-14:
Too many changes in there to pinpoint the original issue. At least it's working now 😄