doudar/SmartSpin2k #335: [BUG] Certificate error when checking for new firmware

eMadman, 2022-03-24:
### Description
User received a new SS2K with a factory reset applied.  Connected to home wifi.  Ends up in a loop, being asked to enter network information.

From the logs, it looks like an SSL certificate error:

### Steps to reproduce

1. Apply factory reset
2. Configure wifi
3. Restart SmartSpin2K

**Expected behavior**
Successful connection to web server, Spiffs and html files download, and device is ready to configure

**Actual behavior**
User is stuck trying to configure their wifi.

### Additional Information

[13:13:18][   222][E](Main): Setting up cpu Tasks
[13:13:18][   222][E](HTTP_Server): Connecting to: Skynet
[13:13:20][  1437][E](HTTP_Server): Waiting for connection to be established...
[13:13:21][  2438][E](HTTP_Server): Waiting for connection to be established...
[13:13:22][  3438][E](HTTP_Server): Waiting for connection to be established...
[13:13:22][  3453][E](HTTP_Server): Connected to Skynet IP address: 192.168.0.76
[13:13:22][  3453][E](HTTP_Server): Open http://smartspin2k.local/
[13:13:22][  3455][E](HTTP_Server): Syncing clock...
[13:13:22][  3456][E](HTTP_Server): Waiting for clock sync...

[13:13:26][  8157][E](HTTP_Server): Clock synced to: 1648127608
[13:13:26][  8160][E](HTTP_Server): Checking for newer firmware:
[13:13:27][  8432][E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():263]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
[13:13:27][  8433][E][WiFiClientSecure.cpp:135] connect(): start_ssl_client: -9984
[13:13:27][  8538][E](HTTP_Server): error downloading version.txt -1
[13:13:27][  8539][E](Main): Start BLE + ERG Tasks
[13:13:27][  8539][E](BLE_Setup): Starting Arduino BLE Client application...
[13:13:27][  8737][E](BLE_Server): Starting BLE Server

doudar, 2022-03-24:
Apply it to the code and give it a try. This could/should also be moved/duplicated and checked for updates in the spiffs partition as well. Good find @eMadman

eMadman, 2022-03-24:
@doudar - do you have any h ints on where I should find it in spiffs?  I only see it referenced in /include/cert.h

doudar, 2022-03-24:
> @doudar - do you have any h ints on where I should find it in spiffs? I only see it referenced in /include/cert.h
> 
> Edit: that might be the only place for it: From settings.h [image]
> 
> And the path to spiffs.bin is referenced in HTTP_Server_Basic.cpp [image]

@eMadman , you’ll find cert.h in /include

eMadman, 2022-03-25:
Still seems to throw an error when I try the new cert.  I've attached the PR
```
[  6772][E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():263]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
[  6774][E][WiFiClientSecure.cpp:135] connect(): start_ssl_client: -9984
[  6879][E](HTTP_Server): error downloading version.txt -1

```

eMadman, 2022-03-26:
Resolved by updating certs.  Latest firmware.bin attached or build from Develop.

[firmware.zip](https://github.com/doudar/SmartSpin2k/files/8355951/firmware.zip)

closed 2022-03-26 by eMadman