doudar/SmartSpin2k #339: [BUG] Disconnects in TrainerRoad

eMadman, 2022-04-11:
### Description
In trainer road, a few of us are seeing disconnects intermittently in rides.  Frequency seems to be hit or miss.  Seems to reconnect on its own

Log didn't indicate anything interesting last time I got this.  Just indicated a disconnect and attempted reconnect.  Will try another ride and capture the log.  I also need to get the exception decoder working on my laptop to see if there's anything worth seeing there.

Length of disconnects appears to be a few seconds.  I think it's running out of reconnect attempts and then starting over.  

Firmware version: Using lower-erg-wattage branch which has the changes from newest_esp32_arduino_master merged into it.

I've shared newest_esp32_arduino_master with a user that's riding with DFC and experiencing this as well.  There might be some fun data in the DFC logs...

jeffbg123, 2022-04-12:
I contacted DFC support and this is their response:

> While there's currently no way for end-users to access DFC logs it is something we have planned for a future release. DFC has a bunch of unused flash memory that  could be used for logs. As you note, that'll help debug issues like you're seeing.
> 
> Have you tried this beta release? It fixes dropout issues on certain devices like the base Fenix 6. We've only received reports of dropouts from a handful of users and this release has fixed that for them.
> https://github.com/intelligenate/dfc/releases/tag/power-cadence-beta.3

I will try their beta release with the ss2k beta and see if that fixes it.

Thanks!

jeffbg123, 2022-04-18:
With the DFC update and the SS2k update the issue has improved drastically. Instead of 10-20 disconnects per hour I have seen 1-2 per hour over the past 3 workouts.

closed 2022-09-06 by doudar