doudar/SmartSpin2k #340: [FR] Wahoo Dircon output.

doudar, 2022-04-14:
Wahoo developed a unit that connects to their trainers and provides BLE FTMS over IP. 

Most apps are now supporting it and Zwift has finally followed suit. 

The service is broadcast on MDNS like:
`MDNS	413	Standard query response 0x0000 PTR Wahoo KICKR 50A4._wahoo-fitness-tnp._tcp.local A, cache flush 192.168.86.80 SRV, cache flush 0 0 36866 Wahoo-KICKR-50A4.local TXT`
[image]
[image]

Then each supported BLE service is replicated on TCP port 36866:
[image]
[image]

The new Dircon branch has most of the advertising prototyped. Zwift is still not recognizing it but Zwift keeps requesting the MDNS repeatedly so it should be very close.  

@MarkusSchneider , this is really up your alley (generating the TCP ports) if you'd like to give it a try. Attached is a Wireshark log from one of the QZ users with filtered packets between the kicker/direct connect and Zwift. 

[Wireshark_dircon_zwift_windows2.pcapng.zip](https://github.com/doudar/SmartSpin2k/files/8492718/Wireshark_dircon_zwift_windows2.pcapng.zip)

Also you can see the QZ implementation at: https://github.com/cagnulein/qdomyos-zwift/issues/756
and 
https://github.com/cagnulein/qdomyos-zwift/blob/master/src/dirconprocessor.cpp

doudar, 2022-04-17:
I've been trying to get advertising working correctly all day I can't get it to show as a Dircon device in Zwift or any other apps . @cagnulein Would you be able to look at the Wireshark capture and see if you can spot any problems? At the end of the file there's a ping test, then a wrong port telnet, then the correct port telnet connection. 

[ss2k_dircon_advertising.zip](https://github.com/doudar/SmartSpin2k/files/8500925/ss2k_dircon_advertising.zip)

[image]

MarkusSchneider, 2022-04-17:
> Wahoo developed a unit that connects to their trainers and provides BLE FTMS over IP.
> 
> Most apps are now supporting it and Zwift has finally followed suit.
> 
> The service is broadcast on MDNS like: `MDNS 413 Standard query response 0x0000 PTR Wahoo KICKR 50A4._wahoo-fitness-tnp._tcp.local A, cache flush 192.168.86.80 SRV, cache flush 0 0 36866 Wahoo-KICKR-50A4.local TXT` [image] [image]
> 
> Then each supported BLE service is replicated on TCP port 36866:
> 
> [image] [image]
> 
> The new [Dircon](https://github.com/doudar/SmartSpin2k/tree/Dircon) branch has most of the advertising prototyped. Zwift is still not recognizing it but Zwift keeps requesting the MDNS repeatedly so it should be very close.
> 
> @MarkusSchneider , this is really up your alley (generating the TCP ports) if you'd like to give it a try. Attached is a Wireshark log from one of the QZ users with filtered packets between the kicker/direct connect and Zwift.
> 
> [Wireshark_dircon_zwift_windows2.pcapng.zip](https://github.com/doudar/SmartSpin2k/files/8492718/Wireshark_dircon_zwift_windows2.pcapng.zip)
> 
> Also you can see the QZ implementation at: [cagnulein/qdomyos-zwift#756](https://github.com/cagnulein/qdomyos-zwift/issues/756) and https://github.com/cagnulein/qdomyos-zwift/blob/master/src/dirconprocessor.cpp

Hi Anthony

MarkusSchneider, 2022-04-17:
Hi Anthony,
I'll try to get some time next week to create the TCP Stack.

Regards
Markus

cagnulein, 2022-04-17:
@doudar i checked your log but the answer from the SS2K is completely wrong, you have to copy this answer https://github.com/cagnulein/qdomyos-zwift/issues/756#issuecomment-1098818031 in particular SRV, A, TXT records

doudar, 2022-04-17:
> @doudar i checked your log but the answer from the SS2K is completely wrong, you have to copy this answer https://github.com/cagnulein/qdomyos-zwift/issues/756#issuecomment-1098818031 in particular SRV, A, TXT records

I've been working off that 😊

My suspicion is the SRV A as well but I'm having an issue getting the esp32 mdns library to put it in the "answers". It sends it, but you'll see it's in the "additional records".

I'm guessing I may have to modify the esp32 mdns library as well 😐

I had hoped to get away without this as I thought you had it working before you added that.

cagnulein, 2022-04-17:
> I'm guessing I may have to modify the esp32 mdns library as well

yes as I did for the QT one ;)

doudar, 2022-10-09:
I just tried working on again and I’m pretty sure the reason Zwift doesn’t attempt to connect to the socket is that the esp32 mdns library adds the “Type A” record to “additional records” instead of the main “Answers” field in response to a MDNS QU from Zwift. The unfortunate part is that I’m fairly certain this is done inside the closed source ESP-IDF, so I don’t have a way to modify it. I’ll submit an issue.

closed 2025-09-16 by doudar