doudar/SmartSpin2k #371: [BUG] Stepper motor stops turning after a period of time

eMadman, 2022-09-04:
### Description

While testing with simulator and on-bike with the power meter, I'm seeing the stepper motor stop responding mid-ride.  Seems to occur in sim mode and erg mode.  

Once it happens, the shifters stop working, though the log shows clicks

The 3rd line of the power table is interesting with the negative values... not sure what's going on there or if that's expected.

<HTML>
<BODY>
<TABLE><TR><TD></TD><TD>107970</TD><TD>[ 43402][E](FTMS_SERVER): 11 00 00 f2 01 28 33 -&gt; Sim Mode Incline 4.980000</TD></TR><TR><TD></TD><TD>107971</TD><TD>[ 43629][E](PowTab): | &nbsp;&nbsp;0| &nbsp;69| &nbsp;88| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0|</TD></TR><TR><TD></TD><TD>107972</TD><TD>[ 43630][E](PowTab): | &nbsp;&nbsp;0| &nbsp;90| &nbsp;85| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0|</TD></TR><TR><TD></TD><TD>107973</TD><TD>[ 43633][E](PowTab): | &nbsp;&nbsp;0|-243| 850| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0|</TD></TR><TR><TD></TD><TD>107974</TD><TD>[ 44109][E](BLE_Common): 00 8c &lt;- 0x180d | 0x2a37 | HRS(HRM)[ HR(140) ]</TD></TR><TR><TD></TD><TD>107975</TD><TD>[ 44115][E](BLE_Server): 44 02 a5 14 9a 00 6f 00 8c 00 00 00 00 00 -&gt; 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(140) CD(77.00) PW(111) SD(285.00) ]</TD></TR><TR><TD></TD><TD>107976</TD><TD>[ 44120][E](BLE_Server): 23 00 6f 00 00 2f 00 62 83 -&gt; 0x1826 | 0x2ad2 | CPS(CPM)[ CD(77.00) PW(111) ]</TD></TR><TR><TD></TD><TD>107977</TD><TD>[ 44121][E](BLE_Server): 00 8c -&gt; 0x180d | 0x2a37 | HRS(HRM)[ HR(140) ]</TD></TR><TR><TD></TD><TD>107978</TD><TD>[ 44824][E](BLE_Common): 2e 07 74 00 b3 d8 04 06 a4 41 &lt;- 0x1818 | 0x2a63 | CPS[ CD(78.37) PW(116) ]</TD></TR><TR><TD></TD><TD>107979</TD><TD>[ 44826][E](BLE_Server): 44 02 e9 14 9c 00 74 00 8c 00 00 00 00 00 -&gt; 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(140) CD(78.00) PW(116) SD(353.00) ]</TD></TR><TR><TD></TD><TD>107980</TD><TD>[ 44832][E](BLE_Server): 23 00 74 00 00 30 00 75 86 -&gt; 0x1826 | 0x2ad2 | CPS(CPM)[ CD(78.00) PW(116) ]</TD></TR><TR><TD></TD><TD>107981</TD><TD>[ 44835][E](BLE_Server): 00 8c -&gt; 0x180d | 0x2a37 | HRS(HRM)[ HR(140) ]</TD></TR><TR><TD></TD><TD>107982</TD><TD>[ 44836][E](FTMS_SERVER): 11 00 00 f1 01 28 33 -&gt; Sim Mode Incline 4.970000</TD></TR><TR><TD></TD><TD>107983</TD><TD>[ 45540][E](BLE_Common): 00 8c &lt;- 0x180d | 0x2a37 | HRS(HRM)[ HR(140) ]</TD></TR><TR><TD></TD><TD>107984</TD><TD>[ 45545][E](BLE_Server): 44 02 e9 14 9c 00 74 00 8c 00 00 00 00 00 -&gt; 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(140) CD(78.00) PW(116) SD(353.00) ]</TD></TR><TR><TD></TD><TD>107985</TD><TD>[ 45551][E](BLE_Server): 23 00 74 00 00 31 00 88 89 -&gt; 0x1826 | 0x2ad2 | CPS(CPM)[ CD(78.00) PW(116) ]</TD></TR><TR><TD></TD><TD>107986</TD><TD>[ 45553][E](BLE_Server): 00 8c -&gt; 0x180d | 0x2a37 | HRS(HRM)[ HR(140) ]</TD></TR><TR><TD></TD><TD>107987</TD><TD>[ 46255][E](BLE_Common): 2e 07 7d 00 93 dc 06 06 aa 47 &lt;- 0x1818 | 0x2a63 | CPS[ CD(79.69) PW(125) ]</TD></TR><TR><TD></TD><TD>107988</TD><TD>[ 46257][E](BLE_Common): 2e 07 7c 00 76 de 07 06 b0 4a &lt;- 0x1818 | 0x2a63 | CPS[ CD(79.38) PW(124) ]</TD></TR><TR><TD></TD><TD>107989</TD><TD>[ 46259][E](BLE_Common): 00 8c &lt;- 0x180d | 0x2a37 | HRS(HRM)[ HR(140) ]</TD></TR><TR><TD></TD><TD>107990</TD><TD>[ 46263][E](BLE_Server): 44 02 2e 15 9e 00 7c 00 8c 00 00 00 00 00 -&gt; 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(140) CD(79.00) PW(124) SD(422.00) ]</TD></TR><TR><TD></TD><TD>107991</TD><TD>[ 46267][E](BLE_Server): 23 00 7c 00 00 32 00 91 8c -&gt; 0x1826 | 0x2ad2 | CPS(CPM)[ CD(79.00) PW(124) ]</TD></TR><TR><TD></TD><TD>107992</TD><TD>[ 46271][E](BLE_Server): 00 8c -&gt; 0x180d | 0x2a37 | HRS(HRM)[ HR(140) ]</TD></TR><TR><TD></TD><TD>107993</TD><TD>[ 46272][E](FTMS_SERVER): 11 00 00 d6 01 28 33 -&gt; Sim Mode Incline 4.700000</TD></TR><TR><TD></TD><TD>107994</TD><TD>[ 46576][E](Main): Shift DOWN: </TD></TR><TR><TD></TD><TD>107995</TD><TD>[ 46977][E](BLE_Common): 00 8d &lt;- 0x180d | 0x2a37 | HRS(HRM)[ HR(141) ]</TD></TR><TR><TD></TD><TD>107996</TD><TD>[ 46985][E](BLE_Server): 44 02 2e 15 9e 00 7c 00 8d 00 00 00 00 00 -&gt; 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(141) CD(79.00) PW(124) SD(422.00) ]</TD></TR><TR><TD></TD><TD>107997</TD><TD>[ 46992][E](BLE_Server): 23 00 7c 00 00 33 00 9a 8f -&gt; 0x1826 | 0x2ad2 | CPS(CPM)[ CD(79.00) PW(124) ]</TD></TR><TR><TD></TD><TD>107998</TD><TD>[ 46995][E](BLE_Server): 00 8d -&gt; 0x180d | 0x2a37 | HRS(HRM)[ HR(141) ]</TD></TR><TR><TD></TD><TD>107999</TD><TD>[ 47136][E](PowTab): | &nbsp;&nbsp;0| &nbsp;69| &nbsp;92| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0|</TD></TR><TR><TD></TD><TD>108000</TD><TD>[ 47137][E](PowTab): | &nbsp;&nbsp;0| &nbsp;90| &nbsp;84| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0|</TD></TR><TR><TD></TD><TD>108001</TD><TD>[ 47142][E](PowTab): | &nbsp;&nbsp;0|-243| 936| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0|</TD></TR></TABLE>

</BODY>
</HTML>

doudar, 2022-09-04:
negative values shouldn't be an issue. Do you see anything like "min position set" in the logs?

eMadman, 2022-09-04:
It came up in the logs here:

<HTML>
<BODY>
<TABLE><TR><TD></TD><TD>115820</TD><TD>[1717919][E](ERG_Mode): PowerTable pairs [20][3]</TD></TR><TR><TD></TD><TD>115821</TD><TD>[1717920][E](ERG_Mode): Min Position Set: -1277</TD></TR><TR><TD></TD><TD>115822</TD><TD>[1717920][E](ERG_Mode): PowerTable pairs [20][3]</TD></TR><TR><TD></TD><TD>115823</TD><TD>[1717922][E](ERG_Mode): Max Position Set: -1276</TD></TR></TABLE>

</BODY>
</HTML>

Full log surrounding this:

<HTML>
<BODY>
<TABLE><TR><TD></TD><TD>115782</TD><TD>[1709494][E](FTMS_SERVER): 11 00 00 1f fe 28 33 -&gt; Sim Mode Incline -4.810000</TD></TR><TR><TD></TD><TD>115783</TD><TD>[1709495][E](PowTab): | &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;78| &nbsp;84| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0|</TD></TR><TR><TD></TD><TD>115784</TD><TD>[1709498][E](PowTab): | &nbsp;&nbsp;0| &nbsp;&nbsp;0|-363| 129| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0|</TD></TR><TR><TD></TD><TD>115785</TD><TD>[1710201][E](BLE_Common): 23 00 7c 00 00 82 00 f8 b6 &lt;- 0x1818 | 0x2a63 | CPS[ CD(87.03) PW(124) ]</TD></TR><TR><TD></TD><TD>115786</TD><TD>[1710205][E](BLE_Server): 44 02 53 17 ae 00 7c 00 00 00 00 00 00 00 -&gt; 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(87.00) PW(124) SD(971.00) ]</TD></TR><TR><TD></TD><TD>115787</TD><TD>[1710909][E](BLE_Common): 23 00 75 00 00 83 00 b2 b9 &lt;- 0x1818 | 0x2a63 | CPS[ CD(88.02) PW(117) ]</TD></TR><TR><TD></TD><TD>115788</TD><TD>[1710914][E](BLE_Server): 44 02 98 17 b0 00 75 00 00 00 00 00 00 00 -&gt; 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(88.00) PW(117) SD(40.00) ]</TD></TR><TR><TD></TD><TD>115789</TD><TD>[1710918][E](FTMS_SERVER): 11 00 00 fb fd 28 33 -&gt; Sim Mode Incline -5.170000</TD></TR><TR><TD></TD><TD>115790</TD><TD>[1711620][E](BLE_Common): 23 00 7b 00 00 84 00 74 bc &lt;- 0x1818 | 0x2a63 | CPS[ CD(87.03) PW(123) ]</TD></TR><TR><TD></TD><TD>115791</TD><TD>[1711621][E](BLE_Common): 23 00 7e 00 00 85 00 26 bf &lt;- 0x1818 | 0x2a63 | CPS[ CD(89.04) PW(126) ]</TD></TR><TR><TD></TD><TD>115792</TD><TD>[1711626][E](BLE_Server): 44 02 dc 17 b2 00 7e 00 00 00 00 00 00 00 -&gt; 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(89.00) PW(126) SD(108.00) ]</TD></TR><TR><TD></TD><TD>115793</TD><TD>[1712333][E](BLE_Server): 44 02 dc 17 b2 00 7e 00 00 00 00 00 00 00 -&gt; 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(89.00) PW(126) SD(108.00) ]</TD></TR><TR><TD></TD><TD>115794</TD><TD>[1712337][E](FTMS_SERVER): 11 00 00 cf fd 28 33 -&gt; Sim Mode Incline -5.610000</TD></TR><TR><TD></TD><TD>115795</TD><TD>[1713001][E](PowTab): | &nbsp;&nbsp;0| &nbsp;&nbsp;0| 111| 126| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0|</TD></TR><TR><TD></TD><TD>115796</TD><TD>[1713002][E](PowTab): | &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;78| &nbsp;84| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0|</TD></TR><TR><TD></TD><TD>115797</TD><TD>[1713005][E](PowTab): | &nbsp;&nbsp;0| &nbsp;&nbsp;0|-670| 129| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0| &nbsp;&nbsp;0|</TD></TR><TR><TD></TD><TD>115798</TD><TD>[1713041][E](BLE_Common): 23 00 7d 00 00 86 00 d8 c1 &lt;- 0x1818 | 0x2a63 | CPS[ CD(89.04) PW(125) ]</TD></TR><TR><TD></TD><TD>115799</TD><TD>[1713046][E](BLE_Server): 44 02 dc 17 b2 00 7d 00 00 00 00 00 00 00 -&gt; 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(89.00) PW(125) SD(108.00) ]</TD></TR><TR><TD></TD><TD>115800</TD><TD>[1713049][E](FTMS_SERVER): 11 00 00 b4 fd 28 33 -&gt; Sim Mode Incline -5.880000</TD></TR><TR><TD></TD><TD>115801</TD><TD>[1713753][E](BLE_Common): 23 00 7e 00 00 87 00 73 c4 &lt;- 0x1818 | 0x2a63 | CPS[ CD(92.11) PW(126) ]</TD></TR><TR><TD></TD><TD>115802</TD><TD>[1713756][E](BLE_Server): 44 02 aa 18 b8 00 7e 00 00 00 00 00 00 00 -&gt; 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(92.00) PW(126) SD(314.00) ]</TD></TR><TR><TD></TD><TD>115803</TD><TD>[1714460][E](BLE_Common): 23 00 7e 00 00 88 00 2d c7 &lt;- 0x1818 | 0x2a63 | CPS[ CD(88.02) PW(126) ]</TD></TR><TR><TD></TD><TD>115804</TD><TD>[1714464][E](BLE_Server): 44 02 98 17 b0 00 7e 00 00 00 00 00 00 00 -&gt; 0x1826 | 0x2ad2 | FTMS(IBD)[ HR(0) CD(88.00) PW(126) SD(40.00) ]</TD></TR><TR><TD></TD><TD>115805</TD><TD>[1714468][E](FTMS_SERVER): 11 00 00 be fd 28 33 -&gt; Sim Mode Incline -5.780000</TD></TR><TR><TD></TD><TD>115806</TD><TD>[1715172][E](BLE_Common): 23 00 7c 00 00 89 00 e7 c9 &lt;- 0x1818 | 0x2a63 | CPS[ CD(88.02) PW(124) ]</TD></TR><TR><TD></TD><TD>115807</TD><TD>[1715173][E](BLE_Common): 23 00 80 00 00 8a 00 99 cc &lt;- 0x1818 | 0x2a63 | CPS[ CD(89.04) PW(128) ]</TD></TR><TR><TD></TD><TD>115808</TD><TD>[1715177][E](BLE_Server): 44 02 dc 17 b2 00 80 00 00 00 00 00 00 00 -&gt; 0x [... cut ...] p;&nbsp;0|</TD></TR><TR><TD></TD><TD>115850</TD><TD>[1723543][E](PowTab): | &nbsp;&nbsp;&nbsp;0| &nbsp;&nbsp;&nbsp;0| -725|-1527| &nbsp;&nbsp;&nbsp;0| &nbsp;&nbsp;&nbsp;0| &nbsp;&nbsp;&nbsp;0| &nbsp;&nbsp;&nbsp;0| &nbsp;&nbsp;&nbsp;0| &nbsp;&nbsp;&nbsp;0| &nbsp;&nbsp;&nbsp;0| &nbsp;&nbsp;&nbsp;0| &nbsp;&nbsp;&nbsp;0| &nbsp;&nbsp;&nbsp;0| &nbsp;&nbsp;&nbsp;0| &nbsp;&nbsp;&nbsp;0| &nbsp;&nbsp;&nbsp;0| &nbsp;&nbsp;&nbsp;0| &nbsp;&nbsp;&nbsp;0| &nbsp;&nbsp;&nbsp;0|</TD></TR></TABLE>

</BODY>
</HTML>

doudar, 2022-09-04:
Problem Identified - Max position is less than Min position. It also looks like it's using crazy positions in the table as well.

eMadman, 2022-09-04:
Same thing happens with the min_brake_watts set to 0

<HTML>
<BODY>
<TABLE><TR><TD></TD><TD>119412</TD><TD>[325959][E](ERG_Mode): PowerTable pairs [20][3]</TD></TR><TR><TD></TD><TD>119413</TD><TD>[325961][E](ERG_Mode): Min Position Set: -739</TD></TR><TR><TD></TD><TD>119414</TD><TD>[325961][E](ERG_Mode): PowerTable pairs [20][3]</TD></TR><TR><TD></TD><TD>119415</TD><TD>[325962][E](ERG_Mode): Max Position Set: -738</TD></TR></TABLE>

</BODY>
</HTML>

doudar, 2022-09-06:
@eMadman , fixed by #372 ?

eMadman, 2022-09-06:
I believe so.  Haven't been able to reproduce it in a couple of tests on bike and on my desk

eMadman, 2022-09-15:
Re-opening this because I've been able to reproduce it somewhat consistently.  Fit file and log
[logging - shifter and motor not responding v2.txt](https://github.com/doudar/SmartSpin2k/files/9575585/logging.-.shifter.and.motor.not.responding.v2.txt)

Using RGT for testing with the Flanders map.  This fit file is what was used to reproduce it - you can also use the Manual BLE simulator.
[2022-01-05-14-23-26.zip](https://github.com/doudar/SmartSpin2k/files/9575602/2022-01-05-14-23-26.zip)

It looks like the motor stops turning in response to elevation or web shifter clicks.  Then occasionally, you'll see it turn the motor again.  

To avoid overtemp issues, Flanders map is relatively flat.  I have the sim multiplier at 5 and motor power at 1000ma

doudar, 2022-09-15:
Well, I see an obvious're using the  problem! 

You're using the software switch to reverse the stepper cables. This results in a reverse power table:
Watts         [676093][E](PowTab): |   0|   67|  87| 148| 196|   0|   0|   0|   0|   0|   0|   0|   0|   0|   0|   0|   0|   0|   0|   0|
CAD           [676094][E](PowTab): |   0|   77|  73|  81|   79|   0|   0|   0|   0|   0|   0|   0|   0|   0|   0|   0|   0|   0|   0|   0|
Resistance [676096][E](PowTab): |   0| 410|  69| -90|-313|   0|   0|   0|   0|   0|   0|   0|   0|   0|   0|   0|   0|   0|   0|   0|

The bottom row of the table *should* normally go small to big. Because of the switch, yours goes big to small. I'll see if I can rethink how the reverse switch is implemented to take this into account. 

In the meantime, for an immediate fix, you could reverse the stepper motor wires.

eMadman, 2022-09-15:
I have it at the default "On" position in the settings toggle

doudar, 2022-09-15:
Interesting....You don't happen to have a bike that the resistance gets higher when the knob turns counter-clockwise do you?

doudar, 2022-09-15:
After talking with @eMadman it appears that this is a flaw in our bench testing process. I'll work on developing a better tool for bench testing. 

This issue was because the testing was being done with the fit streamer in the developer tools, and auto min/max was placing software limits in weird places because knob position was completely de-coupled from watt readings due to the simple bench test. 

I plan to implement another javascript based testing tool which will output Wattages based on Cadence (from the fit file) and resistance received from the SS2K.

closed 2022-09-19 by doudar