doudar/SmartSpin2k #384: [BUG] (short description)

eMadman, 2022-10-13:
### Description
sim ride in Zwift off of BT power meter

After a period of time, it stops adjusting incline or responding to shifts.  It self-healed earlier in the ride (I think around #116597 in the log) and happened again towards the end but didn't recover (last minute or two of logs)

[Custom UDP Receiver (Port 10000).csv](https://github.com/doudar/SmartSpin2k/files/9780884/Custom.UDP.Receiver.Port.10000.csv)

doudar, 2022-10-15:
Confirmed. If max watts gets set lower than current position, it gets stuck until sim mode target position goes below current position. Should be an easy fix.

doudar, 2022-10-21:
Appears to be fixed in #385

closed 2022-11-29 by doudar