doudar/SmartSpin2k #390: Shifters in Stand-alone ERG mode

Flo100, 2022-10-23:
I just found your SmartSpin2k project, build it, adapted it to my already made (and older) hardware setup and I really like it. :)

**Is your feature request related to a problem? Please describe.**
There's not really a problem, it;'s more inconvenience.
Sometimes I use my spinning bike stand-alone and choose the ERG mode from the webserver and set ERG target Watts. But during my excercise I would like to change this target using the shifters, because using the webinterface is annoying during a workout.

**Describe the solution you'd like**
Let the shifters increase and decrease the ERG target Watts in ERG mode. For example in 10Watt steps or so.
Also it would be great that the Watts go to minimum Watts when the shift down button is held for 1,5s. That way you don't have to push the button that many times to release the resistance. (Could also work for gears in SIM mode to switch back faster).

**Describe alternatives you've considered**
Using my phone on the handlebar to change the slider in de webinterface.

I noticed that I can't switch on ERG mode when connected to RGT or Zwift. Would it be possible that the webinterface ERG setting overrides the incline data from RGT so I can ride any event on targeted Watts instead of incline (but still sending all cadence, hr and powerdata to RGT)?

doudar, 2022-10-24:
> I just found your SmartSpin2k project, build it, adapted it to my already made (and older) hardware setup and I really like it. :)
> 
**Awesome! Glad to have you here!!!**

> **Is your feature request related to a problem? Please describe.** There's not really a problem, it;'s more inconvenience. Sometimes I use my spinning bike stand-alone and choose the ERG mode from [...]

**Holding down the shifters - I like it! That's a pretty cool idea. I'll definitely put it on my todo list. Changing target watts in ERG mode is interesting as well. I think it's something that could be done.**

> **Describe alternatives you've considered** Using my phone on the handlebar to change the slider in de webinterface.
>
> I noticed that I can't switch on ERG mode when connected to RGT or Zwift. Would [...]
**This one might be a little bit harder to implement correctly, but I'll think about it and look into it. In the meantime with Zwift, you could work around this by only connecting SS2k as a power meter and not a controllable trainer. **

Flo100, 2022-10-24:
I made a simple version a few years ago with a stepper motor and an A4988 and it works like Smartspin2k when not connected to Zwift, so only changing the resistance by pressing the shifters. I quickly got tired of pushing multiple times to get a large increase or decrease and so I implemented big steps when holding down the button for 1,5s.

It also has a hardware switch and let the stepper "zero" itself when powering on. 
I noticed with SmartSpin2k that when I start on a hill the motor keeps spinning backwards until it is mechanical limited and crashes. (Unless I switch up at the same time offcourse)
How do I tell SmartSpin2k that the stepper is already at minimum steps when I power up? Or do I always have to start with the stepper like halfway my resistance?
Perhaps it should only decrease in SIM mode if the measured power > min_watts? Or does that break something else / do I do something wrong here?

> This one might be a little bit harder to implement correctly, but I'll think about it and look into it. In the meantime with Zwift, you could work around this by only connecting SS2k as a power meter and not a controllable trainer.

That is a possibility offcourse. I mostly start a course in sim mode and when I'm done I tend to keep riding on the same course to do my cooling down. During the cooldown I don't like the automatic resistance change and would rather stay at a (manually) set power level.

doudar, 2022-10-26:
If you’re using a new version of SS2K and have some type of power meter attached, it will generate travel limits as you’re pedaling. These are base off min brake watts and max brake watts and SS2K won’t let you go above or below those respectively.

Flo100, 2022-12-25:
It does that only in ERG mode right? Not in simulation mode?
When I start on top of a hill in simulation mode, my stepper rotates back beyond it's mechanical limit.

doudar, 2022-12-25:
> It does that only in ERG mode right? Not in simulation mode?
> 
> When I start on top of a hill in simulation mode, my stepper rotates back beyond it's mechanical limit.
> 
> 
> 
> Also I think that MIN_WATTS needs to be replaced by userConfig.getMinwatts() in ERG_Mode.cpp?
> 
> `  // set minimum SetPoint to MIN_WATTS if app sends setpoints lower than MIN_WATTS.
> 
>   if (newSetPoint < userConfig.getMinWatts()) {
> 
>     SS2K_LOG(ERG_MODE_LOG_TAG, "ERG Target Below Minumum Value.");
> 
>     newSetPoint = userConfig.getMinWatts();
> 
>   }`

Good catch! Yes, it should be.

Right now, the power table isn't persistent so it takes a few minutes of pedaling at various watt levels before the min and max positions are calculated. Prior to that, the limits are very large and the knob could crash.

What I'd like to do is have the offsets for the table saved and then the table regenerated once a known position is found. That should give you protection after just a few seconds of pedaling.

doudar, 2022-12-25:
> It does that only in ERG mode right? Not in simulation mode? When I start on top of a hill in simulation mode, my stepper rotates back beyond it's mechanical limit.
>
> Also I think that MIN_WATTS [...]

Current changes (you suggested) are going into the FTMS_Resistance branch, as that is currently undergoing testing for merging into develop. If you're interested in working on a persistent power table, I'd absolutely love the help. No pressure of course.

Flo100, 2022-12-27:
> This one might be a little bit harder to implement correctly, but I'll think about it and look into it. In the meantime with Zwift, you could work around this by only connecting SS2k as a power meter and not a controllable trainer. 

I implemented a simple 'solution' to this one.
In SmartSpin_parameters.h change _getERGMode()_ to:
`bool getERGMode() { return ERGMode || simulateTargetWatts; }`
And in BLE_Server.cpp change _case FitnessMachineControlPointProcedure::SetIndoorBikeSimulationParameters: {_  to:
`...`
`port = bytes_to_u16(buf[1], buf[0]);`
`rtConfig.setERGMode(false);`
 `if(!rtConfig.getERGMode()){`
 `  rtConfig.setTargetIncline(port);`
`   logBufLength += snprintf(logBuf + logBufLength, kLogBufCapacity - logBufLength, "-> Sim Mode Incline %2f", rtConfig.getTargetIncline() / 100);`
`}`
 `ftmsStatus = ....`

This way you can "Override" the simulation mode using the webinterface and ride with a fixed Powersetting. I use this when cooling down, because then I don't like the shifting up and down anymore.
Also when RGT (or Zwift) is set to ERG mode RGT is still leading and the Set Power can't be altered during the workout using SmartSpin2k.

> Holding down the shifters - I like it! That's a pretty cool idea. I'll definitely put it on my todo list. Changing target watts in ERG mode is interesting as well. I think it's something that could be done.
I also made changes to the Interrupt Functions of Main.cpp to get this working. 

`///////////// Interrupt Functions /////////////`
`void IRAM_ATTR SS2K::shiftUp() {  // Handle the shift up interrupt IRAM_ATTR is to keep the interrput code in ram always`
`  if (ss2k.deBounce()){`
`    if (!digitalRead(currentBoard.shiftUpPin)) {  // double checking to make sure the interrupt wasn't triggered by emf`
`      if(!rtConfig.getERGMode()) {`
`        rtConfig.setShifterPosition(rtConfig.getShifterPosition() - 1 + userConfig.getShifterDir() * 2);`
`      } else { // -> rtConfig.getERGMode()`
`        int newSetPoint = rtConfig.getTargetWatts() +10;`
`        if (newSetPoint > userConfig.getMaxWatts()) {`
`          newSetPoint = userConfig.getMaxWatts();`
`        }`
`        rtConfig.setTargetWatts(newSetPoint);`
`      } `
`    }// Probably Triggered by EMF, reset the debounce`
`  }`
`}`

`void IRAM_ATTR SS2K::shiftDown() {  // Handle the shift down interrupt`
`  if (ss2k.deBounce()){`
`    if (!digitalRead(currentBoard.shiftDownPin)) {  // double checking to make sure the interrupt wasn't triggered by emf`
`      if(!rtConfig.getERGMode()) {`
`        rtConfig.setShifterPosition(rtConfig.getShifterPosition() + 1 - userConfig.getShifterDir() * 2);`
`      } else { // -> rtConfig.getERGMode()`
`        int newSetPoint = rtConfig.getTargetWatts() -10;`
`        if (newSetPoint < userConfig.getMinWatts()) {`
`          newSetPoint = userConfig.getMinWatts();`
`        }`
`        rtConfig.setTargetWatts(newSetPoint);`
`      }  `
`    }// Probably Triggered by EMF, reset the debounce`
`  }`
`}`

`void SS2K::scanIfShiftersHeld() {`
`  if ((digitalRead(currentBoard.shiftUpPin) == LOW) && (digitalRead(currentBoard.shiftDownPin) == LOW)) {  // are both shifters held?`
` ....`
`  } else if(digitalRead(currentBoard.shiftDownPin) == LOW) {  // left shifter held?`
`    SS2K_LOG(MAIN_LOG_TAG, "Left shifter Held %d", shiftersHoldForScan);`
`    if (shiftersHoldForScan < 2) {  // have they been held for enough loops?`
`      SS2K_LOG(MAIN_LOG_TAG, "Left shifter Held < 2 %d", shiftersHoldForScan);`
`      if(!rtConfig.getERGMode()) { //In Simulation mode?`
`        rtConfig.setShifterPosition(0);`
`        SS2K_LOG(MAIN_LOG_TAG, "Decrease shifters to '0'");`
`      } else { //No, must be in ERG mode`
`        rtConfig.setTargetWatts(userConfig.getMinWatts());`
`        SS2K_LOG(MAIN_LOG_TAG, "Decrease Watts to MinWatts");`
`      } `
`      shiftersHoldForScan = SHIFTERS_HOLD_FOR_SCAN;`
`    } else {`
`      shiftersHoldForScan--;`
`    }`
`  } else {`
`    shiftersHoldForScan = SHIFTERS_HOLD_FOR_SCAN;`
`  }`
`}`

I posted this, so perhaps anyone can also benefit from this.

doudar, 2023-01-05:
Implemented in #420

Flo100, 2023-01-05:
> Implemented in #420

That was quick! :)

There is a bug with the above solution for the ERG mode when switched off again. (I never use it, but it should work offcourse).
When ERG mode is turned off the min and max positions needs to be set to default again otherwise the shifters won't shift as far as they should.

In settings.h I added three lines:
`// Default Min and Max step in Simulation mode. Will be recalculated in ERG mode.`
`#define DEFAULT_MIN_STEP -200000000`
`#define DEFAULT_MAX_STEP 200000000`

In smartSpin_parameters.h I changed line 44 & 45 to:
`  int minStep                = DEFAULT_MIN_STEP;`
`  int maxStep                = DEFAULT_MAX_STEP;`

And the void in line 75 to:
`  void setSimulateTargetWatts(bool swt) `
`  { `
`   simulateTargetWatts = swt; `
`    if (!swt) `
`    {`
`      minStep = DEFAULT_MIN_STEP; //reset minStep`
`      maxStep = DEFAULT_MAX_STEP; //reset maxStep`
`      targetIncline = 0;          //reset targetIncline to avoid big stepper increase when switching from ERG to Simulation or Stand-alone mode`
`      shifterPosition = 0;        //not really necessary, but should give a fixed starting point when turning ERG mode off.`
`    }`
`  }`

doudar, 2023-01-05:
> That was quick! :)
>
> There is a bug with the above solution for the ERG mode when switched off again. (I never use it, but it should work offcourse). When ERG mode is turned off the min and max [...]

Here's how I've done it in #420 

I don't like adding much to ISR's so I instead created a new function that is run in the maintenance loop which checks for shifts in the ISR and then applies them depending on what the current FTMS mode is. That way, customized shifting can be easily implemented in every FTMS mode. Also, if you read this code, I started a refactor on the rtConfig class which you might notice. Watts, Resistance, Cadence and Heartrate were added to "Measurements". 

https://github.com/doudar/SmartSpin2k/blob/fbc9ef99576e07403d29b93d6e6a39c0aeadd94e/src/Main.cpp#L236-L283

Flo100, 2023-01-08:
Nice :)
I flashed the resistance branch to my setup and it took me a while to get it working. Seemed like my Wi-Fi passphrase was too long and I ended up with no free space left in the filesystem. After I increased that it worked like a charm. :)

> There is a bug with the above solution for the ERG mode when switched off again. (I never use it, but it should work offcourse).
When ERG mode is turned off the min and max positions needs to be set to default again otherwise the shifters won't shift as far as they should.

Still, when switching from ERG to simulation mode the min and max stepperpositions needs to be set wide again (because ERG mode limited these points). This is also the case when you turn off ERG mode in RGT.

I like the webinterface (SimTargetWatts) to overrule the incline data when connected to RGT so I made some changes to BLE_Server.cpp to get that working again. Because of the use of FTMSMode a little more lines were needed than my solution above.

I also added the _else_ back in the scanIfShiftersHeld to switch to '0' if the left shifter is held.

doudar, 2023-01-08:
> Nice :) I flashed the resistance branch to my setup and it took me a while to get it working. Seemed like my Wi-Fi passphrase was too long and I ended up with no free space left in the filesystem. After I increased that it worked like a charm. :)

Did you need to increase https://github.com/doudar/SmartSpin2k/blob/84eaafbb4774be8fc1cb499a6eeace65a89d8718/include/settings.h#L234 or something else?

Flo100, 2023-01-10:
I tried that first, but made no difference.
It seemed that Config.txt could not be written due to lack of free space. When I entered less password characters it wrote the file without error.
So I ended up making a custom_partition.csv and expanding the filesystem partition by leaving out the coredump.
Perhaps not the nicest way, but it worked.

Flo100, 2023-01-10:
I also noticed that the BLE tries to reconnect from Main while I have a pm connected and hr set to "none"
Should the 'ifs' in checkBLEReconnect() be something like this:
`if (((strcmp(userConfig.getConnectedHeartMonitor(), "none") == 0) && (spinBLEClient.connectedPM))) {   // Exit if "none" PM and HR is connected`
`    bleCheck = 0;`
`    return;`
 ` }`
Because of the const char compare?

doudar, 2023-01-10:
yes, they need to be cast to string type.
https://github.com/doudar/SmartSpin2k/blob/eaa165a81525f1a3f110c330d2e3754652e3e22d/src/Main.cpp#L560-L577

Flo100, 2023-01-10:
> yes, they need to be cast to string type. It's been fixed in FTMS-resistance. I hope to merge this branch into develop soon

I see now you already fixed this 2 days ago :)
When it's merged I'll flash back to develop and give it a go!
I think we can close this one now.

Flo100, 2023-01-10:
On the other hand maybe we can't close it yet.
I discovered that I can switch 1 step back when I'm already at minResistance and one step up when already at maxResistance.  
And then the stepper slowly steps back to the correct target and keeps repeating over and over because the shifterposition is still to low or to high. 
I changed < and > to <= and >= in line 273 and 274 below so it doesn't switch when it is already on min or max resistance.
`if ((ss2k.targetPosition > rtConfig.getMaxStep()) || (rtConfig.resistance.getValue() >= rtConfig.getMaxResistance()) || (ss2k.targetPosition < rtConfig.getMinStep()) ||
            (rtConfig.resistance.getValue() <= rtConfig.getMinResistance())) {`
https://github.com/doudar/SmartSpin2k/blob/eaa165a81525f1a3f110c330d2e3754652e3e22d/src/Main.cpp#L273-L279

doudar, 2023-01-10:
Whoa! Resistance numbers? That's experimental in this branch - what bike are you using that you're getting resistance feedback? I could really use another tester if you have a bike with resistance! (Especially an Echelon)

Flo100, 2023-01-11:
No sorry, I don't have an Echelon bike.
I made my own setup a few years ago with a simple spinningbike, stepper and nextion display. I mounted a FSR under the felt pad to measure the resistance. I mainly use this when zeroing the stepper at startup and from that point it was just a matter of counting steps and stop at max resistance. I only had the incline from RGT and Zwift working in simulation mode.

I like the ERG mode you implemented, that's why I ported your software to my setup and since it now has resistance too, I thought I'll give it a go.
If I switch fast enough (and keep the stepper spinning) I can still switch beyond the limits because the resistance hasn't changed enough yet for the shifts to be blocked. It obviously did not reach its destination yet (That's why the stepper is still spinning offcourse).
But I don't know if that's normal behaviour for a resistance bike or that it's just my setup.
I use the nextion for displaying the parameters like power, heartrate and cadence and to switch some settings like ERG mode or turn the fan on or off. And offcourse this way I get all the parameters directly on my bike too in stand-alone mode.

doudar, 2023-01-11:
Awesome! That sounds pretty cool!
Your setup is spot on - that's an exact duplicate of the last remaining bug (so far) that we've found in testing with the Peloton.

doudar, 2023-01-11:
Here's a new binary (with @Flo100 's oscillation fix) for anyone else that might be watching this thread. Unzip and make sure it's named exactly firmware.bin then use the web interface admin/admin. 

Also, as a reminder, some of the HTML files have also 
[firmware.zip](https://github.com/doudar/SmartSpin2k/files/10395951/firmware.zip)
changed in this release so if you manually update, you will have to update those from the FTMS_resistance branch individually as well. You'll find them in the data folder. https://github.com/doudar/SmartSpin2k/tree/FTMS_Resistance/data

Flo100, 2023-01-12:
> Awesome! That sounds pretty cool! Your setup is spot on - that's an exact duplicate of the last remaining bug (so far) that we've found in testing with the Peloton.
> 
> The delay is why we have the Peloton resistance limits at 5&98. The actual bike is 0 and 100 but there needs to be a buffer for overshoot.

Before I had BLE working I wanted some sort of reproducable resistance level with each workout and that's why I mounted the FSR. After that used to I find the zero point at startup, reset the stepper to zero and use this as stepper minimum. My FSR gets saturated over time because there is more force on the felt pad than expected and I actually need another type. But I counted the steps to maxResistance once and used that as upper limit.

You could probably do something like that too. Find the zero point at startup and after that use minStep() as boundary. 
You can also find the maxStep this way, only it takes some time, so perhaps implement a calibration setting in config which stores maxStep?
If you reset the stepper (stepper->clearPulseCounter()) at the found zero point minStep will be 0 and you only have to store the number of steps until maxResistance (maxStep).
Offcourse this interferes with ERG mode... so minStep and maxStep needs to be initialized after switching modes or perhaps add minResistanceStep and maxResistanceStep?

doudar, 2023-03-08:
Merged into develop for testing.

doudar, 2023-12-18:
Fixed

closed 2023-12-18 by doudar