doudar/SmartSpin2k #404: Perhaps have SmartSpin2k Poll To Reconnect To Devices When They Are Back On Again

mashaalmemon, 2022-11-25:
**Is your feature request related to a problem? Please describe.**

I'm using my `SmartSpin2k` with a Bowflex C6.

I leave the `SmartSpin2k` plugged in overnight. In that time my Bowflex C6 turns off because it's not in use. When I come back in the morning I need to explicitly "Save Dropdowns" / "Connect To Bluetooth Devices" again. I find myself having to hit the user interface to get this going.

**Describe the solution you'd like**

Is it possible to have the `SmartSpin2k` try to reconnect to the previous device by polling or something like this every 5 seconds perhaps, while it is on, if a connection is lost and no other attempt to connect to another device is made? That way when the C6 is started back up it'll reconnect itself?

**Describe alternatives you've considered**

No other alternatives I can think of other than what I find myself doing now.

**Additional context**

Let me know if you need anything more from me.

doudar, 2022-11-25:
I'll think about implementing a repeating scan if a specific BLE device is selected.

I the meantime what might work for you is if you hold down both shifter buttons for ~5 seconds, it will initiate a scan/reconnect without having to go to the web interface.

mashaalmemon, 2022-11-25:
Hi @doudar,

Thanks for the response. Got it. @eMadman suggested the shifter buttons reset option as well, which I'll try.

I didn't know about the possibility of resetting to defaults. Hopefully I won't do that by accident as its not not clear how to tell if it's rebooting/restarting unless I initiate it (there is not display or lights to indicate the rebooting is in progress).

But if you get a chance to implement the polling, would be super slick. Thanks for considering the option.

doudar, 2022-11-27:
Testing auto reconnect branch. https://github.com/doudar/SmartSpin2k/tree/auto_reconnect/Hardware

doudar, 2022-11-27:
See #405

doudar, 2022-11-29:
new binary for testing in the PR:
https://github.com/doudar/SmartSpin2k/pull/405#issuecomment-1328416065

doudar, 2022-12-07:
Fixed by #405