doudar/SmartSpin2k #413: [BUG] Setting BT source to "none" is resetting to Any

eMadman, 2022-12-16:
### Description
This is affecting a user trying to work with QZ.

The "None" setting on the bluetooth settings page is not sticking - it's going into the "device not found" routine

### Steps to reproduce

1. Set Power Meter or HRM (or both) to none on the Bluetooth Settings Screen
2. Save changes

**Expected behavior**
SS2K stops seeking BLE connection

**Actual behavior**
SS2K reboots and initiates bluetooth scan, connecting to nearby devices

doudar, 2022-12-16:
I'll check into it. Is this only on the newest firmware? Is there a log?

rp8082, 2022-12-16:
I’m the guy that is having the issue. I am able to change the HRM from any to none. But I can’t change power meter from any to none. After pressing “Save Changes”, the software says “setting saved” but the stalls out when refreshing page. It hangs there until I exit out of that screen. I don’t know how to capture a logging file for you b

rp8082, 2022-12-16:
Doudar Great, I just saw your update to Emad. Thanks!

doudar, 2022-12-16:
> I’m the guy that is having the issue. I am able to change the HRM from any to none. But I can’t change power meter from any to none. After pressing “Save Changes”, the software says “setting saved” [...]

No problem - I was able to reproduce it. Thanks for the report! 

This should be an easy fix.

doudar, 2022-12-16:
Fixed by #414 
[firmware.zip](https://github.com/doudar/SmartSpin2k/files/10247392/firmware.zip)

Here's a testing binary. Unzip - rename exactly firmware.bin and upload via the web interface admin/admin

closed 2022-12-16 by eMadman

rp8082, 2022-12-16:
Hello Anthony. Just uploaded the new firmware. Works perfectly.  Also, QZ app is not controlling it all and is taking resistance values from Fulgaz and changing resistance per the grade changes. Thank you for the rapid patch on the firmware!!!

PR #414 (2022-12-16): Dropdown fix
Fixes #413