doudar/SmartSpin2k #426: [BUG] Filesystem gets updated when AutoUpdate is set to false

Flo100, 2023-01-04:
### Description
The filesystem gets updated when a newer version is available when autoupdate is set to false.
Updating of the firmware is ignored as it should.

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

No update

**Actual behavior**

Filesystem gets updated

### Additional Information
* If you provide debug output and it is long, consider using [Pastebin](https://pastebin.com/) and paste a link in this issue to keep the issue easier to read. 
* Provide pictures or links to videos that clearly demonstrate the issue.

In HTTP_Server_Basic.cpp change line 635 [ _void HTTP_Server::FirmwareUpdate()_ ]to:
` if ((availiableVer > currentVer) && (userConfig.getAutoUpdate()) || (updateAnyway)) {`

doudar, 2023-01-05:
Good idea - my concern is that reset to defaults currently wipes the file system and I don’t have a good solution to get the html files loaded again (other than reloading each file individually.) Optimally, they could get loaded with the smartspin2k flasher utility but I don’t have much time to update it.

Flo100, 2023-01-05:
I don't think I understand correctly.
In case of missing index.html (after a 'reset to defaults'?) the filesystem gets updated by the updateAnyway flag, so this should work. 
Offcourse the newest version will be downloaded and I also don't know how to redirect to a specific version.

But when Index.html IS present: the filesystem also gets updated when a newer version is found.
This is because the && userConfig.getAutoUpdate() is missing in the filesystem update check.
It is already present in the formware check.

doudar, 2023-01-23:
Okay - thanks! I should be able to incorporate this into the next version now that the last big push is over.

closed 2023-02-25 by doudar